SUSTAINA-OP/SUSTAINA-OP #1: GankenKun167機 製作によるCADの修正

MasatoKubotera, 2021-02-15:


MasatoKubotera, 2021-02-15:
<h1>M2ねじ下穴ドリル径の修正</h1>

M2ねじ下穴ドリル径は1.6mmであるが，1.8mmとなっていたため修正した．

修正したパーツは下記のとおりである．

- <h2>カメラ_メインフレーム1_KBCR-S02MU.ipt </h2>

  <code>上半身\頭\カメラ_メインフレーム1_KBCR-S02MU.ipt</code>

  (参照スケッチ：カメラ_メインフレーム1_基礎.ipt <code>上半身\頭\カメラ_メインフレーム1_基礎.ipt</code>)

  [image]

- <h2>カメラ_メインフレーム.ipt</h2>
  
  <code>上半身\頭\カメラ_メインフレーム.ipt</code>

  [image]

MasatoKubotera, 2021-02-15:
<h1>側面ねじ径の修正</h1>

- <h2>上腕_メインフレーム_1_1.ipt</h2>

  <code>上半身\腕\上腕_メインフレーム_1_1.ipt</code>

  M3雌ネジであるべきところが，M2となっていた．

  [image]

MasatoKubotera, 2021-02-15:
<h1>ねじ穴の追加</h1>

- <h2>SwitchCov.ipt</h2>

  <code>上半身/コックピット/コックピット_2/SwitchCov.ipt</code>

  <a href="https://akizukidenshi.com/catalog/g/gM-08468/">電圧計</a>取り付け用のねじ穴を追加．

  [image]

MasatoKubotera, 2021-02-24:
<h1>inventor 設計ミス</h1>

下記の2つの部品に側面のM3ネジ穴がありませんでした．

- <h2>コックピット_2_サブフレーム_2.ipt</h2>

  <code>上半身/コックピット/コックピット_2/コックピット_2_サブフレーム_2.ipt</code>

  [image]

- <h2>コックピット_2_サブフレーム_5.ipt</h2>

  <code>上半身/コックピット/コックピット_2/コックピット_2_サブフレーム_5.ipt</code>

  [image]

MasatoKubotera, 2021-02-28:
<h1>inventor 設計ミス</h1>

- <h2>カメラ_メインフレーム.ipt</h2>

  <code>上半身\頭\カメラ_メインフレーム.ipt</code>

  画像の上方向からネジを挿入するが，青色選択部分が曲面になっている．

  [image]

Keita1Mori, 2021-03-01:
足裏の色が大会規定では黒になっているのですが、167の足裏は今現在青色で違反になる可能性があると思うので、塗装するべきだと思われます。それと同様にGFRPが青色だということでこちらも大会規定違反になりかねないので塗装
した方がいいと思われます。

YugoNishio, 2021-03-08:
inventor　設計ミス
・足首_サブフレーム_1_3.ipt

足首_サブフレーム_1_3の穴がCADでは2.4㎜となっていますが、正しくはM3用のガバ穴です。
CADのデータが間違っていたため、M3用のねじ切りがされていました。
CADのデータを3.1㎜に変更し、穴の拡張を行いました。

[image]

NaokiIwasawa, 2021-04-04:
Inventor 設計ミス

・付け根ロール_サブフレーム_6_5
・付け根ロール_サブフレーム_6_6

上記のパーツに関して以前ガタが大きかったことから[issue](https://github.com/citbrains/citbrains_humanoid/issues/1140#issuecomment-496184016)で解決していたが、OpenPlatform_ver3_0では変更が反映されていなかった。そのため167の切削を行うにあたってCADを変更した。
変更内容：はめあいのクリアランスを0.1→0.01㎜にした（2.6→2.51，11.1→11.01）
[image]

AD58-3104, 2021-04-09:
Inventor作図ミス
画像のスペーサの向きが実機と逆だった。実機では広い面が内側になっている。
[image]

yasuohayashibara, 2021-04-12:
こちら意図的かもしれません．
剛性を高めるためにいろいろと試行錯誤をしたので，その時の修正案の可能性があります．

NaokiIwasawa, 2021-06-02:
付け根ロール_メインフレーム_1_1についてz軸の精度問題があり再切削を行ったが、その時にポケット加工は剛性に影響が低く([issue](https://github.com/citbrains/citbrains_humanoid/issues/1140#issuecomment-495840529))かつ切削時間が延びることから必要性がないとの判断をしたので、廃止をした。
[image]

NaokiIwasawa, 2021-06-02:
以上で「GankenKun167製作によるCADの修正」報告が終了したのでissueをcloseする。

closed 2021-06-02 by NaokiIwasawa