SUSTAINA-OP/SUSTAINA-OP #2: GankenKun163機体 整備記録

MasatoKubotera, 2021-04-21:


MasatoKubotera, 2021-04-21:
# 第1回 状態把握 2021/4/21 13:00~17:00

## メンバー
- @MasatoKubotera 
- @Keita1Mori 
- @RyodaiArai 
- @NaokiIwasawa 
- @YukiShigematsu

YukiShigematsu, 2021-04-21:
[image]

YukiShigematsu, 2021-04-21:
[image]

YukiShigematsu, 2021-04-21:
[image]

Keita1Mori, 2021-04-21:
@Keita1Mori @RyodaiArai 
# 右足
## 問題点
- リンクふくらはぎ、リンクふくらはぎサブフレームの素材がアルミではなくPOMになっていました。

[image]
[image]

- 付け根サーボヨー軸方向から膝ヨー軸方向サーボケーブルのカバーが破れていました。その影響で導線が見えていました。

[image]

[image]

- 右足のヨー軸方向のサーボホーンとリンクふくらはぎサブフレームがねじ剤の影響でくっついて取れませんでした。なので故障しているサーボからホーンを取り出して取り替えました。

- 減速ホーンのところですがガラエポを固定するところにプラスネジを使用していますが、本来は六角ねじで固定するので再度注文した後に変更します。

[image]

- サーボを固定するために使うねじの長さが違いました。ので変更します。

[image]

- 付け根メインフレームが歪んでいたので再切削します。

[image]
[image]
[image]

-バリ取り不足があったので再度バリ取りをしました。

hiroki-0001, 2021-04-22:
4/22のハードウェア班のミーティングの議事録です。
https://docs.google.com/document/d/1aOQCK8o2MZA2gD-s7GhieVNz_ME0s32JHOgp3uHcP6s/edit?usp=sharing

MasatoKubotera, 2021-04-28:
状態把握 2日目
メンバー
@MasatoKubotera
@Keita1Mori

対象箇所
右脚

作業時間
13:00〜17:00

Keita1Mori, 2021-04-28:
@MasatoKubotera @Keita1Mori 
# 右足
## 樹脂問題点
- 切削精度の問題により取り外しにくい(付け根_サブフレーム_6_2)
[image]

- 切削精度の問題によりZ軸方向の精度が出ていない(膝_同軸_スペーサ_1_5)
[image]

- POMが摩耗しています(付け根_サブフレーム_1_4)
[image]

## ねじ問題
- サーボ固定に使用しているものが六角ではなくプラスネジを使用していました。
[image]

- 膝の同軸の固定するネジに赤ネジ剤、青ネジ剤、使ってないのがあってバラバラだったため統一した方がいいと思います。
[image]
[image]
[image]

- ネジ剤が溶けて取り外し不可
[image]

## サーボ固定問題
- ネジ剤が溶けてしまってサーボホーンとリンク_ふくらはぎ_サブフレーム_1_4がくっついて取れない状態でした。
[image]

- サーボの軸が曲がっていました。
[image]

yasuohayashibara, 2021-05-05:
しっかり調査していて非常に良いですね．
このような調査を繰り返すことで，ロボットの動きが格段に安定すると思います．
以下，コメントです．

> * リンクふくらはぎ、リンクふくらはぎサブフレームの素材がアルミではなくPOM

こちら指摘する人がいませんでしたので林原が回答しますが，中島君から調査のためにPOMを使用しているとの引き継ぎがされていたような気がします．
実験用に残しても良いですが，統一するという観点からはアルミを使用したほうが良いと思います．

> * 減速ホーンのところですがガラエポを固定するところにプラスネジを使用していますが、本来は六角ねじで固定する

キャップは高さがあるので，低頭を使用しないと干渉するかもしれません．
ご注意ください．

> * サーボ固定に使用しているものが六角ではなくプラスネジを使用

こちらも高さの問題からナベネジを使用しているおそれがあります．

> * ネジ剤が溶けて取り外し不可

樹脂は樹脂用のロックタイトを使用しないと溶けますのでその影響でしょうか．
ちなみに，青色のロックタイトでも樹脂用でないものがありますので，ご注意ください．

> * サーボの軸が曲がっていました

こちら興味があります．
時間があればサーボモータのどこの部分が曲がっているかを更に調査してもらえると助かります．
前世代のロボットのサーボモータ（Futaba製）はギアの最終段の軸の強度が十分ではなく，１年ぐらい経過したサーボモータは十分なトルクを出力できない状態になっていました．
モータの出力軸には繰り返し衝撃力が加わるので，もし最終段の軸が広がっているのであれば，期待した性能が出せないサーボモータになってしまっているおそれがあります．

サーボモータの劣化を測る指標があると良いですね．
もしそれが分かればフルメンテナンス時に交換ということもできますので．．．

MasatoKubotera, 2021-05-06:
# 状態把握 3日目

## メンバー
- @MasatoKubotera 
- @Keita1Mori 
- @NaokiIwasawa 
- @YukiShigematsu 

## 対象箇所
- 左脚
  - @NaokiIwasawa
  - @YukiShigematsu

- 右脚
  - @MasatoKubotera 
  - @Keita1Mori

## 作業時間
17:00~19:00

Keita1Mori, 2021-05-06:
@MasatoKubotera @Keita1Mori 
# 右足首
## 樹脂問題点
- 問題なし
## ねじ問題
- 足裏_メインフレーム_2_3と足裏_サブフレーム_1_1を固定するためのねじの長さが本来８ｍｍのところを６ｍｍで使用されていました
- スパイク部分に使用されている[BRU-306.5S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5S&PNSearch=BRU-306.5S&KWSearch=BRU-306.5S&searchFlow=results2products)ですがねじを切っていたり二連結にされているので[ARU-313S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5S&PNSearch=BRU-306.5S&KWSearch=BRU-306.5S&searchFlow=results2products)を使って固定したいと思います。

- 足裏全体
[image]

- 足裏スパイク部分
[image]

## パーツ変形

- 右足の内側の足首_独立ピッチ_1_1負荷がかかって曲がっていました。
[image]

---

- 右足の外側の足首_独立ピッチ_1_2を足首_独立ピッチ_1_1にはめ込む際に最後まではめ込まないでねじで無理やりつけていたため足首_独立ピッチ_1_2が変形していました。

- 付いている時
[image]

- 表
[image]

- 裏
[image]

---

- 足首_メインフレーム_1_1と足首_メインフレーム_1_1_MIRが負荷がかかっていたため変形していました。

- 足首_メインフレーム_1_1　全体
[image]

- 足首_メインフレーム_1_1　詳細
[image]

- 足首_メインフレーム_1_1_MIR　全体
[image]

- 足首_メインフレーム_1_1_MIR　詳細
[image]

---

- 足首_サブフレーム_1_1も負荷がかかって曲がっていました。
[image]

---

- 足首_サブフレーム_1_3に負荷がかかって曲がっていました。なのではめ合わせをした際にガタガタになっていました。
[image]

---

- 足裏_メインフレーム_1_2と足裏_メインフレーム_1_1は曲がっていませんでした。
[image]

NaokiIwasawa, 2021-05-06:
<h1>左足ひざ下から足首</h1>

<h2>膝裏同軸スペーサの隙間</h2>
片側のみスペーサと膝メインフレームの間に0.3mm隙間があった
[image]

<h2>青ネジ剤が多い</h2>
リンク_ふくらはぎ_サブフレーム_1_2(予備案)とリンクの固定に青ネジ剤が多めに使われていた
[image]

<h2>足首_独立ピッチ_1の歪み</h2>
足首_独立ピッチ_1が左右で歪んでいるた
[image]

<h2>足首_独立ピッチ_1_2の歪み</h2>
外す際にはめ合わせがきつかったが外して確認する歪みがあった。特徴として四か所のネジ部分がもりあっがている。
[image]
[image]
[image]

上記の歪みが解体する前に確認できたので示す
本来は50.5mmであるが49.6mmとなっていた

[image]

yasuohayashibara, 2021-05-06:
> * 足首_サブフレーム_1_3に負荷がかかって曲がっていました。なのではめ合わせをした際にガタガタになっていました。

本ロボットは，嵌め合いで強度を保っているところがあるため，嵌め合いの部分にガタがあると，ロボットは姿勢を保つことができなくなるおそれがあります．
これが新しいロボットの歩行が安定する要因かもしれません．
逆にこの部分を作り直す，もしくは固定すると動きが良くなるかもしれません．

極端な例ですが，これにより歩行がかなり安定しました．
https://github.com/citbrains/citbrains_humanoid/issues/1140#issuecomment-496184016

Keita1Mori, 2021-05-07:
> こちら興味があります．
時間があればサーボモータのどこの部分が曲がっているかを更に調査してもらえると助かります．
前世代のロボットのサーボモータ（Futaba製）はギアの最終段の軸の強度が十分ではなく，１年ぐらい経過したサーボモータは十分なトルクを出力できない状態になっていました．
モータの出力軸には繰り返し衝撃力が加わるので，もし最終段の軸が広がっているのであれば，期待した性能が出せないサーボモータになってしまっているおそれがあります．
>サーボモータの劣化を測る指標があると良いですね．
もしそれが分かればフルメンテナンス時に交換ということもできますので．．．

こちらについてですが今日調査したので指摘されたコメントのところで編集しました。
> - 後の調査からサーボの内側にある出力軸とホーン取り付け部分が曲がっていました。

NaokiIwasawa, 2021-05-07:
# 状態把握 4日目

## メンバー
- @MasatoKubotera 
- @Keita1Mori 
- @NaokiIwasawa  

## 対象箇所
- 左脚
  - @NaokiIwasawa

- 右脚
  - @MasatoKubotera 
  - @Keita1Mori

## 作業時間
17:00~19:00

NaokiIwasawa, 2021-05-07:
<h1>左足足首</h1>

<h2>足首の隙間</h2>
足首_メインフレーム_1_1と足首_サブフレーム_1_1のはめ合わせ部分に0.09と0.3ｍｍの隙間があった
[image]
足首_メインフレーム_1_1と足首_サブフレーム_1_2のはめ合い部分にも隙間がった
[image]

<h2>足首メインフレーム_1_1の歪み</h2>
上記の通りはめ合いに隙間があったので確認すると、足首メインフレーム_1_1が歪んでいた。左右とも歪んでいたが歪んでいる方向は逆だった。
[image]
[image]

<h2>足裏_サブフレーム_1_1と足裏_1_1の固定ネジ</h2>
本来ネジの長さは8mmが適切であるが、かかとの方は6mmのネジで固定されていた。
[image]

Keita1Mori, 2021-05-08:
# 報告

> - ネジ剤が溶けて取り外し不可
- このように報告しましたが今日分解したところ圧力がかかっていたため取れていない状態でした。なので一旦樹脂用のネジ剤案を止めたいと思います。

- 解体前
[image]

- 解体後
[image]

- 詳細
[image]

hiroki-0001, 2021-06-05:
# 2021年6月5日　整備報告
責任者：@MasatoKubotera   
作業者：@hiroki-0001  
活動時間　11:30～
# 足の状態の再確認(状態報告)

## [右足]
### <アルミ部品>
**歪みあったパーツ**
- 足首_メインフレーム_1_1_MIR × 1
- 足首_メインフレーム_1_1 × 1
- 付け根_メインフレーム_2_1 × 1
- 付け根_メインフレーム_2_1MIR × 1
- 足首_独立ピッチ_1_1 × 2
- 足首_独立ピッチ_1_2 × 2
- 付け根ロール_サブフレーム_6_6 × 1
- 付け根_サブフレーム_1_2 × 1
- 足首_サブフレーム_1_1 × 1
- 足首_サブフレーム_1_2 × 1
- 足裏_メインフレーム_1_2 × 1
- リンク_ふくらはぎ_サブフレーム_1_4 × 2
  - 本来は4個あるが、内2つはネジ剤による影響でサーボから取り外しできなくなったためカウントしていない。記載してある個数(2個)はサーボから取り外せたもので歪みがあるパーツをカウントしている。またこの歪みはサーボから取り外す際に発生した歪みである可能性が高い。
---
**歪み + ナットの深さの修正が必要なパーツ**
-  付け根_フリー軸 × 2
-  足首_サブフレーム_1_3 × 1
-  付け根_メインフレーム_6_1 × 1
-  付け根_サブフレーム_1_3 × 1

---
**歪み + タップの穴あけミスがあったパーツ**

- リンク_ふくらはぎ_サブフレーム_1_2(予備案) × 1
- 足裏_サブフレーム_1_1 × 2
---
**歪み + バージョン違い**

- 付け根ロール_サブフレーム_6_5 × 1
- 膝_メインフレーム × 2

---
**歪み + タップミス + 材質違い**

＊(本来はアルミ このパーツはPOMで作成されていた)
- リンク_ふくらはぎ_サブフレーム_1_1(予備案)

---

### <ガラエポ樹脂部品>
**歪みありパーツ**
- リンク_メインフレーム_6_4 × 1  

### <POM部品>
**摩耗していたパーツ**
- 膝_同軸_スペーサ_1_5 × 6
- 付け根_サブフレーム_6_2 × 1
- 付け根_サブフレーム_6_4 × 2
- 足首_スペーサ_1_1 × 4
- 付け根_サブフレーム_1_4
  - 165で確認した同一パーツは高さが変わっていたがこちらでは高さは変わっていなかった。

## [左足]
### <アルミ部品>

### <ガラエポ樹脂部品>

### <POM部品>

# 新たに判明した問題点

hiroki-0001, 2021-07-07:
# 整備報告 2021年7月6日

## 胴体(下側)

[image] 

- 胴体部分のPOMが白い
  - レギュレーション違反になるため再切削
<img src ="https://user-images.githubusercontent.com/53966346/124718690-5103bd80-df41-11eb-98ae-2c0413a8051d.jpg
" width="320px"> 

- 画像のように目に見える隙間が存在する
[image] 

- 画像の胴体固定部分に使用しているネジの長さが短い
  - 165では長さが8mmだが、163で使用されているネジの多くは6mmだった
[image] 

- サーボを固定するネジの長さが一様でない

- パーツに歪みがあった(以下歪みがあったパーツ)
  - 胴体_メインフレーム_1_1
  [image] 
  - 胴体_サブフレーム_1_4(MIRパーツも同様)
  [image] 
  - 胴体_メインフレーム_1_2
  [image]

yasuohayashibara, 2021-07-07:
白ですが，ある程度許容されているかもしれません．
以下のルールブックには禁止と受け取れる表現がありますが，表紙になっているロボットはそもそも白色をしています．

https://humanoid.robocup.org/wp-content/uploads/RCHL-2020-Rules-Dec23.pdf

hiroki-0001, 2021-07-08:
# 整備報告 2021年7月6日
## 胴体(上側)
- 165の整備(https://github.com/citbrains/OpenPlatform_ver3_0/issues/3#issuecomment-859480100)の時とは違い、胴体_サブフレーム_1_9と胴体_サブフレーム_1_11・胴体_サブフレーム_1_10のはめあいの歪みが大きくなかった。

    [image] 

## 頭・首
- 首の根本のパーツ(頭_サーボホーンフレーム)のバージョン違いを発見した。

    [image] 
    [image] 

- 歪みのあったパーツは以下の通りである。
  - 頭_メインフレーム × 2
    - このパーツは複雑に歪んでいる(ロール側・ピッチ側・ヨー側)この歪みにより、首のサーボの角度を基準角度に戻しても、1~5°程度ずれている可能性がある。このパーツは頻度を決めて定期的にパーツ交換を行ったほうがよさそうである。
  
    [image] 

  - 頭_サブフレーム × 3 (うち1つはバージョン違い)
  
    [image] 
    
    [↑↑↑↑↑バージョン違い↑↑↑↑↑↑]

    [image] 

    [image] 

## 肩
- 画像の部分に黒いテープが貼られていた。アルミパーツの反射を防ぐための応急処置だった可能性が高い。

    [image] 

## 腕
- 前腕部分に貼られていた黒いテープを外すと、前腕_サブフレーム_1_1が白いPOMで作成されていたことがわかった。
    [image]

yasuohayashibara, 2021-07-10:
> * 画像の部分に黒いテープが貼られていた。アルミパーツの反射を防ぐための応急処置だった可能性が高い。

肩周辺に貼られている黒いテープは，肩をボールなどと見間違えないための処置となります．
試合前に肩がボールやゴールや白線に認識されることがあり，その場しのぎの対策として肩に黒いテープを貼ることがしばしばあります．
別件になりますが，このようなことが無いように，画像のデータセット収集のマニュアルも作っておいたほうが良いかもしれませんね．

hiroki-0001, 2021-07-19:
# 整備報告 2021年7月19日

  責任者：久保寺  
  担当者：野口

  ## 腕
  ### <右腕>
  - 右肩ロール軸のサーボのフリーホーンのベアリングが外れてしまっていた。またベアリング自体の回転も悪くなっていた。フリーホーンのベアリングは確認していなかったため、再確認の必要性あり？

    [image] 

    [image] 

  - 写真のように,フリーホーンがベアリングから外れかけていた。軸が曲がっているのが原因？

    [image] 

  - 右肘ピッチ軸のサーボのホーンと部品を固定するネジが加工されていた。また加工の精度が写真のように非常に悪い。
    
    [image] 

  ### <左腕>
  - 大きな歪みありパーツ
    - 肩_サブフレーム_1_1 × 1
    [image]   

    - 上腕_サブフレーム_1_1 × 2
    [image] 
    [image] 

    - 上腕_メインフレーム_1_1 × 1
    [image] 

    - 前腕_サブフレーム_1_1 × 1
    [image] 

  ## 腕全体を通しての問題点

  - 写真のようにサーボホーンとパーツを固定するネジが後加工されたものを使用されていた。またこのネジは加工の精度が非常に悪く、サーボホーンのネジ穴を傷つけてしまっている。
    [image] 

    [image]

yasuohayashibara, 2021-07-20:
肩に関して，以下の写真を見るとこすった跡が見られますので，以下が生じたおそれがあると思います．

肩のどこかのネジが緩んで以下の部品と干渉
→ 通常は加わらない大きなスラスト力が加わりベアリングが故障

ラジアル方向の力の場合は，球が潰れることがあっても，外れることはあまり無いです．

[image] 

あと，ネジの教訓として以下があると思います．
- 面倒でも必要なネジは常にストックしておきましょう．

hiroki-0001, 2021-07-28:
# 整備報告 2021年7月9日 (投稿し忘れていましたすいません)
責任者：久保寺
担当者：野口
## 胴体(上側)
- 胴体_サブフレーム_1_11と首ヨー軸のサーボモータを固定するネジの長さが画像のようにバラバラに使用されていた。(ネジの長さは写真より右から11mm, 15mm, 20mmであり、**20mmのネジでしかネジ締結ができていなかった。**)
  [image] 

- 胴体_サブフレーム_1_4でネジ穴付近の傷が均一でなかった。これはネジ締結で加わる力が一定でないことを示している。また過度に傷がついているものは、雌ねじ側のPOMのネジ穴がバカになっているため、起こった現象と見られる。この現象は対照パーツであるMIRパーツにも同様に起こっていた。

    [image] 

    [image] 

## 歪みあったパーツ(胴体)
- 胴体_サブフレーム_1_10 × 1

    [image]

    [image] 

- 胴体_サブフレーム_1_11 × 1

    [image] 

    [image] 

- 胴体_サブフレーム_1_4 × 1

    [image] 

    [image] 

- 胴体_サブフレーム_1_4_MIR × 1

    [image] 

    [image] 

- 胴体_サブフレーム_1_3 × 2

    [image] 

    [image] 

    [image] 

    [image] 

- 胴体_メインフレーム_1_12 × 3

    [image] 

    [image] 

    [image] 

    [image] 

    [image] 

    [image] 

    [image] 

- 胴体_メインフレーム_1_13 × 2

    [image] 
    
    [image] 

    [image] 

    [image] 
  
## 頭

- 165と異なり、カメラ_外装とカメラのレンズ部分のはめあいがきつすぎなかったため、カメラ_外装を破壊することなく、カメラモジュールを取り外すことができた。

    [image]

MasatoKubotera, 2021-08-15:
整備が終了しているため閉じます．

closed 2021-08-15 by MasatoKubotera