SUSTAINA-OP/SUSTAINA-OP #3: GankenKun165機 整備記録

MasatoKubotera, 2021-04-21:
# 165機 整備メンバー

## 責任者
- @MibuchiYuta 

## 作業者
- @hiroki-0001 
- @AD58-3104

MibuchiYuta, 2021-04-22:
# 状態把握 1日目

## メンバー
@MasatoKubotera 
@MibuchiYuta 

## 対象箇所
左足付け根

## 作業時間
14:00〜17:00

MibuchiYuta, 2021-04-22:
# 左足付け根問題点

## ネジ材について

ネジに付着しているロックタイトの量が規定より多いものが多く見受けられました。また、付着させてはいけないところに使われている箇所が複数あったので以下に示す。
[image]

### スペーサ

アルミのスペーサにロックタイトを使用し、意図しないところで固着しています。
[image]

### 樹脂部分

POMのスペーサに接する部分にもロックタイトが大量に付着しており一部にpomの変形が見られた。
[image]

## 切削精度について

z軸精度の問題で以下のパーツがサーボなどに負荷を加える状態でした。
[image]
[image]

## サーボモータの傷

上記の切削精度の問題やビスの締めすぎ/サイズ違いによりサーボモータのボディ破損が見られました。
[image]

hiroki-0001, 2021-04-22:
4/22のミーティングの議事録です
https://docs.google.com/document/d/1aOQCK8o2MZA2gD-s7GhieVNz_ME0s32JHOgp3uHcP6s/edit?usp=sharing

MibuchiYuta, 2021-04-28:
ボルトの長さについての一番最初の項目と同じ箇所の長さに問題があった。

yasuohayashibara, 2021-04-28:
> ## 切削精度について
> z軸精度の問題で以下のパーツがサーボなどに負荷を加える状態でした。

ネジが長いのでは？
この部分は通常の長さがシビアなところです．

yasuohayashibara, 2021-04-28:
こうやってきっちりと整備していくと，ロボットが最高の状態で動作するようになりますので，非常に良いと思います．
ちなみに，制御回路の整備は行わない？
（別のissueで行っているかもしれませんが．．．）

MibuchiYuta, 2021-04-28:
> > ## 切削精度について
> > z軸精度の問題で以下のパーツがサーボなどに負荷を加える状態でした。
> 
> ネジが長いのでは？
> この部分は通常の長さがシビアなところです．

画像がわかりにくいのですが、ナットが圧をかけてる状態でした

MibuchiYuta, 2021-04-28:
# 状態把握 1日目
メンバー
@MibuchiYuta
@AD58-3104 
@hiroki-0001 
@NaokiIwasawa 

対象箇所
右足付け根
右足膝
左足付け根
左足膝
作業時間
15:00〜18:00

hiroki-0001, 2021-04-28:
# 4月28日　作業報告

## 左ひざ

### [ベアリング(MF85ZZ)]
[image]

上記の写真のように、ベアリングが浮いてしまっている。

### [膝_同軸_スペーサ]
[image]

上記の写真のように、膝メインフレームとの間に隙間が生じてしまっている。

[image]

[image]

また、スペーサ自体が削れてしまっていた。

### [膝メインフレーム]
[image]

[image]

膝メインフレーム間の寸法がCADデータ(正常)だと39.5mmに対し、一番誤差が大きいところで0.5mmも小さくなっていた。

### [SETF5-63_5-M3-N3(スペーサ)]
[image]

[image]

上記の写真の部位のスペーサが本来ならば回るはずだが回らなくなっている。 
上記の問題の原因と思わしきものを以下に記載する

gazou

この部位のベアリングのフランジ部分をPOMのスペーサ(膝_同軸_スペーサ)のはめあい部分にはまりすぎていて、ベアリングを外と内で締め付けすぎている。

### [足首_独立ピッチ_1とリンクメインフレーム6_1]
[image]

足首_独立ピッチ_1とリンクメインフレーム6_1を固定するネジの長さが間違っていた。
正しくは6mmだが、実際には8mmが使用されていた。

### [リンク_ふくらはぎ_サブフレーム_1_2]

[image]

写真のように、ネジ穴がずれていた。

## 共通すること

### [赤ネジ剤について]
[image]

赤ネジ剤つけすぎてベアリング側にも赤ネジ剤が漏れてしまっている。

[image]

また赤ネジ剤の大量使用が影響して、M3の六角ネジがなめってしまっていた。

MibuchiYuta, 2021-04-28:
# 右足付け根/膝問題点

## 付け根

### 再切削が必要なパーツ

[image]

[image]

## 膝

### 側面タップのミスについて

側面タップの穴あけミスが確認された。

[image]

[image]

### 青ネジ剤の固着に関して

青ネジ剤のつけすぎにより固着しビスの山がなくなっていた。

[image]

### パーツの歪みに関して

なんらかの力が加わりpomのスペーサの内部が削れていた

[image]

### ベアリングへのグリスの付着に関して

意図しない箇所にグリスが塗られており、ベアリングないにグリスが塗られてた。

[image]

### ネジの違いについて

定乙である必要もないところになべネジが使用されていた

[image]

### 再切削が必要なパーツ

[image]

[image]

[image]

[image]

[image]

[image]

MasatoKubotera, 2021-04-28:
> こうやってきっちりと整備していくと，ロボットが最高の状態で動作するようになりますので，非常に良いと思います．
> ちなみに，制御回路の整備は行わない？
> （別のissueで行っているかもしれませんが．．．）

制御回路＆コックピットの整備は行わない予定です．
回路については，問題点が多く現状の最新バージョンに合わせるよりも新しい回路に移行する方向で，新コックピットともに作成中です．

MibuchiYuta, 2021-05-07:
# 状態把握 2日目
## メンバー
@MibuchiYuta
@kiyoshirou-kawanabe 
@hiroki-0001 

## 対象箇所
右足首
右足裏
左足首
左足裏

## 作業時間
14:30〜18:30

hiroki-0001, 2021-05-07:
# 5月 7日　作業報告

## 左足首

### 足首メインフレーム_1_1

歪み・はめあい部分にガタがあったため再切削が必要
該当個数は2つ

### 足首サブフレーム_1_3
ナットをはめ込んだ際、ナットが浮き出ている。
該当個数は1つ

### 足首_サブフレーム_1_2
歪みがあったため再切削が必要
該当個数は1つ

### 足首独立ピッチ_1_2
歪みがあったため再切削が必要
該当個数は1つ

### ピッチ軸のB3Mのフリーホーン側
フリーホーンを固定する十字ネジのネジ穴が下の画像のように汚れが詰まっていた。

### 足裏_サブフレーム_1_1
足裏メインフレーム_1_1と足裏_サブフレーム_1_1を固定するネジの長さが長すぎる。

また後加工ミス？を発見した。

## 右足首

### 足首メインフレーム_1_1
歪み・はめあい部分にガタがあったため再切削が必要
該当個数は2つ

### 足首独立ピッチ_1_1
歪みあり再切削が必要
該当個数は2つ

### 足首独立ピッチ_1_2
歪みあり再切削が必要
該当個数は2つ

## 共通して発覚した問題。
写真のように、赤ネジ剤をつけすぎて樹脂部分が溶けてしまっている。またそのためネジの取り外しが非常に困難になっていた。

hiroki-0001, 2021-05-20:
# 5/19　活動報告

### 活動時間
15時~16時半

### [参加メンバー]
責任者：野口
作業者：三渕・川鍋

## 実施した内容

- B3Mの個体番号シールを貼った
- 両足の既製品パーツの個数の確認
- 責任者が変わった(三渕 → 野口)ため引き継ぎ事項の確認

以上の内容を実施しました.  
本日の活動は世界大会で必要となるモータのパラメータの検証を優先したため,実施活動は少なめです.

hiroki-0001, 2021-05-28:
# 2021年5月28日　整備報告
## 再切削リスト

- 以下再切削の必要があるパーツをまとめたものである。
https://docs.google.com/spreadsheets/d/1yOL87ms6DhUJ-kO0graTTdItjGoWPmzekN9uscOrE1g/edit?usp=sharing
＊しかし、既製品に関してはまとまっていない

  
## 右足スラストベアリング
---
以下のような関係で組み立てられていた 

付け根ロール_サブフレーム_1_4  
↓  
スラストベアリング  
↓  
付け根ロール_サブフレーム_1_3  
↓  
T = 0.5mm シム(以下呼称　Aシム)  
↓  
T = 1.0mm シム(以下呼称　Bシム)  
↓  
スラストベアリング  
↓  
T = 0.1mm シム(以下呼称　Cシム)  
↓  
付け根ロール_メインフレーム_1_1  

[image]

以降出てくる上面下面は上記の写真に従ったものとする。
---

### Aシム
[image]
[上面]

[image]
[下面]

### Bシム
[image]
[上面]

[image]
[下面]

### Cシム

[image]
[上面]

[image]
[下面]

### シムについて

付け根ロール_メインフレーム_1_1に一番近いシムには特に汚れや傷が多くついていた、

hiroki-0001, 2021-06-04:
# 2021年6月4日　整備報告

# 足の状態の再確認

## [右足]
### <POM部品>

- **付け根_サブフレーム_1_4**  
  - 摩耗しすぎて高さが変化してしまっている。issueの優先度を上げる必要があり。https://github.com/citbrains/OpenPlatform_ver3_0/issues/12

[image]

## [左足]
### <アルミ部品>
- **足首_独立ピッチ_1_1 × 2**  
  - 歪みあり
- **付け根ロール_サブフレーム_6_5 × 2**
  - 歪みあり
- **付け根ロール_サブフレーム_6_6 × 1**
  - 歪みあり
- **付け根ロール_サブフレーム_6_6 × 1**
  - 歪みあり
- **付け根_フリー軸**
  - 歪みあり
- **リンク_ふくらはぎ_サブフレーム_1_2(予備案)**
  - 歪みあり
- **リンク_ふくらはぎ_サブフレーム_1_4**
  - ネジ穴にねじ止め剤が詰まっているため再利用は不可

### <ガラエポ樹脂部品>
- **リンク_メインフレーム_6_1 × 1**  
  - 歪みあり  

- **リンク_メインフレーム_6_2 × 2**  
  - 歪みあり

- **リンク_メインフレーム_6_3 × 2**  
  - 歪みあり

- **リンク_メインフレーム_6_4 × 2**   
  - 歪みあり

### <POM部品>
- **付け根_サブフレーム_1_4**  
  - 上記に記載した内容と同じ問題が発生した(摩擦によるすり減りで高さが違う)
- **付け根_サブフレーム_6_4**
  - 摩耗によってすり減っている。
- **付け根_サブフレーム_6_2**
  - 摩耗によってすり減っている。
- **付け根_スペーサ_1_1**
  - 摩耗によってすり減っている。
- **付け根_スペーサ_1_2**
  - ネジの締めすぎで形状が変化してしまっている。

# 新たに判明した点

hiroki-0001, 2021-06-07:
# 2021年6月5日　整備報告
  責任者：@MasatoKubotera   
  作業者：@hiroki-0001  
  活動時間　18:00 ~ 19:30
## **右足_スラストベアリング**
- 付け根ロール_サブフレーム_1_4
  - T3の厚みにタップ加工にしている
　[image]
  - SLCG5.5-6を取り付ける際に付け根ロール_サブフレーム_1_4を押さえつける必要があるため、ゆがみやすくなる。
  - SLCG5.5-6の雄ねじ部分を3mmに追加工する際の精度が悪い。
     - 既製品に変える必要あり
    　[image]

## **付け根ロール_サブフレーム_1_3**
パーツの状態は以下の写真のとおりである。

   [image]
[裏側]

   [image]
[表側]

裏側の傷がひどい状況になっていた、この部分はスラストベアリングが接触していた部分である。

## **付け根ロール_メインフレーム_1_1**

[image]
[裏側]

[image]
[表側]

hiroki-0001, 2021-06-10:
# 2021年6月10日　整備報告
    責任者：@hiroki-0001    
    作業者：@MasatoKubotera  
    活動時間　18:00 ~ 

## <胴体>
### 腰回り
- **胴体_メインフレーム_1_1**
  - 既製品を加工したスペーサが4箇所使われている
  - 
    - 内一個はバカネジになっていた
      - 面タップが原因？カレイナットの使用で解決できそう 
  - 歪みが大きい
    - 片足の姿勢に耐えられず、部品が曲がってしまっている
  - サーボを固定するネジがナベネジ
- **胴体_サブフレーム_1_4 & 胴体_サブフレーム_1_4_MIR**
  - スペーサとのはめあいがきつすぎる？分解することができなかった。
- 

# 全体を通して
- **腰ヨー軸のサーボと肩ピッチをつなぐケーブルが宙ぶらりんになっているため、固定する必要あり。**
  - 重要度　中？
- **リアバンパーを固定する方法を変える必要あり。**
  - 現在の取り付け方ではTPUバンパーをねじで潰してしまっている。
  - 重要度　高
- **胴体_メインフレーム_1_1の写真の4点固定の内の1点の固定部分の面積が小さすぎる。**
  - 足から力に耐えきれていないかも？構造解析を用いてどれだけ肉抜きするべきかを調査し、再設計する必要あり。
  - 重要度　高？

hiroki-0001, 2021-06-11:
# 2021年6月10日　整備報告
責任者：@hiroki-0001    
作業者：@MasatoKubotera  
活動時間　17:00 ~ 

## <胴体>
---
### 頭・首周り
#### 胴体_サブフレーム_1_10
- サーボケーブルのコネクタ部分が写真のように干渉するため、ケーブル取り出すには首部分を完全に分解する必要がある。
このままでは手間がかかりすぎるので、改良する必要がある。

- 胴体_サブフレーム_1_9とのはめあい部分がおかしくなっていた。これも肩側からの負荷が原因の可能性が高い  
  **大きな歪みあり**
- 写真の部分の強度が足りないため、修正する必要あり

#### 胴体_サブフレーム_1_11
- 胴体_サブフレーム_1_9とのはめあいが写真のようにおかしくなっていた。これは肩側の負荷が影響しているため、はめあいが曲がったものと考えられる。肩側からの負荷に耐えれるように再設計する必要あり
- バージョン違いにより、サーボの軸に接する部分が凹んでしまっていた。しかし現在のバージョンではその問題は解決されている。  
**大きな歪みあり**
- 写真の部分の強度が足りないため、修正する必要あり

-写真部分がフィレット加工されていない

#### 胴体_メインフレーム_1_13
**歪みあり**  
写真のような力が加わっていたため、コの字型に湾曲していた。

#### 胴体_サブフレーム_1_3
- はめあわせの形状がおかしい(参照)
- コックピットが再設計されたため、飾りネジを使用する部分もそれに応じて変更する必要あり。
**歪みある**
肩後ろ部分が接する部分が特に歪んでいる。

- 使用されていないネジ穴あり。写真要参照。

#### 頭_サーボホーンフレーム
- ネジの締付けトルクが高すぎるため、締結部分が写真のように凹んでしまっている。
---
### 肩まわり
#### 肩_サブフレーム_1_1と肩_メインフレーム_1_1の締結について
- 締結する際に使用するネジの長さが10mmで共通していた。なにか意図がある可能性があり。

#### 胴体_サブフレーム_1_4と胴体_サブフレーム_1_4_MIR
あるべきところにネジがない。。写真参照

---
## 全体を通した際のissue
### 胴体部分のサーボの干渉問題
- 現状の胴体のままだと写真のようにサーボの軸の部分がコックピット部分に干渉する。これにより、コックピット側の設計に制限ができてしまうため、

### 胴体_サブフレーム_1_10と胴体_サブフレーム_1_11の統一
- 寸法が同じでサーボの軸を通す穴部分が異なるだけなので、同じ部品にできないかを検討する。

hiroki-0001, 2021-06-15:
# 2021年6月15日　整備報告
責任者：@hiroki-0001    
作業者：@MasatoKubotera  
活動時間　17:40 ~ 19:

# 腕

## 右腕
ネジの長さの問題なし

## 左腕
ネジの長さの問題なし

hiroki-0001, 2021-06-19:
# 2021年6月19日　整備報告
責任者：@hiroki-0001    
作業者：@MasatoKubotera  
活動時間　14:05 ~ 

# [腕]
## <右腕>
### 腕先
他機体と積層方法が異なる。そのため耐久性や重量が違う。

[167の腕先パーツ]

[165(本機体)の腕先パーツ]

# 全体を通してのissue
- 3Dプリンターの充填率や積層方法の調査が必要そう

hiroki-0001, 2021-06-20:
# 165整備報告 各部位ごと問題点まとめ

# 足_付け根

# 足_膝

# 足_首

# 胴体

# 肩

# 腕

# 頭・首

MasatoKubotera, 2021-08-15:
整備が終了しているため閉じます．

closed 2021-08-15 by MasatoKubotera