SUSTAINA-OP/SUSTAINA-OP #4: GankenKun 整備 全機体共通 報告/記録

MasatoKubotera, 2021-04-22:


MasatoKubotera, 2021-04-27:
[再切削・交換部品リスト](https://docs.google.com/spreadsheets/d/1yOL87ms6DhUJ-kO0graTTdItjGoWPmzekN9uscOrE1g/edit?usp=sharing)(GoogleSpreadSheet)を作成しました．
各機体において再切削・交換が必要な場合はこちらに記載してください．

MasatoKubotera, 2021-04-27:
<h1>交換部品について</h1>

<h2>樹脂スペーサ</h2>

<a href="https://jp.misumi-ec.com/vona2/detail/221006201600/?HissuCode=AMP-340E">AMP-340E</a>と<a href="https://jp.misumi-ec.com/vona2/detail/221006202421/?HissuCode=ASU-340">ASU-340</a>は，面幅が異なりAMP-340E：6.0[mm]・ASU-340：5.5[mm]である．inventorではスペーサーの溝として5.6[mm]で設計されている．

部品種類の削減・面幅が異なるため，AMP-340Eは使用を中止し，ASU-340を使用する．

[image]

図 AMP-340E

[image]

図 ASU-340

MasatoKubotera, 2021-04-28:
<h1>部品交換について2</h1>

<h2>スリ割のないスペーサー</h2>

<h3>ARU-305.5</h3>

下図の長穴減速ホーン(1機体あたり2個)でスリ割のないスペーサー([ARU-305.5](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-305.5&PNSearch=ARU-305.5&KWSearch=ARU-305.5&searchFlow=results2products))が使用さているが，スリ割がないため，固定・外すことが困難である．

[image]

そのため，スリ割付きの[ARU-305.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-305.5S)に交換する必要がある．

<h3>BRU-306.5S</h3>

下図の付け根部分(1機体あたり6個)，足裏のスパイク(1機体あたり24個)において，一部の機体でスリ割がない[BRU-306.5](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5)が使われている．

[image]
[image]

そのため，スリ割付きの[BRU-306.5S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5S)に交換する必要がある．

NaokiIwasawa, 2021-04-28:
162,163,164において頭_サーボホーンフレームの高さが設計と異なるものが取り付けられていました。
設計では8mmでるところが上記の機体では5mmの物が取り付けられています。
[image]
写真1:設計通り8mm
[image]
写真2:間違いの5mm
この違いにより首の高さが設計より3mm下がっています。

MasatoKubotera, 2021-04-28:
<h1>ネジの長さの変更</h1>

<h2><a href="https://jp.misumi-ec.com/vona2/detail/110300208270/?HissuCode=SETF5-63.5-M3-N3&PNSearch=SETF5-63.5-M3-N3&KWSearch=SETF5-63.5-M3-N3&searchFlow=results2products">SETF5-63.5-M3-N3</a></h2>

図1のスペーサー(SETF5-63.5-M3-N3)の雌ネジ長は，図2で6mmとわかるがすべての機体において，図3の状況で8mmネジが使用されていた．

[image]

図1

[image]

図2

[image]

図3

よって，すべての機体で1機体あたり6本・ネジ12か所においてスペーサの交換・ネジ長の変更が必要である．

また，スペーサ長も63.7mm(図4)を使わなければならいところ，63.5mmのものを使用しており設計の改善も必要である．

[image]

図4

MasatoKubotera, 2021-05-05:
<h1>材質の統一</h1>

<h2>リンク_ふくらはぎ_サブフレーム_1_1(予備案)</h2>
<h2>リンク_ふくらはぎ_サブフレーム_1_2(予備案)</h2>

一部の機体において，リンク_ふくらはぎ_サブフレーム_1_1(予備案)，リンク_ふくらはぎ_サブフレーム_1_2(予備案)がa2017製ではなく，POM製が使用されてた．

実験のためPOM製が使われていたが，今後すべての機体においてa2017製を使用する．

※この部品は切削でも作成可能であるが，津田沼キャンパス4号館の工作センターで受託加工することができる．

[image]
[image]

<h3>記録 </h3>

- [ ] 161機
- [ ] 162機
- [ ] 163機
- [ ] 164機
- [ ] 165機
- [ ] 166機
- [ ] 167機

yasuohayashibara, 2021-05-06:
放電ワイヤでカットすると思いますので，素材とDXFファイル（寸法線無）をスタッフに渡してください．
DXFに出力するときにサイズが変化していないかを確認してください．
（単位がmmからcmなどに変化することがあるかもしれませんので，注意ください．）
DXFは新しいと読み込めないかもしれないので，2000ぐらいが良いです．
送ってくれればこちらでもデータをチェックします．

Keita1Mori, 2021-05-06:
# スパイクの連結廃止
## 現状報告
- 163の右足の機体整備をしていた際に足裏のスパイク部分が[BRU-306.5S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5S&PNSearch=BRU-306.5S&KWSearch=BRU-306.5S&searchFlow=results2products)をねじを切って使用していたり連結で使用していました。

- 全体
[image]

- 詳細
[image]

- 使用しているスペーサ
[image]

## 問題点
- これにより固定するために加工したり、電ドリでナットが固定できないため（ビットがない）手動でやったりと色々と手間がかかり世界大会中でのメンテで影響することが考えられる。
## 提案
- このようなことから今使用している[BRU-306.5S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-306.5S&PNSearch=BRU-306.5S&KWSearch=BRU-306.5S&searchFlow=results2products)の連結ではなく[ARU-313S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-313S&PNSearch=ARU-313S&KWSearch=ARU-313S&searchFlow=results2products)のメススペーサで足裏の表側からの六角ねじを使用して固定をしようとおもいます。
[image]

- 上記で示した[ARU-313S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-313S&PNSearch=ARU-313S&KWSearch=ARU-313S&searchFlow=results2products)を使用すると反射してしまうので、レギュレーション違反になってしまう可能性があるので、（今現在使用している連結型も黒いテープを巻いて対応している）よって同じ長さで色の違う[ARL-313E](https://jp.misumi-ec.com/vona2/detail/221006201767/?HissuCode=ARL-313E)を使えるかもしれないです。（しかしこのスペーサを使用した場合ステンレス→アルミに素材が変わるため強度問題とか考えられる。）
[image]

- [ ] 161機
- [ ] 163機
- [ ] 164機
- [ ] 165機
- [ ] 166機
- [ ] 167機

Keita1Mori, 2021-05-07:
# 足首_独立ピッチ_1歪み問題
## 現状報告
- 163,165機体の整備をしていた際に足首_独立ピッチ_1が歪んでいました。
[image]
(例：163機体右足外側の足首_独立ピッチ_1)

## 調査
右脚、左脚、足の内側、外側よって歪みの度合いが違うので後日詳しく調査する。

- [ ] 161機
- [ ] 163機
- [ ] 164機
- [ ] 165機
- [ ] 166機
- [ ] 167機

Keita1Mori, 2021-05-07:
# ネジ剤の使用について
## 問題点
- ロックタイトの付け過ぎにより樹脂パーツ（主にスペーサ）が溶けてしまって取れなくなっています。
[image]
(例：163の右足付け根部分、付け根_サブフレーム_6_2)

## 調査
- 樹脂パーツを貫通するネジのロックタイトですが、これからは樹脂用のロックタイトを使用する。
[ロックタイト 樹脂ねじ用 ねじゆるみ止め接着剤 425](https://jp.misumi-ec.com/vona2/detail/223005097491/)

Keita1Mori, 2021-05-12:
# 報告

> - ネジ剤が溶けて取り外し不可
- このように報告しましたが今日分解したところ圧力がかかっていたため取れていない状態でした。なので一旦樹脂用のネジ剤案を止めたいと思います。

- 解体前
[image]

- 解体後
[image]

- 詳細
[image]

NaokiIwasawa, 2021-05-12:
[image]
[image]

[image]
[image]
[image]
[image]
[image]

NaokiIwasawa, 2021-05-12:
## ナット廃止 ##

現状ナットがネジ固定のために埋め込みで使われているパーツがミラーパーツを含め7種類あるが、メンテナンス性が悪いので廃止したいと考えてる。以下にCADの写真を示す

<h3>胴体_サブフレーム_1_4と胴体_サブフレーム_1_4_MIR</h3>
[image]

<h3>胴体_サブフレーム_1_1と胴体_サブフレーム_1_1_MIR</h3>
[image]

<h3>付け根_メインフレーム_6_1</h3>
[image]

<h3>付け根_フリー軸_1_1</h3>
[image]

<h3>足首_サブフレーム_1_3</h3>
[image]

yasuohayashibara, 2021-05-12:
カレイナットなど埋め込みのナットも検討してみてください．
https://jp.misumi-ec.com/vona2/detail/221000555673/

まずはタッピングが候補となりますが，耐久性ではナットの方が上だと思いますので．

MasatoKubotera, 2021-08-15:
報告については，[
GankenKun163機 整備記録 #2](https://github.com/citbrains/OpenPlatform_ver3_0/issues/2 )[
GankenKun165機 整備記録 #3](https://github.com/citbrains/OpenPlatform_ver3_0/issues/3)で行っていました．

共通記録については，記載せずにissueを立ち上げることで解決しています．
そのため，このissueは不要であるため閉じます．

closed 2021-08-15 by MasatoKubotera