SUSTAINA-OP/SUSTAINA-OP #9: 安定化電源の修理

YukiShigematsu, 2021-05-17:


YukiShigematsu, 2021-05-17:
# 安定化電源の現状報告
## 状態
現在、使用不可の安定化電源は以下の４つである
## １つ目
型番　RWS150B-12
<h4>＜状態＞</h4>
・電圧不安定
・使用されているプラグケーブルの定格電圧が安定化電源の入力電圧以下の為交換の必要あり

[image]
[image]
[image]
<h4>＜使用されていた電源ケーブル(7A,125V)＞<h4>
[image]

## ２つ目
型番　EWS150-12
<h4>＜状態＞</h4>
・プラグケーブルが欠損
・出力用ケーブルの圧着部分が不完全な為、再加工の必要あり

[image]
[image]
<h4>＜出力ケーブル＞</h4>
[image]
<h4>＜使用されていた電源ケーブル＞</h4>
不明

## 3つ目
型番　RWS150B-12
<h4>＜状態＞</h4>
・使用されているプラグケーブルの定格電圧が安定化電源の入力電圧以下の為交換の必要あり

[image]
[image]
[image]
[image]
<h4>＜使用されていた電源ケーブル(7A,125V)＞</h4>
[image]

## ４つ目
型番　RWS150B-12
<h4>＜状態＞</h4>
・電圧不安定
・使用されているプラグケーブルの定格電圧が安定化電源の入力電圧以下の為交換の必要あり

[image]
[image]
[image]
<h4>＜使用されていた電源ケーブル(7A,125V)＞</h4>
[image]

YukiShigematsu, 2021-05-24:
# 修理に使用する電源コードの検討

## 現在、安定化電源修理に使用する電源コードとして以下の三つを検討している。

## 一つ目
URL:
https://jp.misumi-ec.com/vona2/detail/110400376040/?CategorySpec=00000260176%3a%3amig00000002615393%0900000260170%3a%3amig00000002581476

定格10A-250V、CCC規格2芯

この電源コードは定格電圧250Vであるため、国外のほとんどの電源電圧に対応できると思われる。
またプラグの形状は、A形の日本・中国での共通型式であるため、国内のコンセントでも使用可能と考えられる。
この製品はPSE規格ではない。
別途圧着工具などの購入が必要

## ニつ目
https://www.amazon.co.jp/gp/product/B00FIXSEO8/ref=ox_sc_saved_image_3?smid=AN1VRQENFRJN5&psc=1

定格12A-125V

この電源コードは定格電圧が125vであるため、国外では一部使用できない可能性がある。
この製品はすでに圧着が行われているため、別途圧着工具等を購入する必要がない。
しかし、使用されている圧着端子が安定化電源のネジ径に合うかは不明。
この製品がPSE規格であるかは不明。

## 三つ目
https://jp.misumi-ec.com/vona2/detail/110500082740/?KWSearch=AC%e3%82%b3%e3%83%bc%e3%83%89%20%e5%88%87%e3%82%8a%e8%90%bd%e3%81%a8%e3%81%97&searchFlow=results2products&relatedKeyword=1&list=PageSearchResult

定格7A-125V

この電源コードは定格電圧が125vであるため、国外では一部使用できない可能性がある。
この製品はPSE規格である
別途圧着工具などの購入が必要

MasatoKubotera, 2021-05-24:
@YukiShigematsu 
一つ目の電源コードは，リンク先のページに
>（ご注意）電案法未取得の為、国内ではご使用いただけません。

と記載されています．形状的には
>A形の日本・中国での共通型式

であり刺すことはできると思いますが，注意として記載されているので使用しないことが無難かと思います．

YukiShigematsu, 2021-06-01:
# 修理に使用する電源コードの検討

### PSE規格などに関する指摘を受け、新たに安定化電源修理に使用する電源コードの候補として以下の三つを示す。

### 一つ目
https://www.amazon.co.jp/gp/product/B00FIXSEO8/ref=ox_sc_saved_image_3?smid=AN1VRQENFRJN5&psc=1

定格12A-125V

この電源コードは定格電圧が125vであるため、国外では一部使用できない可能性がある。
この製品はすでに圧着が行われているため、別途圧着工具等を購入する必要がない。

### 二つ目
https://www.americandenki.co.jp/products/detail.php?product_id=1004

定格20A-250V

この電源コードは定格電圧が250vであるため、国外のほとんどの電源電圧に対応できると思われる。
この製品はPSE規格である。
この製品はすでに圧着が行われているため、別途圧着工具等を購入する必要がない。
コネクタ形状が引掛形である為変換アダプタが別途必要

YukiShigematsu, 2021-06-02:
# 修理に使用する電源コードの検討

## 一つ前の検討で候補に挙げた電源コードのうち、二つ目のものに決定しようとしたが、最初から付属している圧着端子では安定化電源のねじ径が合わず取り付けができないことが判明した。

#### 安定化電源のねじ径　（4mm）
[image]
#### 検討していた電源コード
穴径 3.5mm

ねじ径が4mmな為、圧着端子の穴径は4.5mmが望ましいと考えられる。
そのため、圧着端子と電源コードを別に購入しねじ径の合う圧着端子を使用する必要があると思われる。

## 新たな電源コード部品構成候補を以下に示す
#### 電源コード
125V-12A　PSE規格
https://jp.misumi-ec.com/vona2/detail/110500085360/
#### 圧着端子
内径 4.3mm　絶縁付
https://jp.misumi-ec.com/vona2/detail/110500058510/?CategorySpec=00000041217%3a%3ab%0900000041218%3a%3aa%0900000041220%3a%3amig00000002460645%0900000041226%3a%3ab&list=PageCategory

YukiShigematsu, 2021-06-12:
# 修理に使用する電源コードの購入
購入予定の物品を以下に示す

## 電源コード
EM2P-E12A-3　125V - 12A PSE規格
https://jp.misumi-ec.com/vona2/detail/110500085360/?HissuCode=EM2P-E12A-3
個数：４個
税込単価：￥705

## 圧着端子
RBA2-4BU　内径 4.3mm 絶縁付
https://jp.misumi-ec.com/vona2/detail/110500058510/?HissuCode=RBA2-4BU
個数：１パック（100個入）
税込単価：￥1082

計：￥3902

また、別途圧着工具の購入が必要。
以下は購入申請が通った場合のみ購入
## 圧着工具
NH-32　端子サイズ 0.3~2 mm^2
https://jp.misumi-ec.com/vona2/detail/110500060950/
個数：１個
税込単価：￥13792

YukiShigematsu, 2021-07-19:
# 修理に使用する電源コードの購入
電源コードの単価が高かった為、購入予定の物品を以下のように変更する

## 電源コード
2M-3　125V - 7A PSE規格
https://jp.misumi-ec.com/vona2/detail/110500082830/?HissuCode=2M-3
個数：４個
税込単価：￥313

計 : ￥1252

YukiShigematsu, 2021-09-03:
### 現状報告
圧着端子と電源コードが到着したため、対面活動が再開次第修理を行います。

YukiShigematsu, 2021-09-20:
# 3つ目の安定化電源の修理完了
## 全体
・圧着端子部とコネクタ部には収縮チューブを使用した
・XT90コネクタとXT60コネクタの両方を取り付けた
→現在の機体にはXT90コネクタは不要なため、ほかの安定化電源のコネクタはXT60を取り付けずに作成を行う
[image]

## 電源ケーブルとXT60/XT90コネクタの作成
・電源ケーブルとコネクタの圧着部分
[image]
・XT90コネクタ部分
[image]
・XT60コネクタ部分
[image]

## 不安定な出力電圧の問題解消
・安定化電源の出力電圧の調整には可変抵抗が使用されていた。
しかし、可変抵抗を回すと電源が落ちる現象が見られた。これは、安定化電源使用時に起こると危険である。
そのため、可変抵抗を1KΩの抵抗器に変更し、抵抗値を固定することで電圧の出力を安定させた。
現在、電圧は12.47Vで出力されている。

yasuohayashibara, 2021-09-21:
固定抵抗への変更ありがとう．
一つだけ今後のために述べておくと，可変抵抗は動かしていないと摺動部分が錆びて正常に動作しなくなることがあります．
そのときは繰り返し正回転，逆回転を繰り返すと正常に動作する【可能性】があります．
今回は電圧を調整する必要がありませんので，これで良いと思います．

YukiShigematsu, 2021-09-21:
# 4つ目の安定化電源の修理完了
## 全体
・圧着端子部とコネクタ部には収縮チューブを使用した
・XT60コネクタのみを取り付けた
[image]

## 電源ケーブルとXT60コネクタの作成
・電源ケーブルとコネクタの圧着部分
[image]
・XT60コネクタ部分
[image]

## 不安定な出力電圧の問題解消
・安定化電源の出力電圧の調整には可変抵抗が使用されていた。
しかし、可変抵抗を回すと電源が落ちる現象が見られた。これは、実機使用時に起こると危険である。
そのため、可変抵抗を1KΩの抵抗器に変更し、抵抗値を固定することで電圧の出力を安定させた。
現在、電圧は12.47Vで出力されている。
[image]
[image]

補足：取り外した可変抵抗は保存している

YukiShigematsu, 2021-09-21:
追記：4つ目の安定化電源のコネクタ側の配線が短かったため、長いものを作成し、取り替えました。

YukiShigematsu, 2021-09-24:
# 4つ目の安定化電源の修理完了
## 全体
・圧着端子部とコネクタ部には収縮チューブを使用した
・XT60コネクタのみを取り付けた
[image]

## 電源ケーブルとXT60コネクタの作成
・電源ケーブルとコネクタの圧着部分
[image]
・XT60コネクタ部分
[image]

## 不安定な出力電圧の問題解消
・安定化電源の出力電圧の調整には可変抵抗が使用されていた。
しかし、可変抵抗を回すと電源が落ちる現象が見られた。これは、実機使用時に起こると危険である。
そのため、可変抵抗を1KΩの抵抗器に変更し、抵抗値を固定することで電圧の出力を安定させた。
現在、電圧は12.57Vで出力されている。
[image]
[image]
補足：取り外した可変抵抗は保存している

YukiShigematsu, 2021-09-24:
# 1つ目の安定化電源の修理完了
## 全体
・圧着端子部とコネクタ部には収縮チューブを使用した
・XT60コネクタのみを取り付けた
[image]

## 電源ケーブルとXT60コネクタの作成
・電源ケーブルとコネクタの圧着部分
[image]
配線の取り付け部分の干渉防止用のプラスチックが折れていたため、安全のためにカバーなどを作成する必要がある
・XT60コネクタ部分
[image]

## 出力電圧
現在、電圧は12.63Vで出力されている。
[image]

closed 2021-09-27 by YukiShigematsu