SUSTAINA-OP/SUSTAINA-OP #10: 足首 独立ピッチの再設計

MasatoKubotera, 2021-05-18:
163機 整備において，a2017切削部品の**足首_独立ピッチ_1_1**と**足首_独立ピッチ_1_2**に歪みが見られた．

<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/2#issuecomment-833414675">GankenKun163機 整備記録 #2 Comment</a>

この歪みは，設計段階で強度が足りていないことが原因と考えられる．また，リンク部分の**足首_スペーサ_1_1**は摩耗しているため，ベアリング内に挿入する部材，設計については見直す必要がある．

よって，上記で記載した３つの部品を再度設計する必要がある．

[image]
[図 足首_独立ピッチ_1_1]

[image]
[図 足首_独立ピッチ_1_2]

[image]
[図 足首_スペーサ_1_1]

MibuchiYuta, 2021-06-02:
## 問題点
- 足首_独立ピッチ_1_1,足首_独立ピッチ_1_2が設計上強度に支障をきたすほどに薄くなっている。そのため歪みがひどい。- 
- ベアリング内にpomが使われており、負荷に耐えられていないことが確認されている。
- 足首_独立ピッチ_1_1はアルミニウムにタップ加工がされており、耐久性が保証されていない。

NaokiIwasawa, 2021-06-02:
参考になる[issue](https://github.com/citbrains/OpenPlatform_ver3_0/issues/24)があるのでリンクを貼り、リンク元のissueはcloseします

NaokiIwasawa, 2021-06-30:
# 再設計 ver1 #
変更点は

1. 足首_独立ピッチ_1_1.iptのポケット加工部分削減
2. 足首_独立ピッチ_1_2.iptの廃止
3. スリ割りスペーサーの追加（[BRU-2604S](https://jp.misumi-ec.com/vona2/detail/221006202476/?ProductCode=BRU-2604S)）

[image]
足首_独立ピッチ_1_1

　溝はスペーサーの逃げ分

---

[image]
スリ割りスペーサー

　スペーサー部分4ｍｍ M2.6 外形Φ5（←ベアリングとのはめ合わせの隙間が０なので心配である）

---

[image]
合わせた状態の断面

　ベアリングの内径の面とスペーサーの上面はそろっている

NaokiIwasawa, 2021-06-30:
今回の設計について注意点を以下に示す

- ミラーを作成していないので左右で筋交いの向きが異なる
- 切削でミスをしないようにM2.6の雌ネジと雄ネジを下穴径（2.2mm）で設計している
- 負荷のかかる部分であるがM2.6に変更したため強度が落ちた可能性がある
- 本issueの”ベアリング内にpomが使われており、負荷に耐えられていないことが確認されている。”については**未対策**

NaokiIwasawa, 2021-06-30:
ここまでの設計変更を先輩に確認してもらったので作業branch"ankle"から"ver4_0"へmergeしました。

NaokiIwasawa, 2021-07-06:
## ねじの頭分の逃げを設計し忘れた

足首_独立ピッチ_1_1とサーボホーンをネジ締結し組み立てた際、ネジの頭部分がリンク_メインフレーム_6_4と干渉するのを発見した。設計ミスがあったので従来通りネジの頭分ポケット加工を施す設計にしようと考える。
[image]
CC-0512-10B分の厚さ(1mm)は確保されている

NaokiIwasawa, 2021-08-04:
# 干渉が見つかった

設計ミスでサーボの出力ホーンと足首_独立ピッチ_1_1・スペーサーの雄ネジが干渉している
[image]

以前はパーツを二つ合わせることで両面に窪みを設けられたが、一つのパーツからなる片面切削では条件を満たす設計が難しい。
外側：サーボホーン固定用ネジの頭分の逃げhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/10#issuecomment-874496377
内側：サーボホーンの段差分の逃げ
[image]

NaokiIwasawa, 2021-09-03:
# 再設計案

これまでの案では問題があったので新たに考えた。変形防止のために肉厚をしたりかかと部分の軸をネジからスペーサに変更した
[image][image]

NaokiIwasawa, 2021-09-15:
# 再設計2
再設計案の後に軸部分の基本構成を他の軸と統一する方針になったので再設計2に至った。
変更点

- 転がり軸受けからオイレスの滑り軸受けに
- 足首_独立ピッチ_1_1の厚さをT5に変更
- 足首_独立ピッチ_1_2は廃止

## 軸部分
[image]← 後方
[image]← 前方

基本構成は他の軸と同じだか中央の隙間が2mmであるため挟むものは少ない

### 用部品

外側から
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円 SUSワッシャー
[LFW-0815](https://jp.misumi-ec.com/vona2/detail/221000090734/?PNSearch=LFW-0815&HissuCode=LFW-0815&searchFlow=suggest2products&Keyword=LFW-0815&list=SuggestPreview) 58円 オイレス
T2のGFRP
[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?PNSearch=80F-0603&HissuCode=80F-0603&searchFlow=suggest2products&Keyword=80F-0603&list=SuggestPreview) 32円 オイレスブッシュ
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円　SUSワッシャー
Ankle_side_pitch（旧名：足首_独立ピッチ_1_1）
**後方**[S3-07-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?PNSearch=S3-07-3W&HissuCode=S3-07-3W&searchFlow=suggesthistory&Keyword=S3-07-3W) 埋め込んだ後が出っ張るカレイナット
**前方**[SF319](https://jp.misumi-ec.com/vona2/detail/221000555673/?PNSearch=SF319&HissuCode=SF319&searchFlow=suggest2products&Keyword=SF319) 埋め込んだ後がフラットなカレイナット
軸分[CU-304.5H](https://jp.misumi-ec.com/vona2/detail/221006204401/?PNSearch=CU-304.5H&HissuCode=CU-304.5H&searchFlow=suggest2products&Keyword=CU-304.5H) 中空スペーサー
固定は外側からM3×10mmを使用

## Ankle_side_pitch（足首_独立ピッチ_1_1）
[image]
切削面がサーボ側に来るように取り付ける。肉抜き部分に筋交いを入れた

MasatoKubotera, 2021-10-03:
この独立ピッチにより，サーボモータ締結ネジにアプローチすることが出来ていないため修正する必要があると考える．
ねじにアプローチできるように凹部を作ることで解決することが出来る．

closed 2021-10-19 by MasatoKubotera