SUSTAINA-OP/SUSTAINA-OP #11: 長穴減速機構の再設計

MasatoKubotera, 2021-05-18:
長穴減速機構は，inventorによる設計と実機で仕様が異なっているため，全機体調査しCAD図を更新する必要がある．

実機では，ブッシュ<a href="https://www.hirosugi-net.co.jp/shop/g/g33405/">D6068-4530</a>が加工して使われているため，使用する既製品の再選定，またこの部品は消耗が激しくメンテナンスのたびに交換されているため設計を見直す必要がある．

[image]
[図 長穴減速機構]

kiyoshirou-kawanabe, 2021-06-06:
MISUMI，ヒロスギネットで
最低条件[d=6, d'=6.8,  t=2.0]のブッシュを見つけられなかったため，条件を満たすワッシャの購入を検討しています．
[image]

ワッシャの条件は上の画像より[d=6, D=14, t=0.5]で探しました．
MISUMIにはゴム製のものしかなかったため，ヒロスギネットで見つけた，ポリスライダーワッシャ型番[LL-6014-05]の使用を検討しています．
https://www.hirosugi-net.co.jp/shop/g/g102269/

条件に合う材質がポリスライダーのみだったのですが使用上の問題はありますか？

MasatoKubotera, 2021-06-06:
D=14のワッシャ類では，問題点である既製品を加工している解決ができないような気がしますが大丈夫でしょうか．
現在(実機)では，上記寸法画像のD=14のものが用いられており，干渉するためニッパーで切断しています．
[image]
[image]

そのため，下図で示す半径6.5[mm]->D=13[mm]以下のものである必要があると思います．
[image]

---
ポリスライダーですが，<a href="https://wilco.jp/docs/technical-data/material-commentary/polyslider.html">ポリスライダーの材質解説</a>を見ると，
>特徴】
面圧によるクリープ変形は殆どありません。
・耐クリープ性が良い
・摩擦係数が小さい
・熱伝導性が良い
・密度が小さく軽い
【用途】
自動車関連部品、小型モーター、家庭電気製品、CDプレーヤー、カメラ、産業機器などの摺動部
【使用法】
ポリスライダーテープはプレスで打ち抜いて使用します。
スラストワッシャーに使用の際は相手金属の摩擦面を滑らかにして下さい。またエッジは面取りをして下さい。
金属の少しのカエリでもポリスライダーのテープ面を引っ掻き急速な摩耗の原因となりますのでご注意下さい。

と記載されています．
>金属の少しのカエリでもポリスライダーのテープ面を引っ掻き急速な摩耗の原因となりますのでご注意下さい。

の部分で，長穴の切削部分と擦れるため急速な摩耗が心配されます．

kiyoshirou-kawanabe, 2021-06-09:
ブッシュはサイズがないため購入を諦め，スペーサとワッシャの組み合わせで代用しようと思います．
スペーサの内径は6，外形は6.8，ワッシャは内径6.8，外形は13以下で探します．

kiyoshirou-kawanabe, 2021-06-18:
https://www.hirosugi-net.co.jp/shop/g/g6183/
スペーサは外形８Φのものが一番小さかったため，再切削で画像の部分を広げて対応しようと考えています．

https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=WSJJ10-8-2&PNSearch=WSJJ10-8-2&KWSearch=WSJJ10-8-2&searchFlow=results2products&list=PageSearchResult
ワッシャは外形１０，厚み２．

kiyoshirou-kawanabe, 2021-06-18:
[
[image]
](url)

kiyoshirou-kawanabe, 2021-06-18:
https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult
内径6Φのスペーサです．これと前に送ったワッシャをくっつけて現在のブッシュ部分にしようと思います．
それにあたり上の画像の赤い部分を8.2㎜に広げることが必要になります．
下の画像のものをイメージしています．

kiyoshirou-kawanabe, 2021-06-18:
[image]

KiyoshiroKawanabe, 2021-06-26:
使用既製品
ワッシャ：https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2
スペーサ：https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult
※寸法違うだけでどちらも同じ商品ですが区別のため名前変えてます．

既製品の追加と寸法の変更を行って，”decelerate”というブランチでpushしました．

MasatoKubotera, 2021-06-29:
中心のスペーサーとひとつ外側のワッシャの組み合わせ的に中間ばめになってしまうため，回転時に回らないことが予想されます．
画像の選択しているスペーサー寸法を見直すべきかと思います．
[image]

他場所の作業を行うためにいったんmaegeします．

KiyoshiroKawanabe, 2021-06-30:
現在，スペーサの寸法が合うもの(内径：6.1以上，外形：８)は金属のものしか見つかっていない．
樹脂のものだと外形10が一番小さい．

MasatoKubotera, 2021-07-01:
金属スペーサと外側の樹脂ワッシャを組み合わせた代替案のスペーサーです．
雌ネジ部分が金属製になっており，外形もφ7のため問題解決になるかもしれません．
<a href="https://www.hirosugi-net.co.jp/shop/c/c11131412/">金属付ジュラコンスペーサー／AMR-300E</a>

yasuohayashibara, 2021-07-02:
議論をきちんと理解できていないので，ちょっと的外れのアドバイスかもしれませんが，このような摺動部に最もよく使用されるのはオイレスになります．
ポリスライダーに比べて，限界PV値が10倍程度あります．
摩耗にも強いですが，１年に１回は交換してあげたほうが良いと思います．
数倍の重量を支えるアダルトサイズの長穴減速にも使用されています．

https://www.oiles.co.jp/bearing/oiles_bearing/product/1078/

KiyoshiroKawanabe, 2021-07-06:
@MasatoKubotera 
現在使われているジュラコンブッシュを確認したところ，内径Φ6のため問題ないかと思います．

> 中心のスペーサーとひとつ外側のワッシャの組み合わせ的に中間ばめになってしまうため，回転時に回らないことが予想されます．

KiyoshiroKawanabe, 2021-08-31:
部品はこちらを使用します。
実機での検証後問題がなければcloseしようと思います。

> 使用既製品ワッシャ
> ：[https](https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2)：//jp.misumi-ec.com/vona2/detail/110300235320/？PNSearch = PACK-WSJJ10-8-2＆HissuCode = PACK-WSJJ10-8-2＆searchFlow = suggest2products＆Keyword = PACK-WSJJ10-8- [2座](https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2)
> ：[https](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult)：[//jp.misumi-ec.com/vona2/detail/110300235320/？](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) HissuCode = PACK-WSJJ8-6-5 [＆](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) PNSearch = PACK-WSJJ8-6-5 [＆](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) KWSearch = PACK-WSJJ8-6-5 [＆](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) searchFlow = results2products [＆](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) list [= PageSearchResult](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult)
> ※配置無だけでありもありありです。
> 
> 既菜の長さと残りの長さをてて、「減速」コートでプッシュしました。

KiyoshiroKawanabe, 2021-08-31:
部品はこちらを使用します。
実機での検証後問題がなければcloseしようと思います。

> 使用既製品
> ワッシャ：https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2 [...]

KiyoshiroKawanabe, 2021-09-03:
統合しました

MasatoKubotera, 2021-09-05:
先日のミーティングで議論されましたが，スペーサの外形・ワッシャの内径上，圧入が出来ないため，下図の青色に選択されたスペーサが外側に抜けてしまう恐れがあります．
[image]

以前に提案されていましたが，

> 議論をきちんと理解できていないので，ちょっと的外れのアドバイスかもしれませんが，このような摺動部に最もよく使用されるのはオイレスになります．
> ポリスライダーに比べて，限界PV値が10倍程度あります．
> 摩耗にも強いですが，１年に１回は交換してあげたほうが良いと思います．
> 数倍の重量を支えるアダルトサイズの長穴減速にも使用されています．
> 
> https://www.oiles.co.jp/bearing/oiles_bearing/product/1078/

スペーサ・ワッシャを統合した，オイレスブッシュに変更した方が良いかと思います．
候補のブッシュです．
[80F-0605](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0605)
フランジ厚が1mmであるため，内径φ6程度，外形φ12のワッシャが別途必要になります．

MasatoKubotera, 2021-09-06:
Inventorで可視化してみました．
[image]

オイレスブッシュ：[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)
ジュラコンワッシャ：[CC-0612-10B](https://jp.misumi-ec.com/vona2/detail/221006204759/?PNSearch=CC-0612-10B&HissuCode=CC-0612-10B)

スリ割り丸形スペーサについては，現状[ARU-305.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-305.5S)を使っていますが下図の通り長さLは5.0mmの[ARU-305S](https://jp.misumi-ec.com/vona2/detail/221006202465/?PNSearch=ARU-305S&HissuCode=ARU-305S)で十分そうです．
[image]

MasatoKubotera, 2021-09-06:
上記コメントで指摘した修正をver4_0ブランチに反映しました．
[image]

変更点は下記の通りです．
廃止
- [PACK-WSJJ8-6-5](https://jp.misumi-ec.com/vona2/detail/110300235320/?HissuCode=PACK-WSJJ8-6-5&PNSearch=PACK-WSJJ8-6-5&KWSearch=PACK-WSJJ8-6-5&searchFlow=results2products&list=PageSearchResult) ×2個
- [PACK-WSJJ10-8-2](https://jp.misumi-ec.com/vona2/detail/110300235320/?PNSearch=PACK-WSJJ10-8-2&HissuCode=PACK-WSJJ10-8-2&searchFlow=suggest2products&Keyword=PACK-WSJJ10-8-2)×2個
- スペーサ：[ARU-305.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-305.5S)×2個

新規
- オイレスブッシュ：[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)×2個
- ジュラコンワッシャ：[CC-0612-10B](https://jp.misumi-ec.com/vona2/detail/221006204759/?PNSearch=CC-0612-10B&HissuCode=CC-0612-10B)×2個
- スペーサ：[ARU-305S](https://jp.misumi-ec.com/vona2/detail/221006202465/?PNSearch=ARU-305S&HissuCode=ARU-305S)×2個

MasatoKubotera, 2021-09-06:
再設計が完了したのでissueを閉じます．

yasuohayashibara, 2021-09-15:
指摘が遅くなり申し訳ありませんが，歩行の安定性に関することですので，一応懸念事項を伝えておきます．
部分的な調整になるので，製作後の状態を見て判断していただくので構いません．

できるだけ色々なところで述べるようにしていますが，股ロール関節の剛性をどれだけ高められるかが歩行の安定性に強く寄与します．
GankenKunは当初は人工芝上での安定した歩行が難しかったのですが，股ロール関節の剛性を上げることで，試合中に転倒することなく歩行できるようになりました．

その点で，OBの中島君が試行錯誤して，最終的に長穴を前後から（滑る）樹脂ワッシャで挟むような機構にすることが一番良いという結論になっていたと記憶しています．（曖昧ですが．．．）
もしくはたわまないように，間に何か入れて『強く押し付ける』という方式に変更されたかもしれません．
とにかくたわむ要素が残っていると，歩行が安定しないおそれがあります．
issueを探してもその議論が無いのですが，この部分はキーとなりますので，念のため伝えておきます．

MasatoKubotera, 2021-09-15:
@yasuohayashibara 
>OBの中島君が試行錯誤して，最終的に長穴を前後から（滑る）樹脂ワッシャで挟むような機構にすることが一番良いという結論になっていたと記憶しています．（曖昧ですが．．．）
もしくはたわまないように，間に何か入れて『強く押し付ける』という方式に変更されたかもしれません

現状GankenKunの実機，過去のCADにおいてもこのような機構が搭載されていないため確認できませんでした．

指摘を受け，長穴減速機構部分がたわまないように上記コメントで提案した構成からさらに改良を加えました．
オイレスのブッシュ・ワッシャで長穴部分を両側から押し付けるような構造にしています．

使用部品については下記のようになります．（サーボ側から）
- カレイナット [S3-07-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?HissuCode=S3-07-3W) 
- Slider_crank_horn（A2017 T2 切削）※ver3.0まではGFRP．カレイナットを圧入するためにA2017に変更．
- ワッシャ [UUW-0312-10BH](https://www.hirosugi-net.co.jp/shop/g/g59626/)
- オイレスブッシュ [80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?HissuCode=80F-0603)
- 中空スペーサ [CU-304.5H](https://www.hirosugi-net.co.jp/shop/g/g9069/)
- オイレスワッシャ [LFW-0815](https://jp.misumi-ec.com/vona2/detail/221000090734/?HissuCode=LFW-0815)
- ワッシャ [UUW-0312-10BH](https://www.hirosugi-net.co.jp/shop/g/g59626/)

[image]「長穴減速機構」
[image]「断面図」

MasatoKubotera, 2021-09-17:
ミーティングでの共有が終了したのでcloseします．

MasatoKubotera, 2021-10-17:
## 長穴減速部の加重一点集中への対策

米田先生との対談により判明した問題で、減速機構内のブッシュが円形であるため、付け根ロール軸稼働時に荷重が側面の一点に集中する問問題があった。

解決のために、[脚軸のオイレス 部品の導入 #96](https://github.com/citbrains/OpenPlatform_ver3_0/issues/96)で出たルーテックE ブッシュ（LED）を用いる。

このときの注意として、
>『許容最高PV値』が低め
ということが提言されていたが、ここでのPV値とは

## はめあい部を外さずに分解できる仕様への変更

[image]
[image]
[image]
[image]
[image]