SUSTAINA-OP/SUSTAINA-OP #12: 脚ロール軸 ベアリング回りの再設計

MasatoKubotera, 2021-05-18:


MasatoKubotera, 2021-06-02:
## 追加事案

下図の赤丸で囲まれたベアリング軸を締結するために，機体で唯一M3×9mmが用いられている．
購入物品削減のために，M3×8mm，M3×10mmで代用できるような設計にするべきである．

[image]

NaokiIwasawa, 2021-09-03:
# ネジの長さ対策

現在使用しているネジは10mmのため、1mmの座金を入れて対応する。使用するのは[UUW-0306-10H](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0306-10H&HissuCode=UUW-0306-10H&searchFlow=suggest2products&Keyword=UUW-0306-10H&list=SuggestPreview)で内径3.2外径6mmである。これは現在の状態で最も簡単な対応だが、ベアリング内に入れるPOMの摩耗対策にはなっていないので今後も検討する。この部分はネジが軸として機能するため再設計の際はスペーサなどで代用する。
[image]

yasuohayashibara, 2021-09-15:
見直してみて，気づいたので指摘が遅くなりましたが，懸念事項を伝えます．
次回の設計のときに参考にしていただくので構いません．

１）ワッシャを取り付けずに組み立てる人がいそうな気がします．
　　経験から言うと，無理やり締め込んで大変なことになるか，ガタがある状態で完成とするおそれがあります．
　　入れていないと違和感を感じるような仕組みがあると良いと思います．
　　特に『ここだけ』この構造を採用するとしたら，忘れるおそれが高いと思います．

２）ver.1の設計がそうなっていたと記憶していますが，大きめのオイレスワッシャで全体を抑え込んだほうが良いかもしれません．
　　ベアリングの球が外れる現象が発生していましたが，衝撃力によりそれと同じことが生じるおそれがあります．
　　どの程度スラスト方向に力がかかるかで判断する必要があると思いますが，一応伝えておきます．

NaokiIwasawa, 2021-09-15:
# 再設計の報告

足部の大改良で更に変えたため最新ではありませんでした。報告遅れて申し訳ありません

この部分に限らずなのですが現段階で脚部にあった転がり軸受けは付け根のスラストベアリングを除き全て**滑り軸受け**に変更しました。
一度にすべて変えるのは適切ではないかもしれませんが、オイレスのブッシュとワッシャで駆動部を挟み込む方針で全て設計し直しました。理由は滑り軸受けを知り興味を持ったのと部品点数や場所による構造の違いを出来るだけなくしたかったからです。
固定方法は両サイドからM3×6で締結
アルミと軸スペーサの間にSUSのワッシャを挟み変形を抑える意図があります。

[image]
画像が最新の状態です。(21/09/15)

## 使用部品
手前から
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円
[LFW-0815](https://jp.misumi-ec.com/vona2/detail/221000090734/?PNSearch=LFW-0815&HissuCode=LFW-0815&searchFlow=suggest2products&Keyword=LFW-0815&list=SuggestPreview) 58円 オイレス
T2のアルミ
[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?PNSearch=80F-0603&HissuCode=80F-0603&searchFlow=suggest2products&Keyword=80F-0603&list=SuggestPreview) 32円 オイレス
[CC-0612-20B](https://hirosugi.co.jp/products/POM/CC-B.html) 11円
[CC-0612-25B](https://hirosugi.co.jp/products/POM/CC-B.html) 11円
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円
軸のスペーサ:[ARU-309S](https://jp.misumi-ec.com/vona2/detail/221006202465/?PNSearch=ARU-309S&HissuCode=ARU-309S&searchFlow=suggest2products&Keyword=ARU-309S&list=SuggestPreview) 107円

MasatoKubotera, 2021-10-03:
再設計が終了したためcloseする．

MasatoKubotera, 2021-10-03:
関連性が高いためここに記述する．

MasatoKubotera, 2021-11-14:
再告知が終了したためcloseします．

closed 2021-11-14 by MasatoKubotera