SUSTAINA-OP/SUSTAINA-OP #14: 低頭 六角穴付きボルト使用箇所の調査

MasatoKubotera, 2021-05-18:
現在の機体では大きく分けて 六角穴付きボルト，低頭 六角穴付きボルト，十字なべねじの３種類が使用されている．
十字なべねじについては低頭 六角穴付きボルトに統一する．

そのために，低頭 六角穴付きボルト，十字なべねじの使用箇所ごとにねじの呼び径M，長さL[mm]，頭部厚[mm]を調査する必要がある．

Keita1Mori, 2021-05-24:
このissue自分が担当します.

Keita1Mori, 2021-06-09:
# 低頭　六角穴付きボルト（個数、使用箇所）
## M3×7 頭部厚2mm
[image]

### 使用箇所
- 胴体（両肩）付け根ロール_サブフレーム_1 ×4
- 胴体（両足付け根） 付け根ロール_サブフレーム_1 ×4
---
## M3×10 頭部厚2mm
[image]

### 使用箇所
- 足（両足減速機構部）長穴減速_ホーン_2とARU-305_5固定 ×１
---
## その他の調査
- 今週はここまでしかできなかったので他の箇所の調査を後日行う

DaaFront, 2021-06-29:
# 低頭　六角穴付きボルト(個数、使用箇所)

## M2.6×10 頭部圧2mm

<img src="https://user-images.githubusercontent.com/83112603/123766672-da900b80-d901-11eb-8b5e-a59b8b7fe081.jpeg"
width="320px">

### 使用箇所
- 胴体(両肩付け根)　胴体メインフレーム_1_12 と 胴体_サブフレーム_1_4 ×4
<img src="https://user-images.githubusercontent.com/83112603/123769100-e41a7300-d903-11eb-8623-e41cbf636738.png"
width="320px">

DaaFront, 2021-07-08:
# 十字なべねじ(個数、使用箇所)

## M2.6×7.5 頭部圧1.5

<img src="https://user-images.githubusercontent.com/83112603/124889140-b626e400-e011-11eb-96ca-8c0e7083e251.jpeg"
width="320px">

### 使用箇所
- 脚部(長穴減速機構)　サーボホーン と 長穴減速_ホーン_2 ×4
<img src="https://user-images.githubusercontent.com/83112603/124891259-c63fc300-e013-11eb-8740-e333dced8f2c.jpg"
width="320px">
---
# 十字なべねじ(個数、使用箇所)

## M2.6×16.6 頭部圧1.8

<img src="https://user-images.githubusercontent.com/83112603/124890474-094d6680-e013-11eb-81b8-1913bbf1b81a.jpeg"
width="320px">

### 使用箇所
- 脚部(長穴減速機構)　サーボモータ本体 と 付け根_メインフレーム_6_1 ×2
<img src="https://user-images.githubusercontent.com/83112603/124891419-eff8ea00-e013-11eb-9ef4-8640261a1411.jpg"
width="320px">

DaaFront, 2021-07-08:
# 十字なべねじ(個数、使用箇所)

## M2.6×7.4 頭部圧1.4

[image]

### 使用箇所
- 足首　サーボホーン と 足首_独立ピッチ_1 ×8
<img src="https://user-images.githubusercontent.com/83112603/124897140-fa69b280-e018-11eb-9adf-81d1639f45de.png"
width="320px">

MasatoKubotera, 2021-07-08:
@DaaFront @Keita1Mori 

報告された写真を見ると，ねじの頭部を埋め込むための凹加工がされているようです．
**頭部厚だけでなく，頭部径**についてもinventorなどで**最大〇〇[mm]未満のねじでないと埋め込めない/干渉する**か調査する必要がありそうです．

DaaFront, 2021-07-18:
# 十字なべねじ(追加報告)

## M2.6×7.4 頭部圧1.4

### 使用箇所
- 足首　サーボホーン と 足首_独立ピッチ_1 ×8

inventorにて調べたところ、以下の通りだったので、
**頭部径5mm未満**であれば、干渉せずに使用することが可能であると思われる。

[image]

DaaFront, 2021-07-27:
# 十字なべねじ(個数、使用箇所)

## M2.6×17.4 頭部圧1.9mm

<img src="https://user-images.githubusercontent.com/83112603/127095324-fe2ee42e-d814-4385-9c5b-4dd0e3443a62.jpeg"
width="320px">

### 使用箇所
- 胴体(両肩付け根)　サーボモータ と 胴体_サブフレーム_1_4 ×4
<img src="https://user-images.githubusercontent.com/83112603/127095283-3e7c10a2-6910-4266-bffd-55eaaf9fd3a5.jpeg"
width="320px">

このねじの使用箇所に埋め込み加工はありませんでした。

DaaFront, 2021-07-30:
# 十字なべねじ(個数、使用箇所)

## M2.6×17 頭部圧2mm

<img src="https://user-images.githubusercontent.com/83112603/127610776-0bc1e566-2e87-4e24-9e86-21264216f7cf.jpeg"
width="320px">

### 使用箇所
- 胴体(両肩付け根)　サーボモータ と 胴体_メインフレーム_1_1 ×4
<img src="https://user-images.githubusercontent.com/83112603/127611395-4bc76714-0d11-4ea7-a58d-68f758442ad3.jpeg"
width="320px">

このねじの使用箇所に埋め込み加工はありませんでした。

MasatoKubotera, 2021-08-15:
このissueを参考にネジ購入におけるねじの選定を行いました．
調査が終了し，購入するネジが決定したためこのissueは閉じます．

closed 2021-08-15 by MasatoKubotera