SUSTAINA-OP/SUSTAINA-OP #16: 埋め込みナットの見直し

MasatoKubotera, 2021-05-18:
<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/4#issuecomment-839915933">GankenKun 整備 全機体共通 報告/記録 #4 Comment</a>で提案されたナット廃止だが，

>カレイナットなど埋め込みのナットも検討してみてください．
https://jp.misumi-ec.com/vona2/detail/221000555673/
まずはタッピングが候補となりますが，耐久性ではナットの方が上だと思いますので．

とコメントをいただいた．

埋め込みナットを廃止しタップ加工に変更するか，カレイナットを使用するべきか調査検証を行う．

MasatoKubotera, 2021-05-24:
<h1>ナット数</h1>
<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/4#issuecomment-839915933">GankenKun 整備 全機体共通 報告/記録 #4 Comment</a>で記載されている通り，1機体あたり8種類の部品においてナットを埋め込むための窪みがある．(下記パーツ名を記す．)

1. 胴体_サブフレーム_1_4
2. 胴体_サブフレーム_1_4_MIR
3. 胴体_サブフレーム_1_1
4. 胴体_サブフレーム_1_1_MIR
5. 付け根_メインフレーム_6_1
6. 付け根_フリー軸_1_1
7. 足首_サブフレーム_1_3
8. 腕先

上記８種類の部品で一機体あたり**計18個**のナットが使用されている．

MasatoKubotera, 2021-05-24:
<h1>3Dプリント部品の対応</h1>

３Dプリント部品である**腕先**は埋め込みナットではなく，インサートナット(例：<a href="https://jp.misumi-ec.com/vona2/detail/221005351346/?HissuCode=SP-BRM3-5.3">SP-BRM3-5.3</a>)を熱圧入で固定することで，メンテナンスを上げることが出来るのではないだろうか．

MasatoKubotera, 2021-05-24:
<h1>現状のタップ加工である場所の対応</h1>

アルミ切削部品での埋め込みナットは，全8か所から腕先の3Dプリント部品を除いた7種類である．
しかし，現状ではすでに側面ではない面にタップ加工を行いネジを締結している箇所がある．

この埋め込みナットの見直しを行うためには，上記の箇所を調査する必要があると考えられる．

MasatoKubotera, 2021-08-15:
調査のissue [側面ではない面にタップ加工されている箇所の調査 #20](https://github.com/citbrains/OpenPlatform_ver3_0/issues/20)が終了していました．
そのため，このissueを再開することができます．

MiyawakiKaito, 2021-08-30:
このissueはタップ加工したところにカレイナットを組み込むかどうかを査定するという方針で進めてもよろしいでしょうか。
また、カレイナットを組み込む必要がないところはどのように扱えばよいかも教えていただけるとありがたいです。

MiyawakiKaito, 2021-09-03:
カレイナット　https://jp.misumi-ec.com/vona2/detail/221000555662/
今回使おうと想定しているカレイナットの型番	S3-05-3W

## 胴体_サブフレーム_1_4
胴体_サブフレーム_1_12との接続部分なので多少の強度は必要だと考える。そのためカレイナット(S3-05-3W)の使用を推奨する。

[image][image]

## 胴体_サブフレーム_1_4_MIR
胴体_サブフレーム_1_4と同じ理由により、カレイナット(S3-05-3W)の使用を推奨する。

[image][image]

## 胴体_サブフレーム_1_1
RearBumper_Lowerの再加工は必要になるが、強度が必要な場所ではないのでタップ加工でも可。カレイナットを使う場合はS3-05-3Wの使用を推奨する。

[image][image]

## 胴体_サブフレーム_1_1_MIR
 胴体_サブフレーム_1_1と同じく、タップ加工とカレイナット(S3-05-3W)のどちらでも問題はないと考える。

[image][image]

## 付け根_メインフレーム_6_1
足の部分は強度が必要になってくるので、カレイナット(S3-05-3W)の使用を推奨する。

[image][image]

## 付け根_フリー軸_1_1
付け根_メインフレーム_6_1と同じく、足の部分は強度が必要になってくるので、カレイナット(S3-05-3W)の使用を推奨する。

[image][image]

## 足首_サブフレーム_1_3
[image][image]

MiyawakiKaito, 2021-09-03:
今回のミーティングで選ぶカレイナットを間違っていたことが判明したので、もう一度ナットを選び直す。また、従来の穴はあまり気にしなくていいとのことだったので、従来のパーツも再調整する。

MasatoKubotera, 2021-09-05:
@MiyawakiKaito 
カレイナットについてはす上記の部分については使用する場合はすべて共通のものを使用し，既製品の種類を抑えたいと考えています．
使用するカレイナットについては，厚み3mm部材に収まり，部材厚みT3.0≠カレイナット高さHでないものを選定したいと考えています．
ポケット加工するときにZ方向の高さ公差が+になっている場合，隣接する部品に接触する恐れがあるためです．
かつ，ナット高さHはできるだけ高く取りたいので，3-07シリーズ（H=2.8）で選定をお願いします．

MiyawakiKaito, 2021-09-07:
カレイナット再選定をした結果、https://jp.misumi-ec.com/vona2/detail/221000555662/   のカレイナットがやはり一番条件に合うので、久保寺先輩が指定されたS3-07を用いて、現在下穴を直径4.5㎜、ナット頭部とナットねじ部の穴の深さをそれぞれ2.2㎜、0.8㎜になるよう再加工しています。

MasatoKubotera, 2021-09-07:
@MiyawakiKaito 
カレイナットの選定ありがとうございます．
[S3-07-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?PNSearch=S3-07-3W&HissuCode=S3-07-3W&searchFlow=suggest2products&Keyword=S3-07-3W&isReSearch=1&list=SuggestPrev)を使う方針でよいと思います．
加工方法についても，
> 現在下穴を直径4.5㎜、ナット頭部とナットねじ部の穴の深さをそれぞれ2.2㎜、0.8㎜になるよう再加工しています。

で問題ではないかと思います．

調査段階から変更がありましたのでお知らせします．
- 付け根_メインフレーム_6_1
この部分にナットを使うことがなくなりましたので，修正作業は行わなくて大丈夫です．
理由として，先日この部分にさしているスペーサが既製品追加工であることが発覚し，加工をしようとしないスペーサへの変更が難しいためです．
参照：[既製品の追加工調査/改善 #28](https://github.com/citbrains/OpenPlatform_ver3_0/issues/28)

- 付け根_フリー軸_1_1
廃止されました．

- 足首_サブフレーム_1_3
足首ユニットの変更が行われており，競合するため編集する必要がありません．

MasatoKubotera, 2021-09-09:
@MiyawakiKaito 
Googleドライブで共有された，胴体_サブフレーム_1_4，胴体_サブフレーム_1_1についてver4_0に反映しました．
MIRの自動更新(MIR派生)が機能していなかったため，こちら側で修正しました．
埋め込みナットについても，アセンブリで拘束し実装しています．

[image]「胴体_サブフレーム_1_4の埋め込みナット実装」

[image]「胴体_サブフレーム_1_1の埋め込みナット実装」

これにて作業は終了となります．お疲れさまでした．確認でき次第closeして下ください．

MiyawakiKaito, 2021-09-09:
修正ありがとうございます。無事実装できたようで良かったです。ではこのissueを閉じさせていただきます。

closed 2021-09-09 by MiyawakiKaito