SUSTAINA-OP/SUSTAINA-OP #17: はめ合い寸法の調査

MasatoKubotera, 2021-05-18:


hiroki-0001, 2021-06-02:
はめあいの余白部分が制作した人によって若干異なるので、それらを統一させる必要がある。
そのためCAD上で、はめあいの余白を調査する。

NaokiIwasawa, 2021-06-07:
[image]

# はめ合わせの隙間確認　

足や胴体を基本として調査した（首・手は未調査）

## 足首_1
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 足首_メインフレーム_1_1 | 足首_サブフレーム_1_3 | 2.5 | 1 2 |  |
| 足首_メインフレーム_1_1 | 足首_サブフレーム_1_2 | 0.501 | 1 3 |  |
| 足首_メインフレーム_1_1 | 足首_サブフレーム_1_1 | 0.501 | 1 4 |  |
| 足首_メインフレーム_1_1_MIR | 足首_サブフレーム_1_3 | 2.5 | 5 2 |  |
| 足首_メインフレーム_1_1_MIR | 足首_サブフレーム_1_2 | 0.501 | 5 3 |  |
| 足首_メインフレーム_1_1_MIR | 足首_サブフレーム_1_1 | 0.501 | 5 4 |  |

---
## 付け根_1
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 付け根_メインフレーム_2_1 | 付け根_サブフレーム_1_3 | 0.025 | 1 2 |  |
| 付け根_メインフレーム_2_1 | 付け根_サブフレーム_1_2 | 0.025 | 1 3 |  |
| 付け根_メインフレーム_2_1 | 付け根_メインフレーム_6_1 | 1.0 | 1 4 |  |
| 付け根_メインフレーム_2_1_MIR | 付け根_サブフレーム_1_3 | 0.025 | 5 2 |  |
| 付け根_メインフレーム_2_1_MIR | 付け根_サブフレーム_1_2 | 0.025 | 5 3 |  |
| 付け根_メインフレーム_2_1_MIR | 付け根_メインフレーム_6_1 | 1.0 | 5 4 |  |

---
## 付け根ロール
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 付け根ロール_メインフレーム_1_1 | 付け根ロール_サブフレーム_6_6 | 0.05 | 1 2 | [isuue](https://github.com/citbrains/OpenPlatform_ver3_0/issues/1#issuecomment-813027144)にて変更したはずだったが変わってなかった |
| 付け根ロール_メインフレーム_1_1 | 付け根ロール_サブフレーム_6_5 | 0.05 | 1 3 | [isuue](https://github.com/citbrains/OpenPlatform_ver3_0/issues/1#issuecomment-813027144)にて変更したはずだったが変わってなかった |

---
## 胴体_1下部
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_メインフレーム_1_1 | 胴体_サブフレーム_1_1_MIR | 0.1 0.05 | 1 2 | 長辺側0.1　短編側0.05　**上下2か所とも同様の関係** |
| 胴体_メインフレーム_1_1 | 胴体_サブフレーム_1_1 | 0.1 0.05 | 1 3 | 長辺側0.1　短編側0.05　**上下2か所とも同様の関係** |

---
## 胴体_1中部
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_11 | 胴体_メインフレーム_1_13 | 0.05 | 1 2 |  |
| 胴体_サブフレーム_1_11 | 胴体_サブフレーム_1_9 | 0.1 | 1 3 |  |
| 胴体_サブフレーム_1_11 | 胴体_サブフレーム_1_9 | 0.1 | 1 4 |  |
| 胴体_サブフレーム_1_11 | 胴体_メインフレーム_1_13 | 0.05 | 1 5 |  |

---
## 胴体_1上部
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_11 | 胴体_メインフレーム_1_13 | 0.05 | 1 2 |  |
| 胴体_サブフレーム_1_11 | 胴体_サブフレーム_1_9 | 0.1 | 1 3 |  |
| 胴体_サブフレーム_1_11 | 胴体_サブフレーム_1_9 | 0.1 | 1 4 |  |
| 胴体_サブフレーム_1_11 | 胴体_メインフレーム_1_13 | 0.05 | 1 5 |  |

---
## 胴体_1横
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_3 | 胴体_メインフレーム_1_12 | 0.05 | 1 2 | f |
| 胴体_サブフレーム_1_3 | 胴体_メインフレーム_1_12 | 0.05 | 1 3 | f |
| 胴体_サブフレーム_1_4 | 胴体_メインフレーム_1_12 | 0.05 | 4 2 | f |
| 胴体_サブフレーム_1_4 | 胴体_メインフレーム_1_12 | 0.05 | 4 3 | f |

### 下記の部分において穴の位置がずれているのを発見した

[image]

**胴体_サブフレーム_1_3の上側と下側の両方**

NaokiIwasawa, 2021-06-10:
# はめ合わせの隙間確認追加　　

胴体の部分にはめ合わせの報告漏れがあったので追加する

## 胴体_1
[image]

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_1_MIR | ALSFB6-66 | 0.005 | 1 2 |  |
| 胴体_サブフレーム_1_1_MIR | 胴体_メインフレーム_1_2 | 0.05 | 1 3 |  |
| 胴体_サブフレーム_1_1 | ALSFB6-66 | 0.005 | 4 2 |  |
| 胴体_サブフレーム_1_1 | 胴体_メインフレーム_1_2 | 0.05 | 4 3 |  |

NaokiIwasawa, 2021-06-17:
脚のはめ合わせの統一を行ったので以下に報告する。
足首_1・付け根_1・付け根ロールにおいて片側のはめ合わせ隙間を0.01にしてCAD修正を行った。
branchはfitting_holes

合わせて胴体の修正も行おうとしたが、エラーが発生し自身では対応できそうに
なかったので @MasatoKubotera 先輩に引継ぎをした。

NaokiIwasawa, 2021-06-17:
# はめ合わせの隙間確認追加2　　

胴体の部分にはめ合わせの報告漏れがあったので追加する

## 胴体_1（下面）
[image]
追加で示す部分に関しては隙間が0.00mmで変更後の反対側の隙間を考慮すると片側のみ0.01ｍｍできつそうである。

|partA|partB|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_1 | 胴体_メインフレーム_1_1 | 0.00 | 1 2 | 丸でマークアップ |
| 胴体_サブフレーム_1_1_MIR:1 | 胴体_メインフレーム_1_1 | 0.00 | 2 3 | 丸でマークアップ |

---
## 胴体_1（側面）
[image]
[image]
本部分は少し特殊ではめ合わせの片側がガラエポ、もう片方がサーボモータ、凸部分がPOMとなっている。そしてサーボモータに関しては胴体_サブフレーム_1_3の板厚に比べ半分以下程度しか挿入されていない。

|partA|partB|partC|隙間[mm]|写真中の番号| 備考 |
| :-: | :-: | :-: | :-: | :-: | :-: |
| 胴体_サブフレーム_1_3 | B3M11xx | 胴体_メインフレーム_1_13 | 0.00 0.05 | 1 2 3 | 2 3側が0.00 |
| 胴体_サブフレーム_1_3 | B3M11xx | 胴体_メインフレーム_1_13 | 0.00 0.05 | 3 4 5 | 3 4側が0.00 |

MasatoKubotera, 2021-06-17:
> 合わせて胴体の修正も行おうとしたが、エラーが発生し自身では対応できそうに
> なかったので @MasatoKubotera 先輩に引継ぎをした。

上記の内容を修正した．

また，**はめ合わせの隙間確認追加2**でしている箇所も修正した．
- 胴体_1（下面）では，胴体_サブフレーム_1_1，MIRを修正．
このiptのスケッチが複雑で寸法値を変えただけでは変更できなかったため，面からスケッチを新規作成し対応した．

- 胴体_1（側面）では，胴体_サブフレーム_1_3を修正．

@NaokiIwasawa に引き続きブランチは，fitting_holesで作業し更新した．

MasatoKubotera, 2021-06-19:
調査・修正が終了し，ver4_0ブランチにmargeしたのでcloseします．

closed 2021-06-19 by MasatoKubotera