SUSTAINA-OP/SUSTAINA-OP #18: ケーブルを指定の長さで製作していただけないかを打診する

yasuohayashibara, 2021-05-21:
ケーブルを指定の長さで製作していただけないかを打診する．

yasuohayashibara, 2021-07-03:
こちら遅くなりましたが，近藤科学に打診しました．

yasuohayashibara, 2021-07-06:
検討してくれるとのことですので，以下の見積もりを依頼しました．

１セットの本数
１）　８５ｍｍ　６本
２）１３０ｍｍ　３本
３）２００ｍｍ　５本
４）２４０ｍｍ　５本

yasuohayashibara, 2021-07-06:
１本１５００円だそうです．
ちなみに，今のケーブルを追加で購入すると１本１０００円とのことですので，だいぶ配慮はしてくれていると思います．

一応５セットで見積もりをとってみましたが，この本数では外注できないので，近藤科学内で手作業で作成するとのことでした．
サーボの購入に合わせて５００円の追加で受けてくれるとしたら，リーゾナブルかもしれません．

MasatoKubotera, 2021-09-18:
製品を受け取りました．
今後，USB-RS485変換基板に問題がない機体で検証したいと考えています．

[image]

NaokiIwasawa, 2021-12-27:
# 実装を行うにあたって
近藤科学の特注ケーブルを取り付ける前に圧着端子の様子を確認したところ問題が発生しそうな状態の製品が多かったので報告する

# 確認者
@MasatoKubotera 
@YugoNishio 
@NaokiIwasawa 

# 状態
基本は通常の長さから片方だけを近藤科学側（手作業）でカット及び圧着していただいたようであった。
判断した理由は確認したケーブルのほとんどで片方のみ圧着が適切で、もう片一方は品質が不均一であったからである。そのため以下に問題があると判断した圧着端子の様子を掲載する。

1. コネクターとの固定部分の爪が適切に掛かっていない
[image]
写真の一番右側は適切に爪が掛かっているが残りの部分が掛かっていない

---
2. 圧着端子のカット処理が不十分
[image]
青色線の端子に切損じのバリが残っている。断線の恐れが高まるので可動部の多いロボットでは利用が厳しい

---
3. 被覆部の圧着を失敗しているものが混ざっている
[image]
写真では伝わりにくいが、よく見ると青色の部分の圧着がハート型でなく片方がもう一方の下に入り込んでしまってる。これでは張力負荷が掛かった際に抜けやすくトラブルの原因になるので使えない

---
4. 圧着力が高すぎる
[image]
4つを比較すると黒線の圧着端子が他と比べ潰れすぎているのが分かる。この場合多く食い込んだ分導線と被覆に負荷をかけ耐久性が下がる可能性が高まるため適切でない。

---
以上のような品質が一定数混ざっているため実装は厳しいと思われる。
今後の対応としては他の外注候補も検討したい。
候補：[ダイセン電子工業](https://www.daisen-netstore.com/shopdetail/000000000152/ct15/page1/order/)

NaokiIwasawa, 2021-12-27:
実装に当たっての報告は[こちら](https://github.com/citbrains/citbrains_humanoid/issues/1624)

MasatoKubotera, 2022-01-01:
後継については https://github.com/citbrains/OpenPlatform_ver3_0/issues/217 で扱うためcloseします．

closed 2022-01-01 by MasatoKubotera