SUSTAINA-OP/SUSTAINA-OP #19: ハンドルの再設計

MasatoKubotera, 2021-05-22:


AD58-3104, 2021-05-28:
この問題について具体的に言及されているissueはありますか？

MasatoKubotera, 2021-05-28:
具体的な言及や調査は整備の中で行う予定のため現状ありません．
ハンドルの部材であるPOMが歪んでいる．本体取り付け部分が摩耗しているということは聞いています．

MasatoKubotera, 2021-09-06:
胴体・ハンドル間にあるMSU-308-03（SLCG5.5-8）は1機体あたり4個使用されていたが，機体でスラストベアリング付近で使われているMSU-306-03（SLCG5.5-6）に部品種類を削減するために統一した．

これに伴って，ハンドル本体の長さ135mm丸形スペーサ[AETTF8-135-M3-N3](https://jp.misumi-ec.com/vona2/detail/110300208270/?HissuCode=AETTF8-135-M3-N3&PNSearch=AETTF8-135-M3-N3&KWSearch=AETTF8-135-M3-N3&searchFlow=results2products&list=PageSearchResult)を長さ130mmの[AETTF8-130-M3-N3](https://jp.misumi-ec.com/vona2/detail/110300208270/?PNSearch=AETTF8-130-M3-N3&HissuCode=AETTF8-130-M3-N3)に変更した．

[image]

MasatoKubotera, 2021-09-09:
ハンドル部分のスペーサに関して「ねじがよく緩む」ということがあったため，掴んで持ち上げても緩みにくい六角スペーサ[SLSBF7-130-M3-N3](https://jp.misumi-ec.com/vona2/detail/110300206620/?PNSearch=SLSBF7-130-M3-N3&HissuCode=SLSBF7-130-M3-N3&searchFlow=suggesthistory&Keyword=SLSBF7-130-M3-N3&list=SuggestPreview)に変更した．これに伴い，ハンドル軸部分・ハンドル根元部分もともに六角のはめあいとなりより緩みにくくなっている．

[image]「変更後」

[image]「変更前」

また，根元部分のオスメス六角スペーサの締結が通常のナットから圧入するカレイナットに変更され，整備などによる落下を防ぐ．
参照：[埋め込みナットの見直し #16](https://github.com/citbrains/OpenPlatform_ver3_0/issues/16)

MasatoKubotera, 2021-09-09:
POMの歪みについては，根元部分のはめあいなどを改善したことや，ハンドル幅の現象により多少は改善されると推測する．これにより，再設計は終了しcloseする．
今後歪むようであれば部材や形状の変更についても考える必要がある．

MasatoKubotera, 2021-09-29:
現状のスポンジがあるハンドルでは，認識の妨げとなっている．
そのため，現状のハンドルはバンパーの役割も担っているがこの役割については廃止し，六角支柱のみのハンドルとする．
また，設置場所についても位置を下げ認識の妨げにならないようにする．

このハンドルの再設計については，以前の機体でバンドが用いられていたがピックアップのしにくさからこれは考えないものとする．

yasuohayashibara, 2021-09-30:
おそらく現在の背中の形状からすると腰から着地する可能性が高いと思います．
ハンドルはあまり衝撃吸収の役割を果たさなくなるので，無くても良いと思います．

（以下，このissueとは少々外れた議論になりますので，必要あれば別issueで議論して下さい．）

その分，腰に衝撃吸収の仕組みを入れる必要があると思います．
着地時には100G（重力加速度の100倍）以上の加速度になることもありますので，衝撃吸収する機構が無いとメカニズムもそうですが，特に電子回路が故障する確率が高くなります．
衝撃吸収機構が無いと電池と電子回路を，単純に落下させたのと同じ衝撃力が加わりますので急速に劣化していずれ壊れます．
着地時の衝撃力を緩和することで，電子回路の問題の多くは解決すると思います．
（製作の技量が足りていないとかは除いて）

以下，衝撃吸収を議論しているissueの例
https://github.com/citbrains/citbrains_humanoid/issues/469

ロボットが継続的に動けるかいうのは大きく勝敗を分けますので，従来から衝撃吸収に関しては，かなり時間を割いて検討しています．ちなみに，昔は回路トラブルで停止するロボットが１台でもあると，担当者は罰ゲームを受けていました．それぐらい回路トラブルでの停止は**まれ**でした．まあ，止めないためにコネクタを全て廃止するなど過激なことをしていましたが．．．

こちらにもあります．
https://bit.ly/2Y0Bsyy

@Takehiro-Hasegawa 
GankenKunのデータを計測していたような気がしますが，どこにありましたっけ？

MasatoKubotera, 2021-10-19:
衝撃吸収については、下記の３つのissueで取り扱っている。
- [SideBumperの再設計 #163](https://github.com/citbrains/OpenPlatform_ver3_0/issues/163)
- [RearBumperの再設計 #31](https://github.com/citbrains/OpenPlatform_ver3_0/issues/31)
- [FreundBumper_standの再設計 #150](https://github.com/citbrains/OpenPlatform_ver3_0/issues/150)

Ver3.0機体のハンドルは、衝撃吸収の目的でも使用されていたが、Ver4.0での衝撃吸収については上記などのバンパーで行っていく方針である。

>現状のスポンジがあるハンドルでは，認識の妨げとなっている．
そのため，現状のハンドルはバンパーの役割も担っているがこの役割については廃止し，六角支柱のみのハンドルとする．
また，設置場所についても位置を下げ認識の妨げにならないようにする．

の対応として、応急処置的方法となってしまったがハンドル長（100mm⇒60mm）と幅（139mm⇒132mm）を短くすることで対応した。
[image]「Ver4.0　ハンドル長」
[image]「Ver4.0　ハンドル幅」

下図は比較用のVer3.0機体のものである。

[image]「Ver3.0　ハンドル長」
[image]「Ver3.0　ハンドル幅」