SUSTAINA-OP/SUSTAINA-OP #20: 側面ではない面にタップ加工されている箇所の調査

MasatoKubotera, 2021-05-24:
調査要因は<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/16#issuecomment-847089130">埋め込みナットの見直し #16 Comment</a> を参照．

MibuchiYuta, 2021-06-02:
パーツ名,箇所,厚み,材料を調査する。画像の対象はinventorや実機から撮影する。

MibuchiYuta, 2021-06-09:
## 中間報告

実機での箇所の確認は終わりました。
現在Inventorによる図の作成を行なっております。
今週中には必ず終わります。

MibuchiYuta, 2021-06-15:
## 頭部

#### カメラ_メインフレーム
- POM T5
[image]
#### カメラメインフレーム1_KBCR-S02MU
- POM T5
[image]

MasatoKubotera, 2021-06-19:
上記のコメントでカメラメインフレーム1_KBCR-S02MUのタップ加工位置が指摘されていますが，場所が異なります．再度確認してください．

MibuchiYuta, 2021-06-19:
> 上記のコメントでカメラメインフレーム1_KBCR-S02MUのタップ加工位置が指摘されていますが，場所が異なります．再度確認してください．

上記を変更しました。指摘ありがとうございます。

MibuchiYuta, 2021-06-19:
## 下半身

### A2017 T3
- 付け根_フリー軸_1_1
[image]

- 足首_独立ピッチ_1_1

[image]

### A2017 T5
- 足首独立ピッチ_1_2
[image]

- 足裏_サブフレーム_1_1
[image]

MibuchiYuta, 2021-06-19:
## 胴体
### A2017 T3
- 胴体_サブフレーム_1_4
- 胴体_サブフレーム_1_4_MIR
[image]

- 胴体_メインフレーム_1_1
[image]

- 胴体_メインフレーム_1_1
[image]

MibuchiYuta, 2021-06-19:
## 肩
### A2017 T3
- 胴体_メインフレーム_1_1
[image]
上記と同一のパーツです

MibuchiYuta, 2021-06-19:
以上が側面ではない面にタップ加工されている箇所でした。

closed 2021-08-11 by MibuchiYuta