SUSTAINA-OP/SUSTAINA-OP #21: 座らせたときの足裏のスパイク固定ナットとリンク_ふくらはぎサブフレーム1_1(予備案)の干渉

MasatoKubotera, 2021-05-24:


hiroki-0001, 2021-06-02:
啓発活動の優先度は高、こちらはポスターなどの掲示物や使用説明の際にかならず説明を行うようにする

hiroki-0001, 2021-06-05:
こちらは具体的に説明すると、GankenKunを座らせた際に足裏のスパイク固定ナットとリンク_ふくらはぎサブフレーム1_1(予備案)の干渉することで足先が膝にくっついてしまう問題を解決するためのissueです。

hiroki-0001, 2021-06-05:
こちらの問題は2021年6月5日現在では、足の一部を再設計するよりも、適切な処置を施すことで、足を傷つけることなく外せるのでその処置のやり方をミーティング等の場面でメンバー全員に教えた方が手間的には軽いため、優先度は低く設定されています。

MasatoKubotera, 2021-09-18:
この部分について下半身の再設計に伴い再度検証が必要である．

MasatoKubotera, 2021-10-03:
[脚リンク ピッチ軸サーボの位置変更 #103](https://github.com/citbrains/OpenPlatform_ver3_0/issues/103)によるリンクフレームの変更により，下図のようにねじと筋交い部品が干渉しなくなっている．

しかし，干渉しないのはあくまでも足裏がフレームに対して垂直であるときに限る．

足裏の角度によって干渉する恐れがあるため，干渉した時の対処については，今後共有していきたいと考える．
[image]
[image]

MasatoKubotera, 2021-10-03:
また，https://github.com/citbrains/citbrains_humanoid/issues/1301#issuecomment-932686371 を搭載することにより干渉が起こらない場合もある．

MasatoKubotera, 2021-10-19:
対応したためcloseします。