SUSTAINA-OP/SUSTAINA-OP #24: ベアリング周辺の設計を変更し耐久性と動作性を向上させる　（引継ぎ）

NaokiIwasawa, 2021-05-26:
<h2>このissueは https://github.com/citbrains/citbrains_humanoid/issues/1373 
からの引継ぎである。<h2>

NaokiIwasawa, 2021-05-26:
NaokiIwasawa commented on 5 Apr • 

現在使用しているベアリング(MF85ZZ)において部品の故障が頻発している。そしてネジで固定する際にアキシアル荷重がかかることで動きが重くなるので改善できるように対策を講じる。考えられる対策はベアリングの内径を上げる、設計を変更するなどである。

NaokiIwasawa, 2021-05-26:
”ネジで固定する際にアキシアル荷重がかかることで動きが重くなる”
こちらに関して以前林原先生に図解して頂いたものがあるので以下に示す
[image]

解説が示すようにベアリングとその内側のパーツが同じ高さであることが望ましいが、切削精度によって設計とは異なる状態になることがある。この問題はz軸の精度を上げることで改善できると考える。

一方で足首ピッチに関しては別問題となりそうなので以下に示す
[image]
計測をすると設計の段階からベアリングの内側部品(足首_独立ピッチ_1_2)の方がベアリングより0.1㎜低いとわかる。これではフランジがリンクの駆動部分に圧力を与えてしまい動きが重くなるのではないかと考える。
対策として足首_独立ピッチ_1_2のz軸高さを0.1㎜上げベアリングと合わせることである。

NaokiIwasawa, 2021-05-26:
足首_独立ピッチ_1_2のz軸高さを0.1㎜上げベアリングと合わせるために以下の変更を加えた

- 足首_独立ピッチ_1_1の溝を1.5㎜から1.4㎜に変更
　足首_独立ピッチ_1_2の高さを0.1㎜上げることで対応しようとしたが全体で5㎜のパーツだったため断念
[image]写真1：変更後

- 足首_独立ピッチ_1_2の高さを2か所変更

1. 写真2で示した上の部分を3.5㎜から3.6㎜に
[image]写真2：変更後
2. 写真3で示した下の部分を1.5㎜から1.4㎜に
[image]写真3：変更後

以上3か所の変更を行いベアリングの高さを揃えた
[image]
写真4：ベアリング周辺の高さがそろった後

NaokiIwasawa, 2021-05-26:
現状報告
切削を行おうとしたが、そもそも板の公差で0.1mmの精度が出るのかかが微妙であり、全機体のフルメンテの際に他機体の状態も加味して今後の方針を考える。

yasuohayashibara, 2021-05-26:
12号館の切削装置を使用する場合，エンドミル自身はそれ以上の精度で動いているはずですので，切削の方法を検討したほうが良いかもしれません．

精度の向上が見込まれるのは，
１）切削の領域を狭くする．（固定する面が傾いていることへの対応）
２）速度を下げる．（速度を上げると部品に力を加えながら切削するので精度が悪くなるおそれがあります）
３）エンドミルを変える．（切削時に部品に加える力を下げるという観点から）
４）固定の方法を変える．（両面テープで固定しているので，それが良くないかもしれません．）

NaokiIwasawa, 2021-06-02:
類似の[issue](https://github.com/citbrains/OpenPlatform_ver3_0/issues/10)で扱う内容と被っているのでcloseする。

closed 2021-06-02 by NaokiIwasawa