SUSTAINA-OP/SUSTAINA-OP #25: GankenKunの保管方法が妥当か検討する

AD58-3104, 2021-05-26:
citbrains_humanoid #1392より引継ぎ。
先日新たに製作した167の付け根ロール部分のフレームが既に歪んでいる。まだ殆ど歩行等もさせていない機体なので現段階で歪んでいるという事は現在のフィールド横にしゃがませる保管方法が不適切である可能性がある。
保管方法の妥当性を検討し、もしも不適切だと判断された場合は新たな保管場所を作成する。

AD58-3104, 2021-05-26:
一番簡単に手間がかからなさそうなのがミスミのアルミフレームのサービスです。
https://jp.misumi-ec.com/special/alumiframe/frames/pr/?clkid=clkid_alumi_MJP-VF19025_dtl_t1_sc2028_20202

AD58-3104, 2021-05-28:
保管はGankenKunの股の間に梁を通してそこに跨らせるという形を取るのでそれに合わせて股部分の円形の肉抜きを無しにしようと考えています。

MibuchiYuta, 2021-06-19:
既存の台で直立させたときの様子です
台と機体の接触面の高さはメジャーで計測して367.5mmでした
[image]

AD58-3104, 2021-12-13:
最初に設計した時と比べてGankenKunの身長が上がっていましたので少し設計しなおしました。今の所misumi framesで計算した値段が1台当たり2888円となっています。明日学校で実機と合わせてもう一度確認したら確定させますが、一台当たり2888±200円程度になると思われます。
現在の設計です。
[image]

AD58-3104, 2021-12-13:
上の部分にカバーを付けてみたら500円上がりました(´·ω·`)
値段は左下
[image]

MasatoKubotera, 2022-01-16:
@AD58-3104 

設計データをドライブにアップロードしてもらえますか？

※専用設計ソフトウェアの環境は準備できています．

AD58-3104, 2022-01-19:
@MasatoKubotera 
設計データをドライブのハードウェア/4.1プロトタイプ制作/4.1用保管台にアップしました。

MasatoKubotera, 2022-02-08:
MISUMI FRAMESで設計された保管台を6機体分発注しました．
[image]

1機体あたりの保管台の使用部品は下記の通りです．

型番 | 部品名 | 合計
-- | -- | --
HFSF5-2040-250 | 5シリーズ 長方形 20×40mm 2列溝 3面溝 | 1
NFS5-2040-270 | 5シリーズ 長方形 20×40mm 2列溝 4面溝 | 1
NFS5-2040-350 | 5シリーズ 長方形 20×40mm 2列溝 4面溝 | 2
NFS5-2040-480 | 5シリーズ 長方形 20×40mm 2列溝 4面溝 | 1
HBLFSD5-SET | 突起付反転ブラケット | 2
HBLFSN5-SET | 突起付反転ブラケット | 6
HFC5-2040-B | 5シリーズ(溝幅6ｍｍ) アルミフレーム用フレームキャップ | 1
HSCA5-B | 溝カバー 樹脂タイプ | 1

[image]

すべて単体で購入すると，1機体あたり3,389円ですが，MISUMI FRAMESソフトウェア上で注文まとめを行うことで1機体あたり2,680円（外税），6機体分で16,080円（外税）に抑えることができました．

---
MISUMI FRAMESをダウンロードしないとデータの閲覧が行いにくいので[フォルダ](https://drive.google.com/drive/folders/1itrchhFs53Hp-5LbQ99e1WTVUU8_61Ww?usp=sharing)に下記のものをまとめました．
- rdicf（設計データ）
- 外見寸法図
- 組み立て図
- 組み立て展開図
- 部品表
- STL（3Dデータ）<html xmlns:v="urn:schemas-microsoft-com:vml"

closed 2022-02-14 by MasatoKubotera