SUSTAINA-OP/SUSTAINA-OP #26: 3Dプリンターの資料作成

hiroki-0001, 2021-05-26:
期限：6/5まで

現在の3Dプリンターの使用手順のマニュアルの内容の修正と追加をおこなう。
編集する資料は以下のもの
https://docs.google.com/presentation/d/1WxZWb7rKMKwqMuiG7ODh9XqPMiT-jLJ6awMu3v8MAW0/edit?usp=sharing

MasatoKubotera, 2022-01-03:
3Dモデル共有サイトに便利なものがたくさんあるので活用すると良いと思います．
https://www.thingiverse.com/

なかでもAnycubic i3 Mega Sを使用するにあたって，おすすめなものをいくつか紹介します．

- [Bed Leveling Knob "Tire" for Anycubic i3 Mega](https://www.thingiverse.com/thing:3145976)
レベリングがしやすくなるハンドルです．

- [Anycubic I3 Mega / Mega S bed holder (Passgenau)](https://www.thingiverse.com/thing:3573378)
ベットに負荷をかけずスライドできるもの

- [Smart compact temperature calibration tower](https://www.thingiverse.com/thing:2729076)
フィラメントの最適温度を見つけるのに役に立ちます．
https://github.com/MasatoKubotera/GankenKun_3DPrintParts/tree/master/Temp_Calibration_Tower#temp-calibration-tower-for-tpu で使い方をまとめていますので参考に．

- [Anycubic i3 Mega Bed Level](https://www.thingiverse.com/thing:3244876)
レベリング調整の指標として最適です．

MasatoKubotera, 2022-01-03:
参考資料おすすめ

- [【CURA Tips】テストに最適「レイヤー毎にパラメーターを変える方法」](https://note.com/y_labo/n/nf36e9f2d7cd1)
- [スライサー Ultimaker CURA 設定完全攻略 Ver4.8.0対応版（11/12/20更新）](https://note.com/y_labo/n/nee490a276494)
- [Ultimaker Cura Ver4の基本的なスライサー設定](https://voltechno.com/blog/3dprinting-cura/)

MasatoKubotera, 2022-01-03:
ものづくり系Youtuberさん

新しいフィラメントやパラメータ，造形物のヒントはここから

- [麦茶](https://www.youtube.com/channel/UCn23JisBcbb-Nw-3IhofaNA)
- [Efty【えふてぃー】](https://www.youtube.com/channel/UCJyb9WugOMXPgZFspXsqkLg)

closed 2022-08-08 by MasatoKubotera