SUSTAINA-OP/SUSTAINA-OP #27: GankenKunのメンテナンスの高速化（引継ぎ）

NaokiIwasawa, 2021-05-26:
<h2>このissueはhttps://github.com/citbrains/citbrains_humanoid/issues/1367からの引継ぎである<h2>

NaokiIwasawa, 2021-05-26:
Keita1Mori 

理由：試合中に故障した際に部位ごとによってメンテナンスにかかる時間が大きく変化するので、GankenKunの各部位をユニット化しメンテナンスの簡易化及び高速化を実現させたいと思います。

**例：足付け根のロールとピッチ、足首ロールとピッチ**

上記の部位のサーボモータの交換には、多大な時間と労力がかかる(はめ合い構造部分を外さないと、サーボモータが取り出せないため)

NaokiIwasawa, 2021-05-26:
現状報告
ユニット化をしようとしているが、現状サーボモータや切削パーツの個数が不十分なため進捗していない。
また世界大会中など時間が限られた中でメンテナンスを行う際は有効であると思うが、日ごろの運用でどれ程スピードが重要か考慮することもissueを進める上で必要であると考える。（世界大会ではそもそもロボットごと交代さたり足の移植を行うこともあるそうなので）

MasatoKubotera, 2021-12-12:
こちらのメンテナンスの高速化について，Ver4では，嵌め合いを外すことなくサーボモータを取り出せることができるように設計を行っています．そのため，ユニットを作成する必要性が低くなっています．
これらのことより，上記で提案されているユニット化については見送り，世界大会前に再度交換用のユニットが必要であると判断されるようであればopneしたいと思いますのでこのissueはいったんcloseします．

closed 2021-12-12 by MasatoKubotera