SUSTAINA-OP/SUSTAINA-OP #28: 既製品の追加工調査/改善

MasatoKubotera, 2021-05-28:
既製品について追加工が必要なものがある．
既製品を追加工することは，手作業のため精度や労力が問題である．

このissueでは，追加工が必要な既製品，追加工内容，数量，使用箇所を調査する．

さらにここで調査にあがった既製品で追加工が必要になものについては，追加工が必要ない既製品，または既製品にあった設計に変更する必要がある．

MasatoKubotera, 2021-06-29:
SLCG5_5-8.iptが，追加工した後のモデルとなっていた．
追加工したスペーサと未加工スペーサについてはモデルでも区別したい．

例として，
未加工：SLCG5_5-8.ipt
追加工：edit_SLCG5_5-8.ipt
など命名して区別がつくようにしたい．

[image]

MasatoKubotera, 2021-07-29:
上記で指摘したスペーサについては，追加工を必要としない代替品が見つかった．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/127#issuecomment-889121201

Keita1Mori, 2021-08-19:
# 調査報告
# SLCG5_5-6 （スペーサ）
- 加工内容
  - おねじを6mmから3mmに削っている
- 数量
  - 計32個
- 使用箇所
   - スラストベアリング機構(両手、両足で16個）
[image]

   - 付け根ロール_サブフレーム_1_3と胴体サブフレーム_1_4（両腕で8個）
[image]

   - 付け根ロール_サブフレーム_1_3と胴体_メインフレーム_1_1（両足で8個）
[image]

# ネジ(M3×10)
- 加工内容
  - ネジの長さを10mmから9mmに削っている
- 数量
  -  計2個
- 使用箇所
  - 足付け根ロール（付け根ロール_サブフレーム_6_6 と 付け根_サブフレーム_1_3 の固定）
[image]

MasatoKubotera, 2021-08-19:
現状の機体では，長穴減速機構でのジュラコン製のブッシュ，ワッシャも該当する．
こちらは，現在廃止に向け再設計が行われている．

MasatoKubotera, 2021-08-19:
SLCG5_5-6については，[MSU-306-03](https://jp.misumi-ec.com/vona2/detail/221006202577/?searchFlow=combo2products&KWSearch=MSU-300-03&HissuCode=MSU-306-03&PNSearch=MSU-306-03)にすることで，追加工の必要性がなくなる．
これらが使われている部分については，inventorで修正が必要である．

MasatoKubotera, 2021-09-05:
付け根ユニットで使用されている[BRU-305.5](https://www.hirosugi-net.co.jp/shop/g/g8721/)についておねじを6mmから3mmに加工していることが判明しました．

MasatoKubotera, 2021-09-08:
上記部分については，付け根ユニットの構造の変更により使用しないことになりました．

closed 2021-09-08 by MasatoKubotera