SUSTAINA-OP/SUSTAINA-OP #31: リアバンパーの再設計

MasatoKubotera, 2021-06-01:
コックピットの再開発に伴い，コックピットの長さが変わる．
そのため，新しいコックピットに合ったバンパーに再設計する必要がある．

MasatoKubotera, 2021-06-10:
リアバンパーを固定する方法を変える必要がある．
ネジ締めトルクが強くTPUが凹んでいる．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/3#issuecomment-858475959

[image]

MasatoKubotera, 2021-06-25:
胴体フレームの変更もする必要がある．

MasatoKubotera, 2021-06-25:
関連性があるissue：https://github.com/citbrains/OpenPlatform_ver3_0/issues/88

MasatoKubotera, 2021-09-15:
[TPU Bumper のMIRパーツ廃止 #149](https://github.com/citbrains/OpenPlatform_ver3_0/issues/149)で議論されているが，MIRパーツについては廃止する．よって，左右で共通のRearBumperを設計する必要がある．

MasatoKubotera, 2021-09-26:
@RyodaiArai 

RearBumperの設計をお願いします．

現状のVerでは，RearBumperにねじを挿入しドリルで締めるとTPUが圧縮してしまいます．また，M3のネジ径3mmの２本で力を受け止めているため，締結根元に負荷がかかっています．

そのため，Ver4.0では下図に示すスペーサ軸を導入し，根元の負荷軽減，ドリル締めでも圧縮しないような構造にしたいと考えています．
また，現状では左右で違う（MIR）バンパーを使用しており製作や取付時に手間となっているため廃止します．そのため，左右共通で使用できるものを設計して下しさい．
左右共通でできるようにフレームの角度を調節しています．使用については下図で確認してください．

設計する際のスケッチ作成は「変更しやすいスケッチ」を心がけてください．よろしくお願いします．

最新のCADモデルはブランチ[cross_bearings](https://github.com/citbrains/OpenPlatform_ver3_0/tree/cross_bearings)にあります．

[image]

[image]

参考：
コミット履歴ー[b5a247984](https://github.com/citbrains/OpenPlatform_ver3_0/tree/b5a2479841b77757a9c85f2487e416f9f2649b8c)
[image]「167機 Ver」

yasuohayashibara, 2021-09-30:
ごめんなさい．
だいぶ出遅れました．

以下のリンクの議論はこちらで行ったほうが良かったですね．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/19#issuecomment-930678634

衝撃力を吸収するためには，想定される運動量に対して，
適正な　**ストローク**　と　**ばね係数**　を持つダンパーを設計することが必要です．
これができると，電気回路の問題をかなり減らすことができます．

あと，力の加わる方向は必ずしも後ろからではないこともありますので，斜めの転倒もある程度考慮する必要があります．

議論を行ったOBの論文がありますので，参考までにリンクを示します．
http://www.hayashibara-lab.it-chiba.ac.jp/robocup/document/2015/%E8%A1%9D%E6%92%83%E5%8A%9B/1076012_Ogura.pdf

ちなみに，今まで使用した中で最も衝撃力を吸収したのは，以下のようなラジコン用ダンパでした．
ただし，吸収できる力の向きが限られるので，結局は採用されませんでしたが，衝撃吸収に求められる特性ということでは参考になると思います．

必要なのは以下の２つです．
- 長いストローク（変形する長さ）
- 適正なばね係数（理想的には変形に対して一定の力で押し戻すばねがあると良いです）

[image]

真剣に取り組むとけっこう大変な作業になりますが，電子回路を含めたハードウェアがどれだけ継続的に稼働できるかに大きく影響する部分となります．

RyodaiArai, 2021-11-01:
OpenPlatform_ver3_0にデータをアップロードしました．

MasatoKubotera, 2021-11-04:
上記のデータについて確認しましたが，エラーが出てBodyに統合できませんでした．
@RyodaiArai @NaokiIwasawa

MasatoKubotera, 2021-11-07:
上記で作成されたデータの寸法データが破損ししていたため，作成されたモデル形状をもとに再作成した．

[image]「Trunk_back_bumper」

Ver3.0と比較すると，コックピットが後方に延長されたためその分Bumperも延長している．
また，MIRをなくし左右同じ形状にしており，部品種類の低下につながっている．

[image]「装着図」

Ver3.0では，ネジ締結するとTPUがへこみコックピット側にネジが貫通するということがあったが，ワッシャとスペーサを入れることによってこれを防止している．
欠点としては，装着時にワッシャを2枚入れなければならいことだがこれについては，ほかの場所でも該当するため，フランジソケットの導入を検討する．

[image]「断面図」

MasatoKubotera, 2021-12-12:
こちらのバンパーについて，衝撃吸収能力について評価がまだ行えていないことと起き上がり時に傾いてしまい自己位置がずれることがある．
要因は，先端が細くなっているためであると考えられるためさらなる再設計を行う．

MasatoKubotera, 2021-12-12:
この問題については，制御側デモ取り上げられている．
https://github.com/citbrains/citbrains_humanoid/issues/1607

closed 2022-02-07 by MasatoKubotera