SUSTAINA-OP/SUSTAINA-OP #32: 付け根ユニット ピッチ軸の再設計

MasatoKubotera, 2021-06-01:
リンク_6_3でベアリング(MF85ZZ)に挿入されている，POM 足首_スペーサ_1_1の摩耗への対処．
また，付け根ユニットの軸設計の見直し．

NaokiIwasawa, 2021-07-07:
## 現状
[image]
ベアリングに関しては模擬試合を行った際に何度も破損する。またネジを締めるとアキシアル荷重が摩擦を生みリンクの動きの悪化、部品の摩耗を招いている。

そのため耐久性と動摩擦を改善できるよう再設計を行う

## 再設計
CADはまだ作っていないが、イメージと部品選定を行ったので以下にスケッチを示す。
[image]
メインの軸にはステンレススペーサ[BRU-2612S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-2612S&PNSearch=BRU-2612S&KWSearch=BRU-2612S&searchFlow=results2products&list=PageSearchResult)を使用し、先端は外径5mmのベアリング[MF85ZZ](https://jp.misumi-ec.com/vona2/detail/221000058301/?HissuCode=MF85ZZ&PNSearch=MF85ZZ&KWSearch=MF85ZZ&searchFlow=results2products&list=PageSearchResult)にはめ合わす。アキシアル方向はオイレスのワッシャー[80W-08](https://jp.misumi-ec.com/vona2/detail/221000090802/?HissuCode=80W-08&PNSearch=80W-08&KWSearch=80W-08&searchFlow=results2products&list=PageSearchResult)を2枚重ねることで摩擦低減になると考える。
しかし問題が2つある。1つは内径が最小で8.5mm程度であることだ。このままではスペーサーとの間に隙間が生じる。解決案はスケッチの水色破線で示したように付け根_サブフレーム_6_4の外径の一部をワッシャーの内径に合わせる方法だ。2つ目の問題は外径が17mmであることだ。この場合ワッシャーの方が水色の部分より大きくなるため浮いている部分が発生する。自分としては強度的問題はないと考えネジの頭とも干渉はない(隙間[6.5]-(ワッシャー)[1.5*2]＝3.5でネジの頭高さ3mmより広い)ので結果問題ないと判断する。

この設計の利点は付け根_メインフレーム_2_1にスペーサーを貫通させ付け根_フリー軸1_1でネジ締結している点だ。これによりせん断応力に強くなり歩行の安定にも繋がると考える。林原先生の参考[issue](https://github.com/citbrains/OpenPlatform_ver3_0/issues/46#issuecomment-855321166)

yasuohayashibara, 2021-07-07:
もしかしたら単純に転がり軸受をオイレスベアリングに置き換えると問題が解決するかもしれません．
転がり軸受は衝撃に弱いという弱点があります．
ただし，過去のロボットを見てもこの部分には転がり軸受が使用されてきたような気がしますので，何か林原が気づいていない理由があるかもしれません．
転がり軸受を使用する場合は，サイズを上げると衝撃力に強くなります．

転がり軸受と滑り軸受の違い
https://www.monotaro.com/s/pages/productinfo/bearing_comparison/

yasuohayashibara, 2021-07-07:
今見たらGankenKun ver.1はベアリングを保護するように構造になっています．
あと，ほとんど既製品を組み合わせて構成しているようです．
これだと転がり軸受の長所を活かしながら，強度を保てるような気がします．

https://github.com/citbrains/OpenPlatform/tree/master/mechanism/OpenPlatform_ver1_0

[image]

NaokiIwasawa, 2021-07-22:
# 再設計案2
前回の設計案では強度面で改善が出来ていないので、今回の案では
>もしかしたら単純に転がり軸受をオイレスベアリングに置き換えると問題が解決するかもしれません．

 を参考に考えた。変更にあたり追加した部品は以下の5点
- [BRU-311S](https://jp.misumi-ec.com/vona2/detail/221006202476/?HissuCode=BRU-311S&PNSearch=BRU-311S&KWSearch=BRU-311S&searchFlow=results2products&list=PageSearchResult)
- [GFM-0608-025](https://jp.misumi-ec.com/vona2/detail/221000103492/?CategorySpec=00000229772%3a%3ab%0900000229546%3a%3anvd00000000000017%0900000228412%3a%3anvd00000000000016%0900000228697%3a%3amig00000002761897&list=PageCategory)
- [GFM-0607-06](https://jp.misumi-ec.com/vona2/detail/221000103492/?HissuCode=GFM-0607-06&PNSearch=GFM-0607-06&KWSearch=GFM-0607-06&searchFlow=results2products&list=PageSearchResult)
- [平座金](https://jp.misumi-ec.com/vona2/detail/110300252910/?KWSearch=%e3%83%af%e3%83%83%e3%82%b7%e3%83%a3%e3%83%bc&searchFlow=results2products&list=PageSearchResult)
  - M3で厚さ0.5mmの物
- 内径7mm,外径11mm,厚さ７mmのスペーサ
  - POMから切削

である。部品点数は増えるがラジアル荷重、スラスト荷重共にワッシャが耐えられるように意識して設計した。

## 変更後の様子
[image]

1,2はオイレスのブッシュで3は切削で自作するのを考えている。

また座金はCADに反映できていないが、外側からBUR-311Sをネジで固定する際に使う。座金無しでは0.5mmの隙間がブッシュとBUR-311Sの間に生じる。

NaokiIwasawa, 2021-07-23:
ミーテングで話し合った結果以下の3点について問題が見つかった
1. スペーサーを座金と合わせて締結するにはネジの頭(Φ5.5)の方が座金(内径6.4mm)より小さく不可能な点
2. POMからの切削には精度問題が残る点
3. 前回のスケッチと比べてせん断応力に弱くなっている点

そのため今後解決できるように取り組むが、POMの精度問題に関しては既製品で外径違いでも実現できないか検討する。またポケット加工も避けたいためスペーサを埋め込む際は切り抜きで検討する。
＊ネジの頭の高さ：3mm

NaokiIwasawa, 2021-07-27:
# 再設計案3

[image]

## 使用部品

スペーサーメスメス：[ARU-312.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-312.5S&PNSearch=ARU-312.5S&KWSearch=ARU-312.5S&searchFlow=results2products&list=PageSearchResult)

113円　ステンレス　外径6mm,L=12.5mm

---

座金：[	FFW-0312-20B](https://jp.misumi-ec.com/vona2/detail/221006200946/?HissuCode=FFW-0312-20B&PNSearch=FFW-0312-20B&KWSearch=FFW-0312-20B&searchFlow=results2products&list=PageSearchResult)　

50円　スチール　黒塗装(反射防止に最適)

---

ブッシュ中間：[GSB-061305](https://jp.misumi-ec.com/vona2/detail/221000090879/?HissuCode=GSB-061305&PNSearch=GSB-061305&KWSearch=GSB-061305&searchFlow=results2products&list=PageSearchResult)

68円　充填材入りホ゜リフェニレンサルファイト゛　許容最高面圧 P(N/mm2)	14.5

---

t=2.5の座金　いくつか候補がある

候補1[UUW-0613-25](https://jp.misumi-ec.com/vona2/detail/221006201172/?HissuCode=UUW-0613-25&PNSearch=UUW-0613-25&KWSearch=UUW-0613-25&searchFlow=results2products&list=PageSearchResult)   77円　ステンレスSUS303  

候補2[UUW-0613-25H](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0613-25H&HissuCode=UUW-0613-25H&searchFlow=suggest2products&Keyword=UUW-0613-25H&list=SuggestPreview)  165円　ステンレスSUS304

候補3[FFW-0613-25](https://jp.misumi-ec.com/vona2/detail/221006200946/?HissuCode=FFW-0613-25&PNSearch=FFW-0613-25&KWSearch=FFW-0613-25&searchFlow=results2products&list=PageSearchResult)　42円　スチール　三価クロメート

候補4[FFW-0613-25N](https://jp.misumi-ec.com/vona2/detail/221006200946/?PNSearch=FFW-0613-25N&HissuCode=FFW-0613-25N&searchFlow=suggest2products&Keyword=FFW-0613-25N&list=SuggestPreview) 50円　スチール　ニッケルメッキ

候補5[FFW-0613-25B](https://jp.misumi-ec.com/vona2/detail/221006200946/?PNSearch=FFW-0613-25B&HissuCode=FFW-0613-25B&searchFlow=suggesthistory&Keyword=FFW-0613-25B&list=SuggestPreview) 53円　スチール　ミクロブラック（黒色コーティング）

---
## 特徴
前回問題だった切削精度を気にしなくていいように、全て既製品の組み合わせにした点。そしてスペーサをメスメスにし、座金をかませたことで傷付き防止・安定した固定を実現した。また、せん断に強いようスペーサの先端をT2のアルミに埋め込んだ。その代わり、十字の”付け根_フリー軸_1_1”は廃止した。
以上によりラジアル・アキシアル共に滑り軸受けで荷重を支え衝撃に強くなったと考える。

選定した座金について外見に関わる部分は黒色を選択できるものを探した。そしてコストの低いものを選んだのも特徴である。しかし、t=2.5の座金に関しては候補が多かったので主に表面の仕上げで選定を迷っている。

MasatoKubotera, 2021-07-28:
@NaokiIwasawa 

無給油ブッシュ[GFM-0608-025](https://jp.misumi-ec.com/vona2/detail/221000103492/?HissuCode=GFM-0608-025&PNSearch=GFM-0608-025&KWSearch=GFM-0608-025&searchFlow=results2products&list=PageSearchResult)についてですが，
[image]
凸部分とフランジ部分にR加工（最大0.5）があるようです．通常の輪郭切削するとこの部分がリンクのガラエポと干渉するので注意が必要です．（0.5だけポケット加工すれば干渉は解決できますが．．．）
CADがMiSUMiからダウンロードできないようですが，手書きスケッチではなくInventorに起こして干渉確認をした方が良いかもしれません．

また，下図よりf=0.5ということが分かります．
GFM-0608-025の寸法は，d1=6.0，d2=8.0，d3=12.0，b1=2.5，b2=1.0であるため，ガラエポT2の穴と密着するのは厚さ1となります．主観ではありますが，密着する面積が少ない気がします．全長が4mmである[GFM-0608-04](https://jp.misumi-ec.com/vona2/detail/221000103492/?HissuCode=GFM-0608-04&PNSearch=GFM-0608-04&KWSearch=GFM-0608-04&searchFlow=results2products&list=PageSearchResult)もあるので検討してみてください．
[image]
※[GFM-0608-025 メーカ公式ページ](https://www.igus.co.jp/product/64?artNr=GFM-03040-15)にある，[イグリデュールG カタログPDF](https://www.igus.co.jp/contentData/Products/Downloads/iglidur_G_FM_GBen.pdf)のP13の図より引用，MiSUMiのHPでも確認できる．

ベアリングは廃止できましたが，リンクと付け根ユニット間の摩擦について解決できているか後日議論できればと思います．

NaokiIwasawa, 2021-08-03:
# 再設計案4

[image][image]
写真1：外観　写真2：説明用断面図

## 使用部品

写真2の左から順に

[FFW-0312-20B](https://jp.misumi-ec.com/vona2/detail/221006200946/?HissuCode=FFW-0312-20B&PNSearch=FFW-0312-20B&KWSearch=FFW-0312-20B&searchFlow=results2products&list=PageSearchResult)（50円）　
[GSB-081404](https://jp.misumi-ec.com/vona2/detail/221000090879/?PNSearch=GSB-081404&HissuCode=GSB-081404&searchFlow=suggest2products&Keyword=GSB-081404&list=SuggestPreview)（77円）　
[GSB-081235](https://jp.misumi-ec.com/vona2/detail/221000090879/?HissuCode=GSB-081235&PNSearch=GSB-081235&KWSearch=GSB-081235&searchFlow=results2products&list=PageSearchResult)（77円）　
[80F-0608](https://jp.misumi-ec.com/vona2/detail/221000090789/?CategorySpec=00000229772%3a%3ab&list=PageCategory)（32円）　
[FFW-0312-20B](https://jp.misumi-ec.com/vona2/detail/221006200946/?HissuCode=FFW-0312-20B&PNSearch=FFW-0312-20B&KWSearch=FFW-0312-20B&searchFlow=results2products&list=PageSearchResult)（50円）　

＊80F-0608についてはCADデータがネット上にあったが他はなかったので再現

スペーサーメスメス：[ARU-312.5S](https://jp.misumi-ec.com/vona2/detail/221006202465/?HissuCode=ARU-312.5S&PNSearch=ARU-312.5S&KWSearch=ARU-312.5S&searchFlow=results2products&list=PageSearchResult)（113円）全ネジタイプ

---

## 特徴

前回の無給油ブッシュ[GFM-0608-025](https://jp.misumi-ec.com/vona2/detail/221000103492/?HissuCode=GFM-0608-025&PNSearch=GFM-0608-025&KWSearch=GFM-0608-025&searchFlow=results2products&list=PageSearchResult)だと指摘があった通り、フランジのR加工が問題であったが[80F-0608](https://jp.misumi-ec.com/vona2/detail/221000090789/?CategorySpec=00000229772%3a%3ab&list=PageCategory)に変更するとフランジの根元に加工がないので解決できた。そして、外径違いのGSBを組み合わせることで7.5mmの隙間も埋めることがでた。

GSB-081404の内径部は一部浮いているが1.5mm分はブッシュと接している。1.5mmについて、この分はラジアル荷重がほぼかからないためまったく問題ないと判断する。

@MasatoKubotera 
確認していただきありがとうございます！

>リンクと付け根ユニット間の摩擦について解決できているか後日議論できればと思います．

滑り軸受けとアルミの付け根ユニットが接しているので問題なくアキシアル荷重を逃がせていると考えます。
案4では更に滑り軸受け同志が接触する部分が多いため問題なし

>ネジと板間に入れる座金についてですが、フランジソケット(フランジ付き六角穴付きボルト)の方が取り扱いが楽かも知れません。

フランジソケット付きネジについても探したがフランジ径が12mmの物は見つからなかった。この部分はリンクの曲げモーメントがかかった際にブッシュに応力が集中するのを避ける目的がある。そのため座金は必要で、ねじ止めするにも最低でスペーサの径以上（6mm）が必要。

>交差を考えない理論値上は、スペーサ長ドリンク部材の高さの差は無いのでブッシュに圧力を加え締結することはないかと思います。

仰る通り理論上は交差が無ければ締め付けによる圧力はかかりませんが、リンクに曲げがかかると圧力が加わる恐れがあります。

yasuohayashibara, 2021-08-25:
@NaokiIwasawa 
多くの調査が必要でしたので，だいぶ遅くなってからのコメントで申し訳ありません．

以下，改善したほうが良いと思います．

１）シャフトBRU-312.5Sが付け根_メインフレーム_2_1に固定されていない．
GSB-081235などのスペーサを挟んで抑え込んでいるので，抜けることはないと思いますが，強度とメンテナンス性を考えると固定した方が良いと思います．

２）ネジの頭に六角レンチを入れるためにはモータを外さなければならない．
ネジが緩んだときに締め直すのが大変かもしれません．
→　インサートナットを使用すると良いと思いますが，私も大きなフランジ付きのナットを見つけることができませんでした．
→　もしくは横から工具で固定できる六角ネジ（六角穴付きではない）を使用すると良いかもしれません．

現在の設計を元に，強度とメンテナンス性に配慮すると，以下のような構成があるかなと思います．
おそらく今まであまり馴染みのない部品も含まれていると思いますので，参考にしていただければと思います．
（短期間で考えたので，寸法が間違っているおそれがあります．あくまで参考ということで．．．）

https://photos.app.goo.gl/iQB3mbFZ2xD9ZHTj9

yasuohayashibara, 2021-09-04:
解析しました．結果的に，再考した方が良いと思われます．
従来の機構と異なり，提案する機構はアルミが変形すると歩行しづらくなるおそれがあることも発見しました．
従来の機構もある程度はあったと思われますが，提案する機構は影響が大きくなります．

以下，簡易的ではありますが，解析結果です．
[210905シャフト周辺の解析結果.pdf](https://github.com/citbrains/OpenPlatform_ver3_0/files/7110180/210905.pdf)

解析に用いたデータはこちらにアップロードします．（ファイルサイズが大きいので，現在アップロード中）

NaokiIwasawa, 2021-09-05:
@yasuohayashibara 
お忙しい中解析していただきありがとうございます。数値に基ずく設計能力が未熟なためとても参考になります。
解析を見ると林原先生の案の利点や自身の提案の問題点などがよく分かりました。特に曲げ応力も大いに考慮するべきだったと思いました。
今後は問題点を改善し、より適した設計になるよう先輩と再考してみようと思います。
つきまして、引き続きアドバイスよろしくお願いします。

yasuohayashibara, 2021-09-06:
承知しました．
私も解析してみてver.3の機構の問題点を発見しましたので，学びながら進めている感じです．
リソースが足りていない中で進めていると思いますので，あまり無理をしないようにしてください．

NaokiIwasawa, 2021-09-15:
# 再設計の5
林原先生の助言を参考に他の部分も軸を見直し共通化を実施しました。報告遅れました。

## 脚部の大改良による変更
ピッチ軸の固定フレームLeg_Joint_side_frame（旧名付け根_メインフレーム_2_1）の形状が変更しました。
[image][image]
左画像：前　右画像：後
ピッチのサーボモータは膝に移動しました。

## 軸の解説
### リンク前方のピッチ
[image]

**使用部品**
外側から
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円 SUSワッシャー
[LFW-0815](https://jp.misumi-ec.com/vona2/detail/221000090734/?PNSearch=LFW-0815&HissuCode=LFW-0815&searchFlow=suggest2products&Keyword=LFW-0815&list=SuggestPreview) 58円 オイレス
T2のGFRP
[80F-0603](https://jp.misumi-ec.com/vona2/detail/221000090789/?PNSearch=80F-0603&HissuCode=80F-0603&searchFlow=suggest2products&Keyword=80F-0603&list=SuggestPreview) 32円 オイレス
[CC-0612-20B](https://hirosugi.co.jp/products/POM/CC-B.html) 11円　ジュラコンワッシャー
[CC-0612-20B](https://hirosugi.co.jp/products/POM/CC-B.html) 11円　ジュラコンワッシャー
[CC-0612-25B](https://hirosugi.co.jp/products/POM/CC-B.html) 11円　ジュラコンワッシャー
[UUW-0312-10BH](https://jp.misumi-ec.com/vona2/detail/221006201172/?PNSearch=UUW-0312-10BH&HissuCode=UUW-0312-10BH&searchFlow=suggest2products&Keyword=UUW-0312-10BH&list=SuggestPreview) 36円　SUSワッシャー
Leg_Joint_side_frame T2
[S3-07-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?PNSearch=S3-07-3W&HissuCode=S3-07-3W&searchFlow=suggesthistory&Keyword=S3-07-3W) カレイナット
軸のスペーサ:[BRU-311S](https://jp.misumi-ec.com/vona2/detail/221006202476/?PNSearch=BRU-311S&HissuCode=BRU-311S&searchFlow=suggest2products&Keyword=BRU-311S) 117円

### リンク後方のピッチ
[image]

基本は前方と同じだがカレイナットは不使用

yasuohayashibara, 2021-09-15:
ごめんなさい．
問題があれば次の設計で対応してくれれば良かったのですが，良い設計だと思います．

一点，CC-0612-20BとCC-0612-25Bの見分けが付きにくく製作時に混乱するおそれがあります．
明らかにサイズが異なっていて，製作時に間違いが明らかに分かるようなサイズの組み合わせにしたほうが良いと思います．

NaokiIwasawa, 2021-09-15:
@yasuohayashibara 返信ありがとうございます

> 一点，CC-0612-20BとCC-0612-25Bの見分けが付きにくく製作時に混乱するおそれがあります．

ご指摘の通りだと思います
色違いのCC-0612-25とCC-0612-20がありますので、混乱を避けるために2.5を白色にしようと思います。
[白色CC-06シリーズ一覧](https://hirosugi.co.jp/products/POM/CC.html)

MasatoKubotera, 2021-09-17:
@NaokiIwasawa 

CC-0612-25B から CC-0612-25に変更しました．

[image]

MasatoKubotera, 2021-10-03:
再設計が終了したためcloseする．

closed 2021-10-03 by MasatoKubotera