SUSTAINA-OP/SUSTAINA-OP #33: ゼッケンの再設計

MasatoKubotera, 2021-06-02:


hiroki-0001, 2021-06-05:
こちらのissueは現在のGankenKunは以下の写真のように、**プラバンをマジックテープでくっつけて、簡易的なゼッケンとしている。**
[image]

現状のままだとGankenKunの初期位置を調整する際の邪魔になる、またレギュレーション違反になる可能性があるので、3Dプリンタ等でゼッケンを作成するというissueである。

hiroki-0001, 2021-06-09:
1年生がやってくれると嬉しいな......

FUKUCHIGarbaldigus, 2021-06-26:
## 製作にあたり必要なルール事項

[image]
[原文ルールブック](https://cdn.robocup.org/hl/wp/2021/04/V-HL21_Rules.pdf)Colours項より

・ロボットの色はほとんどが黒か、ダークグレー（RAL 7011 アイアングレー以上）で、無反射でなければなりません。ロボットはアルミのようなシルバー、グレー、ホワイトに着色してもよいが、その場合は足の色を黒にしなければならない。フィールドに使われている色（緑、白）や、相手チームのチームマーカーと似た色は避けなければならない。ロボットの腕、足、胴体はソリッドな形状でなければならない。
・ロボットにはチームマーカーを付けなければならない。このマーカーは、一方のチームには赤、もう一方のチームには青で表示されます。
・チームマーカーは、一方のチームは赤、もう一方のチームは青である。ロボットの腕、脚、胸にあるすべてのチームマーカー（最大20個）の合計可視面積は少なくとも0.06・Htop^2でなければならない。
・両側（左、右、前、後）にある1～5個の大きなチームマーカーの可視領域は、少なくとも0.015・Htop^2でなければならない。
・チームが使用する色はランダムに割り当てられ、ゲームプランで発表されます。
・各チームのロボットは、一意に識別できるものでなければならない。番号や名前が付いていなければなりません。ゴールキーパーロボットは、審判がチームの他のロボットと簡単に区別できるように、独自のマークを付けなければならない。
・2つのチームは互いに識別できる色を身につけなければなりません。
[翻訳ルールブック](https://docs.google.com/document/d/1Ms5g2x1dqoK6ML64X3x4QXYdfRD6G1bDnvPIPmgWV34/edit#)カラー項より

*「Htop^2」は「ロボットの全高の二乗」と思われる。

MiyawakiKaito, 2021-06-26:
GanKenKunの大きさは実測した結果640㎜でした。TDPの情報とも一致しているので間違いないと思われます。

FUKUCHIGarbaldigus, 2021-06-30:
ゼッケンの基本形案を作りました。胸部前面です。
青部分は不透明のフィルムやシート、黒部分はTPUなどを想定しています。
TPUの軟性で広げて覆いかぶせるようにはめて装着します。
ただし、この一個だけでは必要面積である0.015*Htop^2(6144mm^2)を満たさないため、前面はもっと必要そうです。
[image]
[image]
[image]

MasatoKubotera, 2021-07-02:
メモ：
>GanKenKunの大きさは実測した結果640㎜でした。TDPの情報とも一致しているので間違いないと思われます。

により，H_top=640[mm]になるということで，
**合計可視面積**は，少なくとも0.06×H_top^2=0.06×640[mm]^2= **24,576[mm^2]** 必要．
※参考：160[mm]×160[mm]=25,600[mm^2]

両側(左・右・前・後)にある1～5個の大きい可視領域は、少なくとも0.015×Htop^2=0.015×640[mm]^2= **6,144[mm^2]** に必要．
※参考：80[mm]×80[mm]=6,400[mm^2]

MasatoKubotera, 2021-07-02:
@FUKUCHIGarbaldigus 
青い部分(不透明のフィルムやシート)と黒部分(TPU)の接着方法はどのように考えていますか？

MiyawakiKaito, 2021-07-03:
@FUKUCHIGarbaldigus  
新たな目標として、福知君は引き続き胴体部分のゼッケンを設計・改良を、自分は腕の部分のゼッケンを新たに設計することを目標にしたいと思うのですが福知君はそれでいいでしょうか。何かある場合は連絡をください。また、ゼッケンは個数に限りがあるので、なるべく少ない個数で面積をとれるゼッケンを設計してもらいたいとのことです。

FUKUCHIGarbaldigus, 2021-07-03:
@MasatoKubotera 
接着はTPU用の強力接着剤を想定しています
@MiyawakiKaito 
分担の方と面積の方了解しました。私はその方針で大丈夫です。

yasuohayashibara, 2021-07-03:
3Dプリンタで印刷するとしたら積層方向にお気をつけください．
積層方向は比較的簡単に剥離しますので棒状のものを設計するときには注意が必要です．

参考動画
https://youtu.be/nlcI8ruEnIs

MasatoKubotera, 2021-07-04:
@FUKUCHIGarbaldigus 
はじめから接着剤を使用する設計は控えるようにお願いします．
案として，TPUとカラーシート(PP板とか)も着脱式にすることで，TPUの製作コストが1/2になるかと思いますので検討してみてください．

TPUはかなりプリントが難しくPLAのように簡単な素材ではないので，すべてを設計する前に一度プリントしてみた方が良いかもしれません．

FUKUCHIGarbaldigus, 2021-07-09:
基本形の第二案です。部材同士の固定はM3ねじとナットで想定しています。黒の部材のみTPUです。3Dプリンタの積層方向を意識し、サポート材なしでプリント可能かつ裂けにくくしてあります。また、青のフィルム部分は挿入するのみですので、取り換えも簡単です。デメリットとしては、前回案より面積が非効率的であることです。今回は基本形案ですので、これ一つでは正面の面積は補えません。(これのみだと3610mm^2)
[image]
[image]
[image]
[image]

MasatoKubotera, 2021-07-09:
@FUKUCHIGarbaldigus

>部材同士の固定はM3ねじとナットで想定

とありまあすが，GankenKunでナットを使用していると整備や交換時に扱いが大変であり，現在カレイナットORインサートナットに変更を行っています．白色の部材が定義されていないので確かなことは言えませんが，タップ加工でもよいかもしれません．
参考：[埋め込みナットの見直し #16](https://github.com/citbrains/OpenPlatform_ver3_0/issues/16)

>これ一つでは正面の面積は補えません。(これのみだと3610mm^2)

現状の機体を見ると，前側のゼッケン位置は胸×1と足のリンクの筋替え部分×4です．
数値だけ見ると足りなくて心配するような値ではないと思うので，今後設計していければと思います．

前回のコメント
>TPUはかなりプリントが難しくPLAのように簡単な素材ではないので，すべてを設計する前に一度プリントしてみた方が良いかもしれません．

でも提言していますが，一度プリントしてみましょう．
プリントすることで課題や改善点が見つかるかと思います．
また，
>3Dプリンタの積層方向を意識

がプリントを経験することで容易にできるようになると思います．
（TPUのパラメータ調節をしている１年生はいませんので一からなのスタートになります．．．）

FUKUCHIGarbaldigus, 2021-07-18:
正面のゼッケン案がとりあえずできました。
面積は6200mm^2です。前回のものを拡大すれば十分面積が取れることがわかったのでそのまま拡張しました。
M3で白の部材(新人研修でギアボックスに使う素材などを想定)にねじ切りをしたこと、面積が変わったこと以外そのままです。
この形でTPUを出力してみて、試作品を作ってみて改良を加えようかと思います。
[image]
[image]

MiyawakiKaito, 2021-07-19:
<img src = "https://user-images.githubusercontent.com/83056429/126101243-9934f57c-0880-4df0-8807-b62d20646530.png"
width = "320px">

MiyawakiKaito, 2021-07-19:
<img src = "https://user-images.githubusercontent.com/83056429/126101477-05bab7b8-4051-4678-90c1-ba7a5332a626.png"
width = "320px">

NaokiIwasawa, 2021-07-23:
@MiyawakiKaito 
上記の画像では全体と腕に装着後の様子がわからないので判断が出来ません。全体の様子がわかる画像の提示をお願いします。
また面積についても書いてないので示していください。
もし分からないことがあれば連絡ください。

MiyawakiKaito, 2021-07-29:
自分も福知君と同じような方法を用いて腕部分のゼッケンの設計をいたしました。干渉解析している部分もなかったので構造上は問題ないと思います。面積に関しては、必要最低限の可視面積0.015×Htop^2=0.015×640[mm]^2= 6,144[mm^2]には満たない3050mm^2だったので、また新たに足の部分のゼッケンも設計します。 
<img src = "https://user-images.githubusercontent.com/83056429/127413393-8791acaf-9456-4405-bced-b4a2c5c12787.png"
width = "320px">
<img src = "https://user-images.githubusercontent.com/83056429/127413786-18bf9ea1-aa1b-48c4-8b12-0e27a425467f.png"
width = "320px">

MasatoKubotera, 2021-07-30:
@MiyawakiKaito 

TPUで作成する3Dプリンターパーツですが，積層方向によっては曲げにくいということも起きてきます．**特性を知るためにも次の設計に入る前に一度プリントを行ってみましょう．**

設計してくれたゼッケンですが，青色の板材がゼッケンとなり赤バージョンも用意するということでしょうか．．
青色，赤色のこれだけの厚さのPOMなどの部材は用意できないので，スプレーなどで着色することになるかと思います．
表面裏面で青色赤色になっていれば板材が節約できそうですね．

TPUを被締結部材とする場合，相手材とねじ頭の圧力によりTPUが変形します．
例として，現状のリアバンパーがそうで締め付けトルクを一度でも間違えるともとには戻らず相手材とねじ頭間が薄くなっていきます．そのため，TPUにねじを通す場合は何かしらの処置が必要になります．

MiyawakiKaito, 2021-08-11:
現在3Dプリンターが使えない状態なので、新たに足部分のゼッケンを設計しました。面積は2000mm^2でしたので、あと少しゼッケンを付け加えれば必要最低限の可視面積0.015×Htop^2=0.015×640[mm]^2= 6,144[mm^2]を満たすことができます。
<img src = "https://user-images.githubusercontent.com/83056429/128968175-ccc05117-b71e-404b-b6f5-ddce248a5e9d.png"
width = "320px">
<img src = "https://user-images.githubusercontent.com/83056429/128968960-092885df-6998-495a-ae52-ebbe168fec7f.png"
width = "320px">
<img src = "https://user-images.githubusercontent.com/83056429/128968999-3dd150e7-2a98-4253-bbab-fbd13e41f30f.png"
width = "320px">

FUKUCHIGarbaldigus, 2021-08-20:
ねじ穴周りの厚さが薄く脆いとの指摘をうけ、新たに肉盛りを追加しました。また、胸ゼッケンの情報をもとに、背中のゼッケンの設計をしました。胸との違いは固定用のツメの数とサイズ比です。こちらも胸と同じく、一枚で最低面積6144m^2をクリアすることができます。
正面
[image]
背面
[image]
胸ゼッケン
[image]
[image]
背中ゼッケン
[image]

NaokiIwasawa, 2021-08-20:
背中のゼッケンについてですが、コックピット裏はカメラケーブルなどのケーブル類と干渉する恐れがあるので、付け根部分の長さ調節が必要かもしれません。合わせてコックピット裏には現在電圧計が装着されているのでかぶってしまう恐れがあります。
但し、現在コックピットの再設計を行っているので配線周り・フレーム自体は変更があることを知っておいてください。詳細は @MasatoKubotera 先輩に確認お願いします！
[image]

MasatoKubotera, 2021-12-18:
ゼッケンの再設計を行うにあたって，優先順位が低いこと．ゼッケンに予算がかけることができません．
そのため，このゼッケン並びにチームに光学式3Dプリンタを採用することができないため，積層3Dプリンタまたは，現状と同様のPPシート＋色紙などの方法になるかと思います．

MasatoKubotera, 2022-01-05:
リンク筋交いに装着する場合の注意点
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/226

MasatoKubotera, 2022-02-01:
こちらでVer4.1に対応するゼッケンの議論も進めていきたいと思います．

MasatoKubotera, 2022-02-01:
ルールに関するメモ
https://github.com/citbrains/OpenPlatform_ver3_0/issues/33#issuecomment-872765877

現時点での[コミット](https://github.com/citbrains/OpenPlatform_ver3_0/commit/91587230c34ea43f840c650f066e1a70935f5cbf)でのH_topは，640.59mm≒641mmとなります．

[image]

合計可視面積は，少なくとも0.06×H_top^2以上必要であるため，
0.06×H_top^2=0.06×641^2=24,652.86[mm^2]となります．
※参考：160[mm]×160[mm]=25,600[mm^2]

両側(左・右・前・後)にある1～5個の大きい可視領域は、少なくとも0.015×Htop^2必要であるため，
0.06×H_top^2=0.015×641^2=6,163.215[mm^2]となる．
※参考：80[mm]×80[mm]=6,400[mm^2]

MasatoKubotera, 2022-02-01:
前エリア

コックピット（135x68.5=9,247.5mm^2）一か所に貼付することで条件をクリアすることができる．
この場所には機体番号１～６を記載したいと考えている．
文字にとれる面積は，9,247.5mm^2-6,163.215mm^2=3,084.285mm^2（約55x55mm・約68.5*45mm）と十分である．

MasatoKubotera, 2022-02-01:
ゼッケンの素材について

ゼッケンの要件として，脱着が容易であることが望まれる．
現状は，PPシートに色画用紙，マジックテープの構成であり消耗が激しい．

そのため，このissueでは3Dプリンタ製のものが提案されてきたが，現状造形の難しさと技術者不足があるため，Ver4.1では用いらない．

そのため，新たに提案するのは塩ビ板である．

経験談になってしまうが，RoBoCupJuniorSoccerでの機体の側面カバーとしてT1のものを使用していたが貫通することもなく耐久性的には十分である．
https://twitter.com/CreateRoboCup/status/952146280486072320?s=20&t=HGQDuNIl8ncPAMgMRQtrWw

今回はゼッケンでありそこまで厚い必要がないため，T0.5程度でも良いかと思う．
また，この塩ビ板はT１でも手加工（通常のカッター・ハサミ）で加工でき，切削することもできる．
厚みも薄く，ピッチも一層で済むため，手加工するよりも圧倒的に切削の方が時間的には早く，正確であるため，工場が利用できるのであれば切削で対応したいと考えている．

ホームセンターの規模にもよるが，実店舗でも購入できる．

塩ビ板の候補
- [光 塩ビ板 200×300×1 EB231-2 アカ](https://www.amazon.co.jp/%E5%85%89-%E5%A1%A9%E3%83%93%E6%9D%BF-200%C3%97300%C3%971-EB231-2-%E3%82%A2%E3%82%AB/dp/B00AQBAIOG) ４５０円（Amazon）
- [光 塩ビ板 赤 0.5×200×300mm](https://www.amazon.co.jp/%E5%85%89-%E5%A1%A9%E3%83%93%E6%9D%BF-%E8%B5%A4-0-5%C3%97200%C3%97300mm-EB235-2/dp/B00AQBAPS0?th=1) １７０円（Amazon）
- [光E B231-2 [塩ビ板 赤 200x300x1mm]](https://www.yodobashi.com/product/100000001003815532/)　２８０円（ヨドバシ）
- [光 EB231-6 [塩ビ板 青 200x300x1mm]](https://www.yodobashi.com/product/100000001003815536/?gad1=&gad2=g&gad3=&gad4=452613146044&gad5=12323336745747838672&gad6=&gclid=Cj0KCQiA0eOPBhCGARIsAFIwTs7IDCWKQdkJGAVSpWV53ozWMRDhXMD5aAm-r85tOsJT63xzoDQOKMAaAlGXEALw_wcB&xfr=pla)　２８０円（ヨドバシ）
- [光 EB235-2 [塩ビ板 赤 0.5x200x300mm]](https://www.yodobashi.com/product/100000001003815577/)　１８７円（ヨドバシ）
- [光 EB235-6 [塩ビ板 青 0.5x200x300mm]](https://www.yodobashi.com/product/100000001003815581/)　１８７円（ヨドバシ）

MasatoKubotera, 2022-02-01:
横エリア

ForeArm_outside_frameの形に添って加工し，取り付けたいと考えています．

簡単な指標ではありますが，面積的には6465.380 mm²であるため，左右ともにそれぞれ１か所ずつ貼付することで条件を満たすことができます．

[image]

MasatoKubotera, 2022-02-01:
横エリア

ブッシュを使用し，ネジ締め・マジックテープを使わずに装着できる方法を考えてみました．
ゼッケンは上記の塩ビ製で曲げながら装着することを前提に設計しています．

簡単な設計なので，時間があるときに詰めていきたいと思います．

※面積が足りないため，腕を太くして対応する必要あり．

今回の設計で使用したブッシュは[L3035-5810TB](https://www.hirosugi-net.co.jp/shop/g/g86133/)です．
[image]
[image]

MasatoKubotera, 2022-02-09:
上記で提唱していたゼッケン用の板が入手できました．

購入店舗：スーパービバホーム新習志野店

PPクラフトシート
色番：PS-5青，PS-4赤
サイズ：L 565×980ミリ
厚み：0.75
材質：ポリプロピレン
比重：0.9

[image]
[image]
[image]

[image]
[image]
[image]

表裏で仕様が異なり，ツルツルとザラザラの2種でどちらとも非光沢仕様です．

---

新習志野キャンパス前のカインズでも在庫が確認できました．
- [【店舗限定】PPクラフトシート PS－4 赤 厚み0.75mm×490mm×565mm](https://www.cainz.com/g/4953463171045.html)
- [【店舗限定】PPクラフトシート PS－5 青 厚み0.75mm×490mm×565mm](https://www.cainz.com/g/4953463171052.html)
- [【店舗限定】PPクラフトシート PS－5 青 厚み0.75mm×565mm×980mm](https://www.cainz.com/g/4953463170055.html)
- [【店舗限定】PPクラフトシート PS－4 赤 厚み0.75mm×565mm×980mm](https://www.cainz.com/g/4953463170048.html)

MasatoKubotera, 2022-03-02:
https://github.com/citbrains/SUSTAINA-OP/issues/265#issuecomment-1048342753 にて，上記PPクラフトシートでの製作が可能と判断されたため，このPPクラフトシートを使ってゼッケンは作成していきたいと思います．

MasatoKubotera, 2022-06-27:
前後左右のゼッケンを設計して実装しました．

前：Cockpit_front_frame_cloth
後：Cockpit_back_frame_cloth
左右（共通）：ForeArm_outside_cloth

[image]
[image]

前ゼッケン（Cockpit_front_frame_cloth）に関しては，番号を7セグのように押し出し対象を変更することで0から9までの数を表現できるようになっています．3D(ipt)データは，「２」を標準として一つのみ存在し，2D(DXF)データに関しては，Cockpit_front_frame_cloth_No1，Cockpit_front_frame_cloth_No2・・・Cockpit_front_frame_cloth_No7まで用意しています．

3D（ipt）データは各部位階層に．2D（DXF）データは下記の階層にまとめています．
https://github.com/citbrains/SUSTAINA-OP/blob/master/LaserCutter/Cockpit_back_frame_cloth.dxf

MasatoKubotera, 2022-06-27:
実際に製作したものは下図のようになりました．
[image]
[image]
[image]

前：Cockpit_front_frame_cloth，後：Cockpit_back_frame_clothに関してはCockpitにはめ込み，左右（共通）：ForeArm_outside_clothに関してはhttps://github.com/citbrains/SUSTAINA-OP/issues/33#issuecomment-1027283583 で提示した方法で固定しています．固定用の既製品に関しては，L3040-8110TBに変更れています．

MasatoKubotera, 2022-06-27:
再設計が終了しましたのでcloseします．

closed 2022-06-27 by MasatoKubotera