SUSTAINA-OP/SUSTAINA-OP #35: リンク_ふくらはぎ_サブフレーム_1_4の代替既製品を探す

MasatoKubotera, 2021-06-02:


hiroki-0001, 2021-06-05:
[image]

リンク_ふくらはぎ_サブフレーム_1_4は既製品で代用できそうなので代用できる既製品を探す

MasatoKubotera, 2021-07-08:
この部品は，1機体あたりには8個使用されており，サーボモータの出力/フリーホーンにつける リンクフレームの幅を決める部品であり，精度が要求される．
T5部材から，リンクフレームの幅のために4[mm]厚に加工しているが切削時にZ方向の精度を出すことが難しく既製品を導入したいと考える．
凸部は，リンクフレームに埋め込むためのものであるが，組み立て時に落ちにくいというメリットしかないようだ．

T4部材を使用することで精度問題は解決されるが，新しい種類の部材となるため管理が難しいという難点がある．
既製品が見つからない場合は，T4部材の導入の検討を行う．

[image]

MasatoKubotera, 2021-08-11:
[image]

上図のようにT4板を用いることで切削精度が必要ない形状に変更しました．

修正したCADはBranch：[Fix_Link_calf_subframe](https://github.com/citbrains/OpenPlatform_ver3_0/tree/Fix_Link_calf_subframe)にPushしました．

@NaokiIwasawa 確認次第統合お願いします．

MasatoKubotera, 2021-08-11:
ver4_0ブランチに統合されたためissueを閉じます．

yasuohayashibara, 2021-08-22:
指摘が遅くなって申し訳ありませんが，ラジアル方向の力をネジ部で支える構造となっています．
嵌合部分を無くすと，組み立て時の精度が低くなるのと，ラジアル方向の衝撃力がネジ部に集中的に加わるという問題が生じます．
どこかで話したような気もしますが，ネジに大きなラジアル方向の力が加わる設計はできるだけ避けたほうが良いです．
また，力を支える面積が減ると，その分応力が増えます．

加工が難しいということであれば，例えば図の用に太いピンを打ち込むという方法があります．
他の部品の内径をこれに合わせて変更することが前提です．

なお，中実だとサーボホーンのネジにアクセスできなくなりますので，中空パイプにしたほうが良いかもしれません．

[image]
※ネジの位置がずれていますが，ご容赦ください．

MasatoKubotera, 2021-09-02:
修正が完了したので閉じます

closed 2021-09-02 by MasatoKubotera