SUSTAINA-OP/SUSTAINA-OP #36: アルマイト加工する部品の選定

MasatoKubotera, 2021-06-02:


hiroki-0001, 2021-06-09:
レギュレーションで光を反射する素材を使用することはNG。
そのため、GankenKunの部品の中で試合中に目立ちそうな部分を選定し、アルマイト加工を行う。
今回のissueはそのための部品の選定を行う。

MasatoKubotera, 2021-06-18:
初心者向けでもこなせるタスクであると考える．
またルールを再確認し，どの程度までアルマイト加工する必要があるか議論する必要がある．

Kenzo-Fujisaki, 2021-06-26:
#　自分が今週やることリスト
- ルールを確認
-  自分が選定したパーツ

Kenzo-Fujisaki, 2021-07-02:
#### 自分が選定したパーツ
- 正面
    SETF5-63_5-M3-N3
   付け根ロール_サブフレーム_6_6
   肩_サブフレーム_1_1
   リンク_ふくらはぎ_サブフレーム1_1(予備案)
   付け根_サブフレーム_1_3
   足裏_メインフレーム_1_2
- 右から
   付け根ロール_メインフレーム_1_1
   膝メインフレーム
   足首_メインフレーム_1_1
   付け根_メインフレーム_2_1
   胴体_サブフレーム_1_3
- 左から
   付け根ロール_メインフレーム_1_1
   膝メインフレーム  
   足首_メインフレーム_1_1_MIR
   付け根_メインフレーム_2_1_MIR
   胴体_サブフレーム_1_3
- 後ろから
   肩_サブフレーム_1_1
   付け根ロール_サブフレーム_6_5
   リンク_ふくらはぎ_サブフレーム_1_2(予備案)
   足首_サブフレーム_1_1
   ASU-340
   足裏_メインフレーム_1_1

MasatoKubotera, 2021-07-02:
@Kenzo-Fujisaki 

パーツ選定ありがとうございます．
画像などを用いて説明することによってわかりやすくなるかと思います．
またInventorでは，パーツ１つ１つの色を自由に変えることができますので，ある程度は変更する前に完成予想をすることができます．難しければ，ペイントで描きこんでパーツの指摘をしても良いかと思います．

SETF5-63_5-M3-N3というパーツが指摘されていますが，つい火曜日に別のパーツに変更となりました．
CADは逐次更新されていきますので調査する前に最新版をダウンロードするようにお願いします．

Kenzo-Fujisaki, 2021-07-03:
#### 今週やること
- それぞれのパーツが分かりやすい図を作ってくる

Kenzo-Fujisaki, 2021-07-03:
#### 修正版
 - 正面
1. 肩_サブフレーム_1_1
2. 付け根ロール_サブフレーム_6_6
3. 付け根サブフレーム1_3
4. リンク_ふくらはぎ_サブフレーム1_1(予備案)
5. 足裏_メインフレーム_1_2

<img src ="https://user-images.githubusercontent.com/85381022/124347011-e4c14b00-dc1c-11eb-9646-66893d1c737f.jpg"
width="320px">

Kenzo-Fujisaki, 2021-07-03:
#### 修正版
- 横
1. 付け根_メインフレーム_2_1
2. 膝メインフレーム
3. 足首_メインフレーム_1_1

<img src ="https://user-images.githubusercontent.com/85381022/124347158-0bcc4c80-dc1e-11eb-98eb-0cfc1fb3bcb8.jpg"
width="320px">

Kenzo-Fujisaki, 2021-07-03:
#### 修正版
- 後ろ
1. 肩_サブフレーム_1_1
2. 付け根ロール_サブフレーム_6_5
3. 付け根_メインフレーム_6_1
4. リンク_ふくらはぎ_サブフレーム_1_2(予備案)
5. 足裏_メインフレーム_1_1
 
<img src ="https://user-images.githubusercontent.com/85381022/124347598-73839700-dc20-11eb-891d-0f7489f89331.jpg"
width="320px">

yasuohayashibara, 2021-07-03:
検討お疲れさま．

アルマイトをかけるのは，切削した板材のみにしたほうが良いです．
寸法変化もありますので，寸法がシビアなシャフトなどには使用しない方が良いです．
（あとシャフトは表面加工もしてあるおそれがあるので，アルマイトをかけられないかもしれません．）

よく考えると，板材も嵌め合いがシビアなので，試してから使用したほうが良いですね．

参考
https://bit.ly/3jKfSY9

MasatoKubotera, 2021-07-04:
@Kenzo-Fujisaki 
色・反射に関してのルールがどのようになっていたか共有していただけると嬉しいです．

MasatoKubotera, 2021-07-07:
@Kenzo-Fujisaki 
GankenKunのうち最古のバージョンである163機の肩部分のa2017 T5(アルミ 5mm厚)の部品の側面に黒いガムテープが張られていたので報告します．おそらく，反射対策課と思われます．ルールを確認し，どこまでが許容されるのか調査する必要があるかと思います．

[image]

yasuohayashibara, 2021-07-10:
黒いガムテープは画像処理の誤認識を防ぐためとなります．
こちらの方が先に議論されていたようですが，こちらにもう少し詳しい内容を書きました．

https://github.com/citbrains/OpenPlatform_ver3_0/issues/2#issuecomment-877522363

Kenzo-Fujisaki, 2021-07-10:
##### 今日のメモ
- 世界大会の色、反射に関してのルールを翻訳
- 改めて認識班の先輩方に怪しいパーツの相談をする必要がある

Kenzo-Fujisaki, 2021-07-18:
#### 進捗の報告が遅れてしまい申し訳ありません
#### 世界大会の色販社のルールに関しては以下になります
- バーチャル上
・ロボットの色はほとんどが黒か、ダークグレー（RAL 7011 アイアングレー以上）で、無反射でなければなりません。ロボットはアルミニウムのようなシルバー、グレー、ホワイトに着色してもよいが、その場合は足の色を黒にしなければならない。
フィールドに使われている色（緑、白）や、相手チームのチームマーカーと似た色は避けなければならない。ロボットの腕、足、胴体は立体的な形状でなければならない。
・ロボットにはチームマーカーを付けなければならない。このマーカーは、一方のチームには赤、もう一方のチームには青で表示されます。
チームマーカーは、一方のチームは赤、もう一方のチームは青である。ロボットの腕、脚、胸にあるすべてのチームマーカー（最大20個）の合計可視面積は
少なくとも0.06・Htopでなければならない。
・両側（左、右、前、後）にある1～5個の大きなチームマーカーの可視領域は、少なくとも0.015・Htopでなければならない。
・チームが使用する色はランダムに割り当てられ、ゲームプランで発表されます。
・各チームのロボットは、一意に識別できるものでなければならない。番号や名前が付いていなければなりません。ゴールキーパーロボットは、審判がチームの他のロボットと簡単に区別できるように、独自のマークを付けなければならない。
・2つのチームは互いに識別できる色を身につけなければなりません。
https://cdn.robocup.org/hl/wp/2021/06/V-HL21_Rules_v4.pdf

- 実機
・ロボットの色はほとんどが黒か、ダークグレー（RAL 7011 アイアングレー以上）で、無反射でなければなりません。ロボットはアルミニウムのようなシルバー、グレー、ホワイトに着色してもよいが、その場合は足の色を黒にしなければならない。
フィールドに使われている色（緑、白）や、相手チームのチームマーカーと似た色は避けなければならない。ロボットの腕、足、胴体は立体的な形状でなければならない。
・ロボットにはチームマーカーを付けなければならない。このマーカーは、一方のチームには赤、もう一方のチームには青で表示されます。
チームマーカーは、一方のチームは赤、もう一方のチームは青である。ロボットの腕、脚、胸にあるすべてのチームマーカー（最大20個）の合計可視面積は
少なくとも0.06・Htopでなければならない。
・両側（左、右、前、後）にある1～5個の大きなチームマーカーの可視領域は、少なくとも0.015・Htopでなければならない。
・両チームの意見が一致しない場合は、試合の1時間前にコインを使ってチームカラーを決定します。
・各チームのロボットは、一意に識別できるものでなければならない。番号や名前が付いていなければなりません。ゴールキーパーロボットは、審判がチームの他のロボットと簡単に区別できるように、独自のマークを付けなければならない。
・2つのチームは互いに識別できる色を身につけなければなりません。また主審と副審とも区別できる色でなければならない。
https://humanoid.robocup.org/wp-content/uploads/RCHL-2020-Rules-Dec23-changesMarked.pdf

Kenzo-Fujisaki, 2021-07-18:
#### 連続ですいません
#### 久保寺さんと神戸さんにお話が聞けたのでパーツを追加します 肩_メインフレーム_1_1となります
<img src="https://user-images.githubusercontent.com/85381022/126054793-7601561c-5867-4d92-85f7-7832874709fb.png"
width = 320px>

Kenzo-Fujisaki, 2021-07-19:
#### 部品の選定は終了したので調査は終了します

closed 2021-07-19 by Kenzo-Fujisaki