SUSTAINA-OP/SUSTAINA-OP #38: リンク筋変えの向きの決定

MasatoKubotera, 2021-06-02:


hiroki-0001, 2021-06-09:
[image]
筋交いの向きが現状統一されていないため、それらを統一する。過去に参考になりそうなissueがあればそれを持ってきて、向きを決定させる

aiba0921, 2021-09-01:
前回のメカ班ミーティングで出たリンク_ふくらはぎ_サブフレーム_1_2の「ネジ穴を両側で一つずつ減らして筋交いを中心を通るようにする」再設計をしたデータをドライブに載せておきました。再設計に伴って内側の長方形の形が若干変化して上下の幅が0.5mmずつ短くなりました。（一方、この幅を3mmに保とうとすると左右の幅が6mmになります。）
[image]
[image]
https://drive.google.com/file/d/10eYu1ZQxzWHd0h3VMgg5SwoRmp9FtFET/view?usp=sharing

MasatoKubotera, 2021-09-02:
上記で議論されている，ねじを片側3本から2本に減らすことについて，ver4_0ブランチに反映した．
この時に，上記で修正済みのリンク_ふくらはぎ_サブフレーム_1_2.iptが共有されていたが，縁の幅が狭くなるという指摘があり，構造解析による検証が行えず不確定要素であったため，
>筋交いを中心を通るようにする

については採用していない．これを行う場合には今後検証する必要がある．

[image]

aiba0921, 2021-10-02:
リンク筋交いの向きは組み立て時にケーブルを固定するため、リンク_ふくらはぎ_サブフレーム1_1とリンク_ふくらはぎ_サブフレーム1_2の筋交いが交差する向きにした方がいいかもしれない

closed 2022-02-07 by MasatoKubotera