SUSTAINA-OP/SUSTAINA-OP #39: MIRパーツの削減/調査

MasatoKubotera, 2021-06-02:


hiroki-0001, 2021-06-09:
現状MIRパーツは切削の面倒くささや管理のしづらさなどから、ある程度廃止できるところは廃止する必要がある。
現在使用されているMIRパーツの調査を行い、該当パーツが廃止対象になるか検討する。

hiroki-0001, 2021-08-11:
こちら、MIRパーツの削減まで行うとデータが錯綜して大変なことになるので、**このissueでは調査のみをお願いします**。

MiyawakiKaito, 2021-09-02:
<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

部位 | パーツ一覧 | 個数
-- | -- | --
胴体(コックピット) | コックピット_2_メインフレーム_1_MIR:1 | 1
胴体(コックピット) | コックピット_サブフレーム_1_5_MIR:1 | 1
胴体(コックピット) | コックピット_2_メインフレーム_2_MIR:1 | 1
胴体(コックピット) | Antenna_MIR:1 | 1
胴体 | 胴体_サブフレーム_1_1_MIR:1 | 1
胴体 | 胴体_サブフレーム_1_4_MIR:1 | 1
胴体 | RearBumper_Lower_MIR:1 | 1
胴体 | FreundBumper_stand_MIR:1 | 1
下半身(足首)(左足) | 足首_メインフレーム_1_1_MIR:1 | 1
下半身(付け根)(左足) | 付け根_メインフレーム_2_1_MIR:1 | 1
下半身(右足) | 下半身_6_MIR:1 | 1
下半身(右足) | 足裏_1_1_MIR:1 | 1
下半身(右足) | 膝_6_1_MIR:1 | 1
下半身(右足) | 付け根_1_MIR:1 | 1
下半身 (右足) | 付け根_メインフレーム_2_1_MIR:1 | 1
下半身(右足) | 足首_1_MIR:1 | 1
下半身(右足) | 足首_メインフレーム_1_1_MIR:1 | 1

</body>
</html>

MiyawakiKaito, 2021-09-02:
<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

部位 | パーツ一覧 | 個数
-- | -- | --
胴体(コックピット) | コックピット_2_メインフレーム_1_MIR:1 | 1
胴体(コックピット) | コックピット_サブフレーム_1_5_MIR:1 | 1
胴体(コックピット) | コックピット_2_メインフレーム_2_MIR:1 | 1
胴体(コックピット) | Antenna_MIR:1 | 1
胴体 | 胴体_サブフレーム_1_1_MIR:1 | 1
胴体 | 胴体_サブフレーム_1_4_MIR:1 | 1
胴体 | RearBumper_Lower_MIR:1 | 1
胴体 | FreundBumper_stand_MIR:1 | 1
下半身(足首)(左足) | 足首_メインフレーム_1_1_MIR:1 | 1
下半身(付け根)(左足) | 付け根_メインフレーム_2_1_MIR:1 | 1
下半身(右足) | 足裏_1_1_MIR:1 | 1
下半身(右足) | 膝_6_1_MIR:1 | 1
下半身(右足) | 付け根_1_MIR:1 | 1
下半身 (右足) | 付け根_メインフレーム_2_1_MIR:1 | 1
下半身(右足) | 足首_1_MIR:1 | 1
下半身(右足) | 足首_メインフレーム_1_1_MIR:1 | 1

</body>
</html>

MiyawakiKaito, 2021-09-02:
:1を入れてしまっていたため、削除して再投稿しました。
<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

部位 | パーツ一覧 | 個数
-- | -- | --
胴体(コックピット) | コックピット_2_メインフレーム_1_MIR | 1
胴体(コックピット) | コックピット_サブフレーム_1_5_MIR | 1
胴体(コックピット) | コックピット_2_メインフレーム_2_MIR | 1
胴体(コックピット) | Antenna_MIR | 1
胴体 | 胴体_サブフレーム_1_1_MIR | 1
胴体 | 胴体_サブフレーム_1_4_MIR | 1
胴体 | RearBumper_Lower_MIR | 1
胴体 | FreundBumper_stand_MIR | 1
下半身(足首)(左足) | 足首_メインフレーム_1_1_MIR | 1
下半身(付け根)(左足) | 付け根_メインフレーム_2_1_MIR | 1
下半身(右足) | 下半身_6_MIR | 1
下半身(右足) | 足裏_1_1_MIR | 1
下半身(右足) | 膝_6_1_MIR | 1
下半身(右足) | 付け根_1_MIR | 1
下半身 (右足) | 付け根_メインフレーム_2_1_MIR | 1
下半身(右足) | 足首_1_MIR | 1
下半身(右足) | 足首_メインフレーム_1_1_MIR | 1

</body>
</html>

MiyawakiKaito, 2021-09-03:
コックピットのパーツとアセンブリパーツが間違って入れてしまっていたので削除しました。

<html>
<body>
<google-sheets-html-origin><style type="text/css"></style>

部位 | パーツ一覧 | 個数
-- | -- | --
胴体 | 胴体_サブフレーム_1_1_MIR | 1
胴体 | 胴体_サブフレーム_1_4_MIR | 1
胴体 | RearBumper_Lower_MIR | 1
胴体 | FreundBumper_stand_MIR | 1
下半身(足首)(左足) | 足首_メインフレーム_1_1_MIR | 1
下半身(付け根)(左足) | 付け根_メインフレーム_2_1_MIR | 1
下半身 (右足) | 付け根_メインフレーム_2_1_MIR | 1
下半身(右足) | 足首_メインフレーム_1_1_MIR | 1

</body>
</html>

MiyawakiKaito, 2021-09-03:
調査の作業は終えたので、次は該当パーツが廃止対象になるか検討する。

MiyawakiKaito, 2021-09-07:
## 調査報告

## 胴体_サブフレーム_1_1_MIR
胴体_サブフレーム_1_1_MIRは、赤丸で囲んだ段差部分をなくし、埋め込みナットの穴をタップ加工に直すことによってMIRパーツでなくすことが可能。
[image][image]

## 胴体_サブフレーム_1_4_MIR
胴体_サブフレーム_1_4_MIRは、埋め込みナットの穴をタップ加工に直すことによってMIRパーツでなくすことが可能。
[image][image]

##  RearBumper_Lower_MIR
 RearBumper_Lower_MIRは、MIRパーツでなくすことは可能だが、優先度は低いと考える。
[image][image]

## FreundBumper_stand_MIR
FreundBumper_stand_MIRもRearBumper_Lower_MIRと同様に、MIRパーツでなくすことは可能だが、優先度は低いと考える。
<img src="https://user-images.githubusercontent.com/83056429/132314230-b6c7937a-0cc2-4cb6-b198-2571a3e79a5a.png)
" width=282>[image]

MasatoKubotera, 2021-09-07:
@MiyawakiKaito 
調査お疲れ様です．

- 胴体_サブフレーム_1_1_MIR
段差については，ver4_0新規コックピットでは不要にですので埋めることができます．
ナット部分ですが確かに削減することが出来ますが，後方に倒れたときの衝撃がもろにここに来るのでカレイナットに変更してもこのポケット加工については残していきたいと思います．

- 胴体_サブフレーム_1_4_MIR
この部分については，持ち手で後方への転倒同様に衝撃が加わる箇所のため，容易にポケット加工を廃止することが出来ません．

- RearBumper_Lower_MIR
- FreundBumper_stand_MIR

上記2つの3Dプリントパーツについては，優先度は低いですがMIRパーツをなくしたいと考えています．
理由としては，この部分がTPU素材で作成されているためです．3Dプリントパーツは切削部品以上に消耗があり，TPUでは曲がりやすいことが挙げられ力をかけ続けると簡単に変形してしまいます．また，こすれて内部が露わになっているものも現状存在します．交換時にMIRパーツの種別があると，管理，製作が大変になりますので余裕があればMIRパーツをなくしたいと考えます．

MiyawakiKaito, 2021-09-08:
今回のissue自体はMIRパーツの調査ということだったので、このissueはここでクローズさせていただきます。RearBumper_Lower_MIRとFreundBumper_stand_MIRのMIRパーツの削減については、私自身で時間があるときに挑戦してみたいと思います。

closed 2021-09-08 by MiyawakiKaito