SUSTAINA-OP/SUSTAINA-OP #41: 長穴減速ホーン バージョン違いによる減速比の差

MasatoKubotera, 2021-06-03:
162機に取り付けられている，ガラエポ **長穴減速_ホーン_1_2**が左右でバージョンが異なっていた．

[image]
[image]
[image]

2枚の長穴減速_ホーン_1_2を比較すると，軸間距離が異なり実測で1[mm]違った．

長穴減速機構の減速比(算出方法：<a href="https://qiita.com/Ninagawa_Izumi/items/0b1d607e41396d3fc307">長穴減速を解く</a>)が左右で異なっている状態であった．

MasatoKubotera, 2021-06-03:
@hiroki-0001 が左右の両方 167機切削時のあまりに交換した．

[image]

ロール軸の減速比が異なっていたため，歩行パラメータなどの調整が必要である．

kiyoshirou-kawanabe, 2021-06-09:
ブッシュはサイズがないため購入を諦め，スペーサとワッシャの組み合わせで代用しようと思います．
スペーサの内径は6，外形は6.8，ワッシャは内径6.8，外形は13以下で探します．

closed 2021-06-09 by kiyoshirou-kawanabe