SUSTAINA-OP/SUSTAINA-OP #45: スラストベアリングの改良

MasatoKubotera, 2021-06-05:


hiroki-0001, 2021-06-09:
現状の整備でスラストベアリング関連の悪い点の調査を行い、それの修正を行うことは難しいので、一個のissueとして取り扱う。このissueに調査・修正を記録を記入してください。

Keita1Mori, 2021-06-09:
このissue自分が担当したいと思います。

MasatoKubotera, 2021-06-10:
スラストベアリングを固定している，4本のスペーサーについての報告が記載されています．https://github.com/citbrains/OpenPlatform_ver3_0/issues/3#issuecomment-858475959

Keita1Mori, 2021-07-06:
# 調査
## 165機体の肩のスラストベアリング
### 解体前構造
 上　肩_メインフレーム_1_1
 ↓　SIM　厚さ：1mm
 ↓　スラストベアリング
 ↓　付け根ロール_サブフレーム_1_3
 ↓　スラストベアリング
 下　肩ピッチ_サブフレーム

[image]
[image]
[image]
[image]

---

### 全体の問題
- ネジを締めた状態で隙間があった
[image]
- グリスが塗られていなかった
- 定格トルクで締めると動かない
---

### 肩_メインフレーム_1_1
- 微妙な歪みあり
[image]

---

### 付け根ロール_サブフレーム_1_3
- inventor上と実際の厚さが違う
[image]

- 傷がついている
[image]

---

### 肩ピッチ_サブフレーム
- inventor上と実際の厚さが違う
[image]

- 胴体_サブフレーム_1_4とサーボモータを固定するためのネジM2.6の頭の部分と干渉していて傷ついている
[image]
[image]
[image]

- 六角スペーサー加工あり
[image]

MasatoKubotera, 2021-07-07:
スラストベアリングの付け根ロール_サブフレーム_1_3と胴体間の六角スペーサ(<a href="https://jp.misumi-ec.com/vona2/detail/110302672270/?HissuCode=SLCG5.5-6">SLCG5.5-6</a>)について

付け根ロール_サブフレーム_1_3のねじ締結部分の厚みが1.5[mm]，六角スペーサの雌ねじ長がMiSUMiからダウンロードしたCADより4.090[mm]であった．
そのため，最新機体でも長さ6[mm]ねじを用いていたが，適切ではないと判明した．
5[mm]もしくは4[mm]ねじを使用する必要がある．

または，中空のスペーサを用いて胴体側のフレームにねじを貫通してもよいかと考える．

[image]
[図 付け根ロール_サブフレーム_1_3]

Keita1Mori, 2021-07-13:
# 調査
## 163機体の右肩のスラストベアリング
### 解体前構造
165機体の左肩のスラストベアリングの時と変わらず
> 上　肩_メインフレーム_1_1
　↓　SIM 厚さ：１ｍｍ
　↓　スラストベアリング
　↓　付け根ロール_サブフレーム_1_3
　↓　スラストベアリング
　下　肩ピッチ_サブフレーム
---
### 全体の問題
- ネジを締めた状態で隙間があった
- スラストベアリングにグリスが塗られていなかった
---
### 肩_メインフレーム_1_1
- 歪みなし
- 傷あり
[image] 
- バリ取り不足で手を切る可能性あり

---
### 付け根ロール_サブフレーム_1_3
- 歪みなし
- inventor上と同じ厚さ
- 汚れあり
[image]

---
### 肩ピッチ_サブフレーム
- 傷なし
- バリ取り不足で手を切る可能性あり
- 六角スペーサ加工あり
[image]

---
### 解体後
- パーツクリーナで汚れを落とした
- バリ取り不足のところをバリ取りした

MasatoKubotera, 2021-08-17:
[スラストベアリングの六角スペーサーの既製品の加工について #127](https://github.com/citbrains/OpenPlatform_ver3_0/issues/127)でスラストベアリングについて問題提起がされています．

Keita1Mori, 2021-09-17:
# 報告
## 構成

[image]

赤く囲ったものは切削部品になります。
バツ印の物は既製品を使用します。
以前スラストベアリングだったパーツを[ワッシャー](https://jp.misumi-ec.com/vona2/detail/221000090802/?CategorySpec=00000228989%3a%3amig00000002502089%0900000229746%3a%3anvd00000000000001&list=PageCategory)に変えます。
次に問題になっていた付け根ロール_サブフレーム_1_3を二つに分け平たい四角の面を切削で製作し、円柱の部分を[金属ワッシャー](https://jp.misumi-ec.com/vona2/detail/110300231910/?PNSearch=FWSSS-D27-WVE23-T3.5&HissuCode=FWSSS-D27-WVE23-T3.5&searchFlow=suggest2products&Keyword=FWSSS-D27-WVE23-T3.5&list=SuggestPreview)にし、圧入する形になりました。

## アセンブリー

[image]

yasuohayashibara, 2021-09-18:
方向性は良いと思いますが，ここは一番困難な箇所となりますので，苦労を強いられるかもしれません．
おそらく１回で完成させるのは難しいと思って取り組んだ方が良いです．
試行錯誤（および経過観察）が必要になるおそれが高いです．

検討したほうが良い点を２点挙げます．（股ヨー軸として説明します．）

１）金属ワッシャ

> 平たい四角の面を切削で製作し、円柱の部分を金属ワッシャーにし、圧入する

平たい四角の板が薄いので，圧入には向かないと思います．加工が困難ということであれば，この際なくしても良いかもしれません．主なデメリットは，股ヨー軸のモータに関して

- ラジアル方向の力がモータ軸に直接加わる

です．シュート時などに発生する並進力がそのままモータ軸に加わります．無くしても回転力に関しては受けてくれます．あと，平たい四角の板の強度が下がるおそれはありますが，上下から機構が挟まれているため，それほど影響はなさそうな気がします．
次にメリットは，

- 軸のズレによりモータに加わる力が軽減される

があります．あまり，意識されていないかもしれませんが，モータとこちらの機構の回転軸を２つ直列につないでいるため，これが完全に直線状に配置されていないと大きなラジアル力がモータ出力に加わります．一般的にはこのような機構では，カップリングが使用されますが，この機構では，組み立てる人の技量を信頼してこのような機構が追加されていません．この関節は組み立てた人の技量で全く別物の関節になります．軸が完全に一致するところを見つけて固定することが必要になります．（これ伝わっていますかね？）
https://www.nbk1560.com/resources/coupling/article/couplicon-about/

締め込むと動かなくなるというのは，実は上記の要因もありそうな気がします．軸がずれたまま無理に動かすとモータの出力軸のベアリングが壊れます．壊れないまでもスムーズに動いてはくれなくなります．
精度高く作れるとラジアル方向の外力を受けてくれる部分となりますが，精度が低いと逆に動くたびにモータ軸にラジアル方向の力が加わりモータが急速に劣化します．

２）幅の調整
図の赤で示した幅がありますが，おそらくは値がばらつくおそれがあると思います．
御存知の通り，隙間があると歩行が不安定になり，きつすぎると股ヨー軸の動きがぎこちなくなります．
ネジを締めると動かなくなるということは，すでに体験した人は多いと思います．

現在はどちらかというと広めに製作して，シムを入れて調整していますが，下記の構造では薄いシムを入れづらいかもしれません．シムは金属の薄板ですので，押付力以外の横方向の力（こすりつけるような力）が加わると金属粉を出して徐々に変形していくおそれがあります．現在は，スラストベアリング用の金属の板がありますので，その下に入れることでこれをできるだけ回避しています．ただし，回避できているかは微妙ですが．．．

なお，ベアリングを樹脂にすることによりある程度の変形が期待されるので，このような調整をしなくても問題なく動く可能性があります．

全ては製作してみないと分からないというところはありますが，その分試行錯誤を強いられる部分だと思います．
製作や調整が難しい部分ですので，耐久性を犠牲にしてでも作りやすくしたほうが良いかもしれません．
ただし，あまり手を抜きすぎると足先がぐらぐらになり歩行できなくなりますので，良い妥協点を探ることが必要となります．

[image]

Keita1Mori, 2021-09-18:
# 報告
## 構成
切削面に既製品を使用することに問題点があるとのことだったので、変更しました。
以前提示したワッシャーの厚みを0.5mm削減しその代わりに[SIM](https://jp.misumi-ec.com/vona2/detail/110302677870/?list=PageCategory)を使用します。
変更後のワッシャーは[こちら](https://jp.misumi-ec.com/vona2/detail/221004981388/?HissuCode=DBB24W&PNSearch=DBB24W&KWSearch=DBB24W&searchFlow=results2products&list=PageSearchResult)を使用します。

# アセンブリー
## 断面図
[image]

## 胴体_メインフレーム_1_1との拘束
[image]

MasatoKubotera, 2021-09-18:
コアタイムで口頭でお伝えしてしまった部分をまとめます．

現状のスラストベアリングは，ポケット加工が含まれる3つの部品と公差が±0.1ある六角スペーサで構成されており調整がとても難しい状態となっています．ポケット加工が含まれるということは，それほど切削精度を出さなければならずXY平面上の傾斜がついてしまうこともあります．そのため，できる限りポケット加工が少ない部品設計がこの部分では求められると思います．切削精度が出せれば問題視する点ではありませんが，現状の技術，環境的にはかなり厳しいところがあります．

切削面に既製品を使用することが問題ではなく，ポケット加工をした部分にオイレス部品を装着することが問題視されます．これは，ポケット加工のバリがオイレスワッシャの寿命低下につながる恐れがあるためできるだけ避けたいということです．この部分については，切削した部品とオイレスワッシャ間に何か挟むことで解決することが出来ます．

MasatoKubotera, 2021-09-18:
タップ追加工を行っている箇所があると思いますが，そのタップ加工は経験によって大きく精度が異なってきます．タップ後のバリ取りなども行わなければならず，加工方法にも注意が必要な個所です．

MasatoKubotera, 2021-09-21:
昨日の議論で出たクロスローラベアリングですが，導入するとなると取付穴があるものを用いることで部品点数を減らし，精度が必要ない加工方法で脚ピッチ軸を製作できます．

候補のクロスローラベアリングは，[CRBTF305AT](https://www.mekasys.jp/series/detail/id/IKO_0218/)です．
[カタログ](https://www.ikont.co.jp/product/newproduct/pdf/CRBTF.pdf)

MasatoKubotera, 2021-09-23:
# スラストニードルベアリングの廃止

上記で議論が行われていますが，スラストニードルベアリングを取り入れた設計が難航し，どうしても切削・製作精度に頼る設計になってしまう．切削精度を出すことは現状の技術量・環境では限界があり，現状切削後の組み立ての際に調整する職人がいなければならず，今後のロボット製作におて大きな障害となっている．

現在使用している，胴体-脚の付け根と胴体-肩間に計4か所で採用されているが，この部分については廃止したいと考える．

# クロスローラベアリングの新規導入検討

[image]

## 導入検討への経緯
スラストニードルベアリングを廃止するとして，候補として真先に上がるのはオイレスワッシャによる滑り軸受けであるが，これは，スラストニードルベアリングと同様の形状であるため，上記で述べるように設計が困難である．

この場所については，[脚のロール軸方向のガタの解明と対策 #1140](https://github.com/citbrains/citbrains_humanoid/issues/1140)でも議論されているようにガタを生じさせている要因の一つでもある．

## 利点
クロスローラベアリングの利点としては下記がある．
>クロスローラーリングはあらゆる方向の荷重を負荷できる、回転精度が高いローラーベアリングです。
円筒ころが直交配列されているため、あらゆる方向の荷重を負荷することができます。

参照：[クロスローラーリング/複列アンギュラローラーリング](https://www.thk.com/?q=jp/node/6727)

## 選定
クロスローラベアリングを選定するにあたって，取付穴が付属しているものを第一に選定する．理由として，ベアリングホルダを作る必要がなく設計が容易であることが挙げられる．
欠点としては，穴付きクロスベアリングの種類が少なく，穴なしに比べ高価であることである．

上記のことを踏まえて選定したクロスローラベアリングは，[CRBTF105AT](https://www.mekasys.jp/series/detail/id/IKO_0218/)である．寸法詳細の[カタログ](https://www.ikont.co.jp/product/newproduct/pdf/CRBTF.pdf)

外形43mm 内径10mm 厚み5mmであり，穴付きのクロスローラベアリングとしては小型の部類である．

[image]

欠点として，下記の要項が挙げられる．
- 取付穴が，内輪：M2.5，外輪：φ2.9であり特殊である．
- 日本版MiSUMiでの取り扱いがなく，価格帯が確定できない．
  - 米版MiSUMiでの取り扱いは確認できた．[CRBTF105ATC1](https://us.misumi-ec.com/vona2/detail/221303549852/?HissuCode=CRBTF105ATC1)

## 設計

CADは，ブランチ「[cross_bearings](https://github.com/citbrains/OpenPlatform_ver3_0/tree/cross_bearings)」で確認することが出来る．

設計するにあたって，下記のことに注意している．
- ポケット加工など切削Z精度を必要としない設計．
- スラストベアリングのように専門知識を持たずとも誰でも簡単に組み立てすることが出来る．

### 使用部品
片脚分の使用部品を下記に示す．

|名称|型番|数量|価格(税込)|
|---|---|---|---|
|クロスローラベアリング |[CRBTF105AT](https://www.mekasys.jp/series/detail/id/IKO_0218/)|1||
|オイレスワッシャ|[83W-21](https://jp.misumi-ec.com/vona2/detail/221000090813/?PNSearch=83W-21&HissuCode=83W-21)|1|￥53|
|ユニファイNo.4-40六角亜案付きボルト|[SNSS-#4-40X3/16](https://jp.misumi-ec.com/vona2/detail/221000233238/?PNSearch=SNSS-%234-40X3%2f16&HissuCode=SNSS-%234-40X3%2f16&searchFlow=suggest2products&Keyword=SNSS-%234-40X3%2f16&list=SuggestPreview)|6|￥389|
|ばね座金・平座金組込六角穴付ボルトM2.5-10|[CSHHNDI3TM-SUS-M2.5-10](https://jp.misumi-ec.com/vona2/detail/221005020574/?HissuCode=CSHHNDI3TM-SUS-M2.5-10)|6|￥186|

M2.5ねじについては，ベアリングにねじ穴加工が施されてており，変更の余地はない．M2.6ねじをサーボモータ周りでは使用しているが，頭部高さを大きく異なることで区別をつけるようにしたいと考えている．また，この部分については高価なベアリングにねじ材を使用するべきではないと判断し，ばね座金・平座金組込六角穴付ボルトを採用している．

インチねじについては，取付ガバ穴φ2.9に合うねじを選定することで生じた．No.4-40のネジ径は約2.8mmである．このネジについては，メートル並目でも[ストリッパ・リーマ・ショルダーボルト](https://jp.misumi-ec.com/vona2/mech_screw/M3301000000/M3301030000/)などを検討したが採用できるものが見つからなかった．
本格的にこのベアリングを採用するのであれば，下図のようなねじを浅井製作所に製作を依頼しても良いかと考える．

### 必要な工具
インチネジ[SNSS-#4-40X3/16](https://jp.misumi-ec.com/vona2/detail/221000233238/?PNSearch=SNSS-%234-40X3%2f16&HissuCode=SNSS-%234-40X3%2f16&searchFlow=suggest2products&Keyword=SNSS-%234-40X3%2f16&list=SuggestPreview)を導入するにあたって下記の工具も必要である．

|名称|型番|数量|価格(税込)|
|---|---|---|---|
|3/32インチ面幅 六角ビット|[BH3/32](https://jp.misumi-ec.com/vona2/detail/223012337531/?PNSearch=BH3%2f32&HissuCode=BH3%2f32&searchFlow=suggest2products&Keyword=BH3%2f32&list=SuggestPreview)|1|￥1,126|
|No.4-40 ハンドタップ|[IS-S-HT-NO.4NC40-3](https://jp.misumi-ec.com/vona2/detail/223009428213/?PNSearch=IS-S-HT-NO.4NC40-3&HissuCode=IS-S-HT-NO.4NC40-3&searchFlow=suggest2products&Keyword=IS-S-HT-NO.4NC40-3&list=SuggestPreview)|1|￥688|

---

[image]「断面図(横から)」

[image]「断面図(前方から)」

[image]「横から」

[image]「斜め後方から」

[image]「前方から」

[image]「下方から」

[image]「クロスローラベアリングの取付」

[image]「ベアリングとサーボの接続」

[image]「オイレスワッシャ」

[image]「オイレスワッシャと胴体の隙間を埋める部品」

[image]「新規ベアリングに合わせた胴体フレーム」

MasatoKubotera, 2021-09-23:
## 肩スラストベアリングの代替

肩のスラストベアリングについても廃止したいと考えるが，上記で提案したクロスローラベアリングは高価であり腕に使用することは難しいため，技術・コストを第一に考えた下記の案を提案する．

- ベアリングを装着せずに関節（部品）を取り付け，肩バンパーなどによる転倒時の衝撃緩和

yasuohayashibara, 2021-09-24:
林原が理解できていないおそれが高いので直接聞いたほうが良さそうな気がしますが，クロスローラベアリングが腰のパーツに固定されているように見えません．

クロスローラベアリングの外側は腰のパーツに固定する必要があります．

あと，HIWIN社のCRBDの方がおそらく安価だと思います．CRBAはミスミにもありますが，CRBDは見積もりを取らないと価格は分かりません．それほど品質は悪くなさそうです．
https://premium.ipros.jp/hiwin/product/detail/2000523991/

MasatoKubotera, 2021-09-24:
動力が伝わってないとんでもない設計でした．修正します．

MasatoKubotera, 2021-09-24:
フランジ穴付き外輪分割型CRBDシリーズについては，最小サイズで内径20mm，外形70mm，幅12mmのCRBD02012AでGankenKunに対して大きいと考えます．

そのため，製作の手間は増えますが，穴がないCRBAシリーズを採用しようかと思います．
候補として，	[CRBA02508WWC8P5](https://jp.misumi-ec.com/vona2/detail/221006309957/?PNSearch=CRBA02508WWC8P5&HissuCode=CRBA02508WWC8P5&searchFlow=suggest2products&Keyword=CRBA02508WWC8P5&list=SuggestPreview)　￥7,817を考えています．

yasuohayashibara, 2021-09-25:
すみません．手間が増えるのであれば，HIWINを選択しないほうが良いと思います．
私も少し調査してみます．

MasatoKubotera, 2021-09-25:
手間が増えることが装うされますが，見積を取るまで価格が不明である点，MiSUMiでの取り扱いがないベアリングを使用することで開発に遅れが出ると思い，取付穴がないクロスローラベアリング [CRBA02508WWC8P5](https://jp.misumi-ec.com/vona2/detail/221006309957/?HissuCode=CRBA02508WWC8P5) ミスミ価格：￥7,817/個を用いて設計しました．

>林原が理解できていないおそれが高いので直接聞いたほうが良さそうな気がしますが，クロスローラベアリングが腰のパーツに固定されているように見えません．
クロスローラベアリングの外側は腰のパーツに固定する必要があります．

の指摘部分については，自分の勉強不足でした．今回のモデルは腰のパーツに固定しております．

取付穴があるクロスローラベアリングに比べ，切削部品点数，既製品点数も増えますが，以前自分が検討していた[CRBTF105AT](https://www.mekasys.jp/series/detail/id/IKO_0218/)とは大きく異なり，インチネジ・インチタップ，M2.5ネジを使用していないことを踏まえると，取付穴にこだわらなくても良いかと思いました．（メートル並目に合う取付穴付きのものがあればなお良いですが．）

必要な切削精度については，スラストベアリングベアリングの時のような厳密さはいらない設計かと思いますが，-0.0~-0.1の精度は必要かと思います．

部材についてはA2017 T4，T1.5を新たに用いる設計となっています．

ベアリング厚がT8のため，厚い部材を用いる必要がありますが，切削時間の増加・部材価格によりスペーサにて代用しています．このスペーサについては，すでに脚のロール軸やピッチ軸に使われているものと同じ型番のものを用いることで１機体あたりのコストを抑えています．

また，クロスローラベアリングを胴体に固定する必要がありますが，アルミにタップすることを避けるためにカレイナットを用いています．このカレイナットについても，すでにほかの部位で使用予定のものを用いているためコストを抑えた設計になっています．

モデルについては，ブランチ[cross_bearings](https://github.com/citbrains/OpenPlatform_ver3_0/tree/cross_bearings)にPushしています．

価格の比較（部材を抜いた単純計算）
<html>
<body>

Ver3.0 |   |  
-- | -- | --
価格 | 数量 | 価格
MSU-306-03 | 8 | ¥554
BA2542 | 1 | ¥429
合計 |   | ¥983

</body>
</html>

<html>
<body>

今回の設計 |   |  
-- | -- | --
型番 | 数量 | 価格
BRU-307S | 8 | ¥748
BRU-311S | 4 | ¥814
S3-07-3W | 8 | ¥192
SF319 | 4 | ¥408
CRBA02508WWC8P5 | 1 | ¥7,817
合計  |   | ¥9,979

</body>
</html>

[image]「側面」
[image]「側面断面図」
[image]「前方45度断面図」
[image]「下方」

MasatoKubotera, 2021-09-25:
上記でBRU-311Sを使っている部分について.サーボホーンのネジ径がM2.6でしたのでBRU-2611Sに変更になります．
後日CADは修正しますが，構造的に大きな変化はありません．

NaokiIwasawa, 2021-09-26:
上記の変更確認しました
短時間で再設計いただきありがとうございます。かなり複雑ですので設計が大変だったと思いますが自分は大きな問題はないように思います。

但し、少し気になった点をあげます（適切か自信はないです）。

1. 青マーク：このカレイナットの向きではキック時などに外れやすい気がします
スペーサーとOuter_support_standのポケットを調整するかTrunk_horizontal_frame_01に上面からポケットを加えカレイナットを上から完全に埋めるなど上面から圧入できるようにした方がいいのかなと思います
2. 紫マーク：何か所か壁の厚さが薄すぎる気がします。（計測が正しければ最小距離0.19mm）
3. ネジの頭高さが1.1mmに再製作で変わるので多少余裕が出来そうです！
 

 
[image]

MasatoKubotera, 2021-09-26:
@NaokiIwasawa 

>1. 青マーク：このカレイナットの向きではキック時などに外れやすい気がします
スペーサーとOuter_support_standのポケットを調整するかTrunk_horizontal_frame_01に上面からポケットを加えカレイナットを上から完全に埋めるなど上面から圧入できるようにした方がいいのかなと思います

この部分については，下図の青色で選択されたパーツをはめ合わせていることにより，下からの圧入しかできない状況となっています．他部分については上方からの圧入もできますが，この部分に合わせる設計となっています．
[image]
[image]

また，スペーサのおねじ長が6mmであるため，上方から圧入してしまうと雄ねじが届きません．
下図の青色に選択された部品にすでに1mmポケット加工されていますが，上方からの圧入の場合さらに+2mmポケット加工とスペーサ長を延長しなければならず，加工時間の増大と部材の残り厚2mmで強度が保てるかという懸念があります．（構造解析すればよいことですが．．．）
[image]

>2. 紫マーク：何か所か壁の厚さが薄すぎる気がします。（計測が正しければ最小距離0.19mm）

この部分については，かなり薄くなっていますが，ベアリングの外周スペーサ８本をできるだけベアリングから離さないような設計を優先としていました．スペーサ間隔を広くすることでカレイナットの設計規定より，部材端あで5mm必要なため，相応の重量の増加が懸念されます．
上記を踏まえてサーボ締結ネジが干渉しない最大限まで厚く変更しました．

[image]「変更前」

[image]「変更後」

[image]「変更前 Outer_flange.ipt」

[image]「変更後 Outer_flange.ipt」

[image]「変更前 Outer_support_stand.ipt」

[image]「変更後 Outer_support_stand.ipt」

>3. ネジの頭高さが1.1mmに再製作で変わるので多少余裕が出来そうです！

現状0.1mmの余裕しか確保出来ていなかったので良かったです．
納品確定次第，ねじCADモデルの方も変更できればと思います．

---

自分自身で指摘した
>上記でBRU-311Sを使っている部分について.サーボホーンのネジ径がM2.6でしたのでBRU-2611Sに変更になります．
後日CADは修正しますが，構造的に大きな変化はありません．

についても修正しました．

最新版も同ブランチにpushしました．

MasatoKubotera, 2021-09-26:
カタログの指示台フランジで指定されている幅を残し軽量化しました．
Outer_support_stand.ipt・Outer_flange.ipt合わせて9.562gの軽量化となっています．
[image]「軽量化前」
[image]「軽量化後」

MasatoKubotera, 2021-09-26:
肩のスラストベアリングについては廃止ししました．

[image]「左：スラストベアリングなし，右：スラストベアリングあり」

MasatoKubotera, 2021-09-26:
Trunk_vertical_frame_04とそのMIRについてスラストベアリングの取付穴が不要となったので廃止した．
[image]

yasuohayashibara, 2021-09-27:
スラストベアリングを無くすのは良いと思いますが，片持ち梁でラジアル方向の力をそのまま受けると故障しますので，以下のようなワッシャを入れたほうが良いと思います．
過去のロボットにもほとんど入っていると思います．

https://www.vstone.co.jp/robotshop/images/4562179391090.jpg

MasatoKubotera, 2021-09-27:
面受けのワッシャついてはテフロンシートが多く用いられるようです．

参考書籍 [ROBO-ONEのための二足ロボット製作ガイド](https://www.amazon.co.jp/ROBO%E2%80%90ONE%E3%81%AE%E3%81%9F%E3%82%81%E3%81%AE%E4%BA%8C%E8%B6%B3%E6%AD%A9%E8%A1%8C%E3%83%AD%E3%83%9C%E3%83%83%E3%83%88%E8%A3%BD%E4%BD%9C%E3%82%AC%E3%82%A4%E3%83%89-RoboBooks-ROBO%E2%80%90ONE%E5%A7%94%E5%93%A1%E4%BC%9A/dp/4274087492)

MasatoKubotera, 2021-09-28:
ご指摘いただいたラジアル方向の荷重による対策として，PTFEワッシャ [TT-2532-10](https://www.hirosugi-net.co.jp/shop/g/g32088/)：￥3,168/20個を導入しました．

[image]
※赤色の部品がPTFEワッシャ TT-2532-10

同サイズのジュラコン（POM）製[CC-2532-10B](https://www.hirosugi-net.co.jp/shop/g/g31554/)：￥1,188/20個も存在します．

PTFE製を採用した理由として，POMよりも低摩擦係数であることが挙げられます．
動摩擦係数：PTFE 0.10，POM 0.18

参考：
- https://wilco.jp/docs/technical-data/material-commentary/ptfe.html
- https://www.jushiplastic.com/wear-resistance

---

ワッシャのずれ防止のために肩部品にポケット加工を追加しました．
[image]

受けの胴体側ネジの凸をなくすために部材厚をT3からT4に変更しました．
[image]

これにより，ハンドル用のナット部分まで埋め込むカレイナットを採用するとポケット加工が両面になってしまうため，六角スペーサ軸側を埋め込む形に変更しました．スペーサもオスメスからメスメスに変更となり，両側からねじ挿入が出来る長さになっております．

[image]

MasatoKubotera, 2021-09-28:
過去の報告で首サーボにおいて故障が確認されておりません．
転倒時の胸バンパーの役割が大きいかと思います．

首にラジアル荷重対策のワッシャを入れるとなると，サーボホーンに取り付けられたPOM製部品とサーボ締結用のねじの凸をなくさなければならず，変更箇所も多く，対処として部材厚が増すことでさらなる増量が考えられます．

[image]

そのため，この部分については今後の課題としたいと考えております．

MasatoKubotera, 2021-09-29:
下図で青色で選択された部品について当初A2017 T1.5部材で切削予定だったが，T1.5がこの場所にしか使用されていないためT2に変更した．

[image]

yasuohayashibara, 2021-09-29:
良い設計だと思います．
以下の点が気になりますので，ご検討ください．

[image]

- カレイナットの向き
岩澤君からも指摘がありましたが，カレイナットの向きが気になります．
この向きで使用することができると記載されていれば大丈夫ですが，タップでも良いかもしれません．

- 噛み合う長さが短い
M3とすると2mm以上必要ですが，この場合先端を使用していますので2mmよりも短くなるかもしれません．
撃力を受けるところですので，長くしたほうが良いかもしれません．

ねじのはめあい長さについて
https://bit.ly/3iiGMEF

- 並進力がサーボモータに伝わる
図の矢印の並進力が加わると，ベアリングが力を支えますが，摩擦締結になっていますので，静止摩擦力以上の力が加わるとモータに力が伝わります．
嵌め合いにしたほうが良いと思われます．

- モータとベアリングの中心軸を合わせる仕組み
この機構で一番難しいのはモータとベアリングの中心軸を微調節して合わせることだと思います．
モータの回転軸を精度高く腰に固定することはおそらく難しく，さらにベアリングを取り付けると移動させることができなくなります．
（モータの穴の議論がありましたが，それを見てもあまり精度を期待しない方が良いと思います．）
それを考えると，以下の手順で微調整することが必要だと思われます．

モータの固定　→　モータの出力軸とベアリングの内輪の固定（中心軸を合わせて）　→　ベアリングの位置の調整　→　ベアリングの外輪と腰部品の固定（位置を調整して）

案外軸がずれていても問題なく動作する可能性はありますが，問題になる可能性も高いと思います．

芯ずれに関して
https://www.tts-inspection.com/ja/learn-maintenance/1505maintenance_32j/

MasatoKubotera, 2021-09-30:
ご指摘ありがとうございます。

>カレイナットの向き
岩澤君からも指摘がありましたが，カレイナットの向きが気になります．
この向きで使用することができると記載されていれば大丈夫ですが，タップでも良いかもしれません．

カレイナットについては，[SF319](https://jp.misumi-ec.com/vona2/detail/221000555673/?HissuCode=SF319&PNSearch=SF319&KWSearch=SF319&searchFlow=results2products&list=PageSearchResult)を用いることを考えていました．
こちらの商品ページ中に「引張強度」「押し込み強度」が記載されており，どちらの向きでも使用できるようです．
※六角沈込みで空転防止　溝で抜け防止空転とも記載あり．

>噛み合う長さが短い
M3とすると2mm以上必要ですが，この場合先端を使用していますので2mmよりも短くなるかもしれません．
撃力を受けるところですので，長くしたほうが良いかもしれません．
ねじのはめあい長さについて
https://bit.ly/3iiGMEF

ご指摘の通り，カレイナットのネジ溝高が1.9mmであるため，2mmよりも短くなっております．
この部分について2mm以上ねじを挿入すると胴体に垂直に立つA2017 T5部材のはめあいと干渉するために長くできない状況でした．
ねじのはめあい長さについてhttps://bit.ly/3iiGMEF の記事も踏まえておねじ挿入長を2,0mmから3.5mmに変更し，加えてカレイナットを廃止し，タップに変更しました．
カレイナットを上面から挿入することは，ちょうどよいカレイナット[S3-15-3W](https://jp.misumi-ec.com/vona2/detail/221000555662/?PNSearch=S3-15-3W&HissuCode=S3-15-3W&searchFlow=suggest2products&Keyword=S3-15-3W&list=SuggestPreview)がありましたが，全体高が3.6mmであり圧入する関係で上面と合わせなくてはならず下面から0.4mm空洞になってしまい，ねじ締結部が3.5mm-0.4mm=2.9mmとまたねじ締結部が短くなることがあり導入を断念しました．また，導入することでさらなる既製品種類の増加にもなり悩みどころです．

>並進力がサーボモータに伝わる
図の矢印の並進力が加わると，ベアリングが力を支えますが，摩擦締結になっていますので，静止摩擦力以上の力が加わるとモータに力が伝わります．
嵌め合いにしたほうが良いと思われます．

こちらについて，[カタログ](https://www.hiwin.tw/download/tech_doc/br/Crossed_Roller_Bearings-(J).pdf)で指示がありましたが反映できていなかったため，修正しました．

>ねじの頭高さについて

ご指摘いただいた箇所以外にも，ねじ頭と部材間が0.1mmである個所が多々あるため修正します．
浅井製作所の[ページ](http://nejikouba.com/img/susroku.htm)を見ると，ねじ頭高の公差は+-0.1mmのようですのでこちらを参考に修正していきます．
※頭径，ねじ部長についてはマイナス公差であるようです

---
ご指摘をもとに修正しました．
モデルについては，ブランチ「[cross_bearings](https://github.com/citbrains/OpenPlatform_ver3_0/tree/cross_bearings)」で保管しています．

[image]「クロスベアリング周りの断面図」

[image]「胴体フレーム」

MasatoKubotera, 2021-10-12:
> 米田先生が相談に乗ってくれるそうです． こちらの都合で申し訳ありませんが，10/14(木) 11:00~米田先生の自室で相談する予定です． こちらに関連する方で，参加できる方はご参加下さい． なお，あまり多人数で行くと密になりますので，林原＋２名ぐらいが限界だと思います．

対応ありがとうございます．
岩澤と久保寺で参加させていただきます．

できるだけ早く，自分が考察した組み立て手順についてissueを更新します．
よろしくお願いします．
---
@NaokiIwasawa

NaokiIwasawa, 2021-10-17:
[image]
[image]
指摘の部分についてまだ切削を行っていないので変更の問題はないです。（10/18に該当のInner_flange.iptを含むT2の残りを切削予定）
嵌め合いの追加部分については青部分に赤同様のポケット加工を追加する方針でしょうか？

MasatoKubotera, 2021-10-17:
>本日考えていてふと気づいたのですが，以下のハイライトした部品ですが，できればこちらもクロスローラベアリングの内側にはめ合う部分があった方が良いと思います．
というのは，ここで嵌め合いを作らないと，モータの軸とクロスローラベアリングの軸が一致しないおそれがあります．
例えば六角スペーサが真っ直ぐに立てられていないとすると，サーボの出力の部分でサーボモータの中心とベアリングの中心がずれてしまい，回転するたびに以下の部品とベアリングが擦り合わされることになるおそれがあるかなと思います．
気にし過ぎかもしれませんが，念のためお伝えしておきます．
すでに切削済みの場合は，しばらく動かしてチェックするで良いと思います．[image]

指摘いただいたInner_flangeに嵌め合いを設けなかった理由ですが、[カタログ](https://www.hiwin.tw/download/tech_doc/br/Crossed_Roller_Bearings-(J).pdf)の「支持台とフランジの固定方法と設計」において、フランジが嵌め合いを含み設計されていなかったことが挙げられますが、カタログと違い部材圧をT5よりも厚くできないことからスペーサを導入しており、垂直に立てられないことも考えられます。

しかし、これについては岩澤君が上記でも提案してくれていますがはめあいを設けるとなると、A2017 T2からさらに厚みを増し、T3以上にする必要があります。この時に、スペーサのおねじが5mmであるがために、サーボホーンに挿入する長さを確保するためには、Inner_flangeにスペーサを食い込ませるポケット加工が必要となります。
現状の切削技術では、ポケット加工の精度が保証されていないために水平に切削できないく、
切削により、さらに
>六角スペーサが真っ直ぐに立てられていない

という問題が生じます。そのためこの部分については、ポケット加工を必要としない嵌め合いのないフランジタイプでよいかと自分的に考えます。

また、水平がとれたとしてもポケット加工のZ軸高さ精度により、外輪の嵌めあい（Outer_support_stand）とも関係し、スラストベアリング以上の精度調節が必要となります。板厚公差も関係しますが、現状板厚の公差まで気にするとその他ほかの部分も考えなければいけないため、無視して考えています。

[image]

MasatoKubotera, 2021-10-20:
>モータとベアリングの中心軸を合わせる仕組み

について米田先生に相談し、出力軸側のサーボモータ締結については、クロスローラベアリング装着後にアプローチ出来ない出力軸に近い２点についは、締結しない方針となった。

理由として、クロスローラベアリングの支持台やフランジなどですでに胴体に固定されているため、過剰にサーボモータを締結する必要はなく、サーボモータが回転しない程度に固定するだけで十分である。

[image]

サーボモータと締結するためのねじ穴については、モータとベアリングの中心軸を合わせるために余裕を持った大きさとしている。締結にはM2.6ネジを使用するが、ガバ穴径はφ3.0としている。（ネジ頭径：φ4.8）

[image]
[image]

下図POM T5 のカバー（Trunk_upper_waist_frame）のサーボモータ嵌め合い部分についても余裕を持った寸法にしているが、これらの余裕については今後検証が必要であると考える。

[image]

MasatoKubotera, 2021-11-07:
改良が終了したためcloseします．

closed 2021-11-07 by MasatoKubotera