SUSTAINA-OP/SUSTAINA-OP #49: 適切なネジの長さを選定する

MasatoKubotera, 2021-06-10:
適切なネジの長さを選定し，inventorのネジ深さを変更する．

MasatoKubotera, 2021-06-14:
inventorモデルにネジを付け加えるかも検討する．

MasatoKubotera, 2021-06-18:
<a href="https://www.inventor-navi.net/inventor-how-to-use/1769/">詳細レベルリプレゼンテーション</a>を活用することで，ネジを追加したアセンブリも容易に開くことが出来るのではないだろうか．

hiroki-0001, 2021-06-18:
現状のGankenKunのモデルはネジ周りの設定はおかしくなっているため、適切なネジの長さを決めたあと、CADに反映させる。

MasatoKubotera, 2021-06-19:
上腕_メインフレーム_1_1に締結するねじは，現状ではM3x6mm，4mmの組み合わせである必要がある．
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/3#issuecomment-864358329

MasatoKubotera, 2021-07-07:
スラストベアリングにおいて改良が施されない場合，M3×4mmを使用しなければならない箇所がある．
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/45#issuecomment-875466145

また，短い六角スペーサについては，雌ねじ部分の長さと非締結部品の関係を調査する必要がある．

MasatoKubotera, 2021-08-15:
[ねじの軽量化 #100](https://github.com/citbrains/OpenPlatform_ver3_0/issues/100)で提言されていた，ネジの本数による軽量化についてもこのissueで取り扱いたいと思います．

MasatoKubotera, 2021-09-24:
同様なissue[CADモデルにねじを追加する #159](https://github.com/citbrains/OpenPlatform_ver3_0/issues/159)が存在するため，こちらのissueをcloseします．

closed 2021-09-24 by MasatoKubotera