SUSTAINA-OP/SUSTAINA-OP #50: 軽量化の検討

MasatoKubotera, 2021-06-10:


hiroki-0001, 2021-06-18:
現状ではGankenKunの重量は重くなる一方なので、軽量化のための設計を行いたい。

hiroki-0001, 2021-06-18:
軽量化のための余地があった場合にはこのissueに書き込んでください。

yasuohayashibara, 2021-06-28:
- 肉抜きの量を限界まで増やす．
ついでにワイヤーで切れる構造にすると開発が楽になります．
- メカニズムをシンプルにできないか検討する．
強度は下げずに
メンテナンス性も考えて
- 金属材料を樹脂にできないかを検討する．
切削時間も減らせるので
- 消費電流を減らしてバッテリのサイズを下げる
利用時間を短くすることは戦略の開発を妨げるので，現状の利用時間を変えない方が良いです．
理想は１時間ぐらい使えたほうが良いです．
（CPUの消費電力を限界まで下げると可能ではないでしょうか？）
- ネジの本数を減らす
実はネジがSUSなので重いです．
嵌め合いでずれないように固定できれば，あとは**ちょっとだけ押さえつける**だけでよくなります．
本来のネジの使い方です．

MasatoKubotera, 2021-07-08:
関連性があるissue：
[ねじの軽量化 #100](https://github.com/citbrains/OpenPlatform_ver3_0/issues/100)
[付け根ユニットの再設計 #103](https://github.com/citbrains/OpenPlatform_ver3_0/issues/103)
[リンク筋交いの使用材料について #79](https://github.com/citbrains/OpenPlatform_ver3_0/issues/79)
[カメラUSBケーブルの変更 #104](https://github.com/citbrains/OpenPlatform_ver3_0/issues/104)

MasatoKubotera, 2021-08-15:
現状必要な部分でしか低頭ねじを用いていなかったがすべての個所で低頭ねじを使う方針となった．
参考：[ねじの軽量化 #100](https://github.com/citbrains/OpenPlatform_ver3_0/issues/100)
そのため，ねじによる軽量化の影響は大きいと考える．

また，ねじの本数を減らすことに関しては，適切な長さを選定する際に行いたいと考えている．
参考：[適切なネジを選定する #49](https://github.com/citbrains/OpenPlatform_ver3_0/issues/49)

MasatoKubotera, 2021-09-14:
上記で告知しているように，ネジによる軽量化が行われることとなったため，この提案issueはcloseします．

closed 2021-09-14 by MasatoKubotera