SUSTAINA-OP/SUSTAINA-OP #52: メンテナンス台 作成

MasatoKubotera, 2021-06-10:
現在， @hiroki-0001 が作成したメンテナンス台を使用しているが，TPUバンパーに干渉に変形の要因となっている．

MasatoKubotera, 2021-06-26:
https://github.com/citbrains/citbrains_humanoid/issues/1292

HirokiHasegawa777, 2021-07-02:
現在の作成案です。
[image]

HirokiHasegawa777, 2021-07-17:
モデリングできました！
[image]

HirokiHasegawa777, 2021-07-17:
現在、Aパーツと下のフレームとをオスメスにして固定することを考案中です。

HirokiHasegawa777, 2021-07-28:
[image]
ブリムで座布団のようにプリントしてみたところ
[image]
のように、座布団の一部が部品に完全にくっついたままになってしまい、剥がすことが出来ませんでした。

[image]
φ5の穴はサポート材無しで綺麗に仕上げることが出来ました！

今後については、とりあえずこの部品をダメ元で加工し、上手く行けばこのまま形にしようと思います。上手くいかなければ蝶番などを買って代用しようと思います。そもそもですが、今回考えた台はプリントに時間がかかり過ぎてしまうため、2個目以降は買ったパーツで組み立てる案を考えることになります。

MasatoKubotera, 2021-07-28:
@HirokiHasegawa777 
>座布団の一部が部品に完全にくっついたままになってしまい、剥がすことが出来ませんでした。

[image]

画像中のラフト間のラップをいじることでラフトとモデルの隙間が広がり，取れやすくなるかと思います．モデルとラフトの隙間にスクレーパーを入れながらはがすと設定を変えずに取れることもあり，取り方もラフトとモデルが結合してしまう原因でもあります．次の機会に検証してみてください．

>φ5の穴はサポート材無しで綺麗に仕上げることが出来ました！
サポートなしでもプリントできてよかったです！！

>今回考えた台はプリントに時間がかかり過ぎてしまうため、2個目以降は買ったパーツで組み立てる案を考えることになります。

時間を削る方法として，，，
1. 積層ピッチ(Z)を荒くする
2. （主に中間層の）プリント速度を早くする．
3. プリントする体積をそもそも減らす．
4. ラフトの余白部分を最低限にする．
などなどいろいろ挙げられますので，上手くいったときの時短技として検証してみてください．

既製品の丁番を使ったものを作るときは，3Dプリントではなくて切削などの方が早いかもしれません．．．

HirokiHasegawa777, 2021-11-06:
[image]

[image]
上記の写真は、折りたたんだ状態の写真と通常の状態の写真です。パーツは全て揃ったため、これから金属棒を差し込み、接着剤で固定します。
なるべく急ぎでということでしたので、表面精度を上げるためのラフトでプリントするという試みは、量産する場合に再開しようと思います。

MasatoKubotera, 2021-12-12:
現状のリソース的に開発を行う必要性がないためcloseします．
Ver４については，全面がBumperを外すとフラットになるため，メンテナンス台なしで整備が可能となっています．

closed 2021-12-12 by MasatoKubotera