SUSTAINA-OP/SUSTAINA-OP #55: 機体no消費電流の計測

MasatoKubotera, 2021-06-13:


hiroki-0001, 2021-06-18:
サーボモータの消費電流を調査したい、現状その方法が存在しないため方法を確立させる。
一応プログラム側で対処もできそうだが比較のため、実世界からも計測できるようにする。

yasuohayashibara, 2021-06-28:
実はB3Mは過剰なぐらいインテリジェントなモータで各モータの中で電流値を計測しています．
アドレス0x48に電流値が入っていますので，読み出せば電流値を知ることができます．
（その他，様々なデータを読み出すことができます．）
https://kondo-robot.com/w/wp-content/uploads/B3M_SoftwareManual1.2.0.3.pdf

なお，各モータの電流値が必要なければ安定化電源に表示される電流値でも良いと思います．
CPUの電流値はほぼ一定ですので，差し引けばおおよその値が得られると思います．

MasatoKubotera, 2021-07-02:
各モータの電流値が必要なため，計測用コードの作成を依頼した．
<a href="https://github.com/citbrains/citbrains_humanoid/issues/1523">ロボットの動作中にサーボの電流を表示する機能の実装 #1523</a>

メインの基板からは，19個のサーボモータを計4分割(下半身2分割，上半身2分割)にして電源供給を行っているが下半身を中心に定格を上回っていることが予想される．
また，最大電流がXHコネクタの定格を超えている場合は分割を増やすなど，基板の再設計やさらなる対処が必要となる．

AD58-3104, 2021-07-08:
こちらはhumanoidの方にissueが立ったのでもう閉じて良いと思われます

MasatoKubotera, 2021-07-08:
@AD58-3104 
humanoidに新たに立ったissueは機能実装であって計測に関わることではないのでこちらのissueは残したいです．

計測結果や今後の対策を示す場としては，humanoidよりもこちらのOpenPlatformになるかと思います．

MasatoKubotera, 2021-07-15:
安定化電源の電圧計・電流計を用いて簡易計測を行った．
プリセットは，電圧：12.50[v]，電流は最大の10.00[A]とした．

動画は，CPU電源投入後・サーボ電源投入前から撮影し，サーボ電源投入，hr，前方歩行，サーボ電源切まで記録した．
[検証動画](https://drive.google.com/file/d/1voQCK5pGlf1CSqirTgdoYQ8zbDNTO2kn/view?usp=sharing)

この時，記録された最大電流は6.52[A]であり，歩行時の片足立ち状態であった．
[image]
キックモーションの際に最大電流が流れると予想し，今後ほかのモーションでも検証を行っていく方針である．

今後の測定には，[電流センサーIC ACS712ELCTR-30A-T](https://akizukidenshi.com/catalog/g/gI-12292/)を用いて計測することも考えている．

TakuyaNagamine, 2021-07-17:
## キックモーション時の最大電流の測定
・安定化電源の電圧計・電流計を用いて簡易計測を行った
・プリセットは，電圧：12.50[v]，電流は最大の10.00[A]とした．
・キックモーションの際に流れた最大電流は7.50[A]であり,インパクトの直後であった.
・[検証動画](https://drive.google.com/open?id=1OU0heFuLG3Kn2569atruj0caLhcywt9D&authuser=s21c1090ke%40s.chibakoudai.jp&usp=drive_fs=sharing)
[image]

TakuyaNagamine, 2021-07-17:
## 横移動時の最大電流の測定
・安定化電源の電圧計・電流計を用いて簡易計測を行った.
・プリセットは，電圧：12.50[v]，電流は最大の10.00[A]とした．
・横移動の際に流れた最大電流は7.58[A]であり,動き出してから約6秒後であった.
・[検証動画](https://drive.google.com/open?id=19T2JsM3oK5dh-tmA5ilP3VWQLuxlx31a&authuser=s21c1090ke%40s.chibakoudai.jp&usp=drive_fs=sharing)
[image]

sekitorimisaki, 2021-07-18:
・安定化電源の電圧計・電流計を用いて簡易計測を行った.
・プリセットは，電圧：12.50[v]，電流は最大の10.00[A]とした．
・仰向けからの起き上がりの際に流れた最大電流は19.84[A]であった.
[image]
[検証動画](https://drive.google.com/file/d/1-6OP6UVkbqodbI1TgRJ8WCJ3gbql1lQs/view?usp=sharing)

sekitorimisaki, 2021-07-18:
・安定化電源の電圧計・電流計を用いて簡易計測を行った.
・プリセットは，電圧：12.50[v]，電流は最大の10.00[A]とした．
・うつぶせからの起き上がりの際に流れた最大電流は7.29[A]であった.
[image]
[検証動画](https://drive.google.com/file/d/1-2Yv1d8yqCI0q1YmdvvWDv5EeYK2a9yD/view?usp=sharing)

yasuohayashibara, 2021-07-22:
歩行が不安定になると，多くの電流が流れます．
見ると不安定な歩行となっていますので，通常よりも多くの電流が流れていると思います．

MasatoKubotera, 2021-07-23:
サーボモータによる電流測定も実行しており，作成した@AD58-3104 によると歩行が不安定になっている要因は測定プログラムによるものと考えられるようです．

MasatoKubotera, 2021-07-23:
安定化電源の供給最大電流である10Aに到達したため，以後の検証はより大きい電流が供給できる安定化電源，もしくはLIPOバッテリを用いることを考えています．

MasatoKubotera, 2021-07-24:
機体全体の消費電力を計測する装置を作成した．
ケーブルコネクタがXT60とVHコネクタのもので安定化電源ORバッテリと機体間に接続して計測を行うためのものである．

今後，CPUの消費電流を測定する用の装置も作成する必要がある．

MasatoKubotera, 2021-08-17:
[電源基板のノイズ調査 #125](https://github.com/citbrains/OpenPlatform_ver3_0/issues/125#issuecomment-900200535)で詳細を記載したが，Jetson TX2の消費電力に関しては下記のコマンドで計測することが出来る．
```
cat /sys/bus/i2c/drivers/ina3221x/0-0041/iio:device1/in_current0_input
```

MasatoKubotera, 2022-01-01:
- https://github.com/citbrains/OpenPlatform_ver3_0/issues/112 

の完成待ちになります．一つの指標となるため，計測したいと考えています．

TakuyaNagamine, 2022-04-16:
現在、緊急で行う必要がないため一旦クローズします

closed 2022-04-16 by TakuyaNagamine