SUSTAINA-OP/SUSTAINA-OP #56: 3Dプリンタ 修理

MasatoKubotera, 2021-06-13:
ANYCUBIC i3-Mega 1号機のノズル温度が上昇しつづける問題の解決．

YukiShigematsu, 2021-06-19:
# ３Dプリンタの故障個所
[image]
上の写真の赤丸で囲まれたFETが故障しているため、新しいFETに交換する

もともとついていたFET
FDD-8708
https://www.digikey.jp/product-detail/ja/on-semiconductor/FDD8780/FDD8780CT-ND/1305821

交換に使用したFET
NP52Ｎ06SLG
https://akizukidenshi.com/catalog/g/gI-09928/

交換後、FETの各端子と基板との導通を確認できた。
動作確認はまだ行っていない。

RyodaiArai, 2021-06-26:
動作確認完了．
問題なく使用時の温度(230℃)付近に到達し，温度が上昇し続ける問題も解決しました．

MasatoKubotera, 2021-07-04:
7/3(土)に現状を確認した．

167機製作の時に作成したgcodeで動作確認を行い次の問題点が発覚した．
- ノズルから出るフィラメント(PLA)が細い(Filament IN)
- フィラメント(PLA)の排出が一定ではない．(プリント時)
- ビルドプレートに定着しない

問題解決のため次のことを実施し，詳細を示す．
- **フィラメント(PLA)を新品に交換**
  湿気の吸い過ぎによる問題か判断できなかったため．
- **ビルドプレートに3M製3Dプリンターシートを張る**
  ビルドプレートの消耗への対応
- **ノズル・温度センサユニットの交換**
  FET故障によりノズル内部が破損していた．
- **押し出し機構の分解/調査**
 問題なし
  [image]

再度動作確認を行ったが，状態は変化せず通常使用はできない．

MasatoKubotera, 2021-07-08:
現状では対処のしようがないためcloseします．

closed 2021-07-08 by MasatoKubotera