SUSTAINA-OP/SUSTAINA-OP #57: 12号館4階のCNC面だしをする

NaokiIwasawa, 2021-06-14:
6月14日現在、12号館の4階のマシニングセンタ（CNC）のPOM製ベットは免許取得の講習のために外されている状態で、
POMに両面テープを貼る方法での切削は出来ない。そのため切削を行うためはPOMを固定し面だしを行う必要がある。

NaokiIwasawa, 2021-06-17:
6/16日に面だし完了

MasatoKubotera, 2021-06-17:
今後も定期的に行っていくことなので，面出しの注意事項や方法，camを共有ができればと思います．

MasatoKubotera, 2021-06-19:
面出しは，φ9のエンドミルを使用する．

面出し範囲はx300[mm]，y300[mm]のテーブルである．
気を付ける点として，y軸の可動範囲は問題がないが，x軸の可動範囲は実測で305.098[mm]であった．
そのため，面出しを行うテーブル，CAMパスをこの範囲内に収める必要がある．
今回実践する方法として，x軸の可動範囲の最小値から2.5[mm]((305.098-300)/2≒2.5)離して配置・固定した．

パス作成の注意点として，可動範囲外に余裕がないため進入退出動作をなくす必要がある．また，削るのは高さ0.5[mm]程度で十分である．

NaokiIwasawa, 2021-07-03:
今回の面出しで利用した[gcode](https://drive.google.com/file/d/1uuD13BoVkDFmoEphaWV74o0rtHdsdQUo/view?usp=sharing)です。

動きの概要
- z軸基準から0.5ｍｍ切削
- 外側から内側に向かって正方形を描いて移動
- 面出しをCAMで選択すると両辺からの侵入でオーバートラベルになってしまうため、ポケット加工を選択している。
- 侵入については垂直切り込み
- 切削時間は30分程度
- エンドミルはΦ9
- パラメーターは[こちら](https://docs.google.com/document/d/1MHM7T2BDNjA8xqrB54MIc1QrvLaiD6PXMtrIb-hadEg/edit)を参考にした

元のCAMについては誤って保存を忘れてしまったので残ってないです。

closed 2021-07-03 by NaokiIwasawa

NaokiIwasawa, 2022-04-20:
切削に使ったCAMが当初紛失していたと思われたのですが，再度探した結果見つかった．
[以下](https://drive.google.com/drive/folders/1PLaVBB_uevN_jeRBwg03xgredOmvV9xF)にアップロードしたので必要に応じて参考して欲しい