SUSTAINA-OP/SUSTAINA-OP #58: ネジ穴加工 下穴径の統一

MasatoKubotera, 2021-06-15:


MasatoKubotera, 2021-06-18:
GankenKun167機を作成した際に，面タップ加工するときに切削時に下穴をあけるがこの下穴径がネジ径により統一されていないことがあった．

参考までに，M2下穴径はΦ1.6，M3はΦ2.5である．
<a href="http://jitukawa.net/kikaku2.html">ミリネジ下穴径（ドリル径）</a>

面タップ加工，側面タップ加工の穴径を調査し，統一する必要がある．

MasatoKubotera, 2021-06-18:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/20 と関連性があるisuueである．

MibuchiYuta, 2021-06-25:
穴径に違いがあることは多数に認められた。いっぽうで、下穴をcadで表現している箇所とねじ穴を表現している箇所の二つがあった。これらを変更するべきか検討する必要がある。
## 下穴が表現されている
[image]
## ねじ穴が表現されている
[image]

MibuchiYuta, 2021-06-25:
[image]
[image]

ジオメトリ投影による穴径の指定があったため、編集できない状態です。以下のファイルが該当しています。
- 胴体サブフレーム_1_1
- 胴体サブフレーム_1_1_MIR
- 足裏_サブフレーム_1_1
- 足裏独立ピッチ_1_1
- 肩メインフレーム_1_1
- 肩ピッチ_サブフレーム
- カメラ_メインフレーム1_KBCR-S02MU

MibuchiYuta, 2021-06-25:
上記の問題のため、それ以外の穴径の修正で止まっています。作業ブランチは　https://github.com/citbrains/OpenPlatform_ver3_0/tree/change_hole_size_before_threading　です。

MasatoKubotera, 2021-06-25:
パーツ名を正しく記載してください．inventorでの検索にヒットしませんでした．
上記のブランチURLがミスがあるので修正お願いします．

MasatoKubotera, 2021-06-25:
スパイクがなくネットワークを確認したところ．最新バージョンから派生を行っていないので，全てやり直しになります．
@NaokiIwasawa が機体のほぼすべてを更新しており統合することが出来ません．
ブランチを削除し再度作業をお願いします．
[image]

HirokiHasegawa777, 2021-08-24:
https://docs.google.com/spreadsheets/d/1y_O0oOIOrdUXkXNpVGKIjBIUrqWA2n85PWvWeApBHBM/edit?usp=sharing
上記のスプレッドシートに統一すべき下穴を部品ごとにまとめました。

ブランチによる問題を回避するため、後日一斉に修正することになっています。

MasatoKubotera, 2021-09-14:
上記スプレッドシートに記載しているパーツについては変更しました．

MasatoKubotera, 2021-09-14:
Ver4.0で統一した下穴径は，
- M3 : φ2.5
- M2 : φ1.6

である．

MasatoKubotera, 2021-09-14:
修正が終了し，下穴径が統一されたためcloseします．

closed 2021-09-14 by MasatoKubotera