SUSTAINA-OP/SUSTAINA-OP #60: 腕先 3Dプリンタパーツのインサートナット化

MasatoKubotera, 2021-06-15:


MasatoKubotera, 2021-06-29:
@sekitorimisaki @RyodaiArai

変更するパーツは，腕先.iptの3Dプリントパーツです．現状は，埋め込みナットで締結されていますがこの部分をインサートナットに対応するように変更してください．
<img src="https://user-images.githubusercontent.com/53966390/123739249-4746de00-d8e1-11eb-9fce-32f3a7aae64a.png
" width="320px">

<a href="https://www.monotaro.com/s/pages/productinfo/insatonato_type/">インサートナットの種類と特長</a>にインサートナットについてまとめられてるので参考にしてみてください．

使用する，インサートナットについては<a href="https://jp.misumi-ec.com/vona2/detail/221000748262/?PNSearch=UD-23001&HissuCode=UD-23001&searchFlow=suggest2products&Keyword=UD-23001">UD-​23001</a>にしようかと思っています．

リンク内のカタログに，どのような寸法にすればいいのかが記載されていますので，探してモデルを変更してみてください．

sekitorimisaki, 2021-07-02:
# 腕先の改良
## 改良内容

・腕先のナットを下のリンクをもとにインサートナット化した。
http://www.tokai-mmc.co.jp/insert/seihin_ud2.html

[image]

sekitorimisaki, 2021-07-02:
@MasatoKubotera 購入はどうすればいいですか？

MasatoKubotera, 2021-07-02:
@sekitorimisaki 
設計が早くて正直驚いています．．．申し訳ないですが，予算の関係上すぐに購入することができません．
3Dプリンタのプリントパラメータの調整に時間がかかると思いますので，まず3Dプリンターの使い方を学び，プリントしてみましょう．

また，@sekitorimisaki が設計してくれたCADモデルを全体に反映するためにGitHubの知識が必要となります．
追々やっていければと思います．

sekitorimisaki, 2021-07-03:
設計したものをもとに印刷してみます。

sekitorimisaki, 2021-07-07:
[image]
印刷し終わりました。
赤丸の部分が本当は凹む予定なのですが凹みませんでした。

MasatoKubotera, 2021-07-08:
@sekitorimisaki 
inventorのモデルが共有されていないので確かなことは言えませんが，
積層ピッチ(レイヤー高さ)[mm]が凹みよりも浅い場合，プリントすることができません．
プリントする前にCuraでシミュレーションできますので「凹まない」ことは確認できるかと思います．

sekitorimisaki, 2021-07-16:
[image]
インサートナットが裏から飛び出てしまい頑健君に上手くはまりません

sekitorimisaki, 2021-07-16:
[image]
インサートナットが裏から飛び出てしまい腕にはめることができませんでした。

MasatoKubotera, 2021-07-19:
モデルを変更し調節することもできるが，長さが短いインサートナットへの変更することで対応する．
そのため，一時保留とする．

MasatoKubotera, 2021-08-06:
長さが短いインサートナットとして，[UD-43040](https://jp.misumi-ec.com/vona2/detail/221000748262/?HissuCode=UD-43040&PNSearch=UD-43040&KWSearch=UD-43040&searchFlow=results2products&list=PageSearchResult)を使用する．

[カメラ_外装の再設計 #67](https://github.com/citbrains/OpenPlatform_ver3_0/issues/67)と共用のインサートナットです．

KentaTsukino, 2021-08-18:
新しいインサートナット(UD-43040)に対応させました。
[image]

MasatoKubotera, 2021-09-02:
こちらの部品は，3Dプリントのため緊急性がないため期限を延長します．

TakuyaNagamine, 2021-09-03:
@MasatoKubotera 
スケッチの量を最低限にし,上記に記されていた問題点を修正して再設計しました.
・[試作品データのリンク](https://drive.google.com/drive/folders/1SsOy6xFHybhIJwtBxnrvtVpbqODPwvvF?hl=ja)
## 写真
[image]

MasatoKubotera, 2021-09-05:
@TakuyaNagamine 

返信が遅くなりました．スケッチの削減ありがとうございます．
スケッチエラー表示がなくなっていることを確認しましたが，アセンブリを行い統合するときに干渉エラーが出ました．
下図に示しておきます．

[image]

要因は，スケッチ2だと思われます．
良く分からない数値で拘束されているので前腕_サブフレーム_1_1を派生させてスケッチで利用するか，同等のスケッチを再現した方が良いです．
[image]

3Dプリンタはここまで精度が出るものではないので見逃すこともできますが，全体の干渉検証でエラーが出てしまうのでできれば対応していただきたいです．

TakuyaNagamine, 2021-09-07:
@MasatoKubotera 
先輩にご指摘いただいた部分を改良しました!
・[試作品データのリンク](https://drive.google.com/drive/folders/1SsOy6xFHybhIJwtBxnrvtVpbqODPwvvF?hl=ja)

MasatoKubotera, 2021-09-07:
@TakuyaNagamine 
お疲れ様です．コアタイムの方で，アセンブリを行って干渉がなかったということなので今後ver4_0に反映していきたいと思います．

MasatoKubotera, 2021-09-08:
@TakuyaNagamine 
遅くなりましたが，ver4_0ブランチに反映しました．
干渉もなく無事に反映することができ，スケッチエラーにいても解決されていました．
確認でき次第，このissueを閉じてください．お疲れさまでした．

補足ですが，ドライブの共有にルール_1というエクセルファイルがありました．これはエクセルファイルの数値を変えることでスケッチを開かずともスケッチ寸法を変更するためのものです．前腕_メインフレーム_1_1.iptで読み込みされていましたがエクセルファイルに記載された寸法値が参照されていなかったので，前腕_メインフレーム_1_1.iptの設定をエクセルファイルを参照しないように変更しました．

TakuyaNagamine, 2021-09-09:
確認したのでcloseします!

MasatoKubotera, 2021-09-26:
下図に書き込んだが，最長のM3-10ねじを挿入しても長さ4mmのインサートの半分の2mm程度しか締結できないことが確認できた．

[image]

MasatoKubotera, 2021-10-03:
下図の通り，最小距離が3.726mmであるため，長さ4のインサートナットが完全に挿入できないことも判明した．
[image]