SUSTAINA-OP/SUSTAINA-OP #62: ガラエポ 購入先決め

MasatoKubotera, 2021-06-15:


hiroki-0001, 2021-06-18:
2mmのガラエポがルール違反しまう白色のものしかない。海外での購入も視野に入れ、購入先を検討する

MasatoKubotera, 2021-06-19:
脚の色については，ルールを要再確認．

yasuohayashibara, 2021-06-28:
現在，GFRPは千葉工業大学に出入りしている谷津商事という業者にお願いして調達してもらっています．
色を指定して購入することができるかもしれません．

hiroki-0001, 2021-08-23:
http://www.yatsushoji.co.jp/index.html
谷津商事さんのサイトを発見したので、こちらからメール等で黒のGFRPが購入できるか問い合わせてみます。

hiroki-0001, 2021-08-24:
以下のような内容で問い合わせてみます。

[image]

hiroki-0001, 2021-08-24:
以下のような内容で問い合わせてみようと思いますが、よろしいでしょうか？
[image]

yasuohayashibara, 2021-08-24:
テキストで送っていただければこちらで添削できますが，とりあえずコメントを送ります．

- スポンサ依頼ではありませんので，実績は伝えなくて構いません．
- だいたいのサイズを伝えたほうが良いです．
サイズによって入手可能かどうかが決まります．
ただし，素材の場合は～程度とした方が良いです．
- 林原が窓口になることを伝えたほうが良いです．
個人での購入は嫌われます．
取引のある担当者を通じて購入することを伝えたほうが良いです．
- 見積もりまでもらったほうが良い．
２度手間になるので，見積もりまでもらったほうが良いです．

hiroki-0001, 2021-08-25:
谷津商事さんにはGFRP2種類のみ注文する予定でしたが、A2017のT5の板材も合わせて購入したほうがいいと話し合いの結果なったので、それも合わせてお問い合わせしたいと思います。

hiroki-0001, 2021-08-25:
@yasuohayashibara 現在[こちらのドキュメント](https://docs.google.com/document/d/1q_4cEqjKXNEYhzL0yYBiqp898IkN4fs5Rz3gTj1qXXI/edit?usp=sharing)でお問い合わせ内容の文を編集しています。
上記で指摘していただいた点を踏まえて、書き直しました。添削をお願いします。

yasuohayashibara, 2021-08-25:
文言は添削しましたが，はじめから多くの枚数を購入するのは危険だと思います．
メカニズムの議論を見る限り板厚の誤差など許容できない項目がありそうですので，サンプルを購入して確認することが必要かなと考えます．
あと，黒の場合は反射するおそれがありますので，ロボット周辺に常に白線が見えるようになるおそれがあります．
色々不足の事態が想定されますので，試用してから購入したほうが良いと思います．
なお，板厚の精度が要求されるのであれば，その旨伝えたほうが良いと思います．
ただし，精度を保証する部品は価格が数倍になるおそれがありますので，ご注意ください．

あと気になった点として，T5のアルミはこんなに必要でしょうか？
２枚あればロボット１台分作れたような気がします．

GankenKun0.5のA2017S 5mmの切削用データ
https://github.com/citbrains/GankenKun_0_5_dxf/tree/master/A2017/5mm

5mmはそれなりに高価ですので，他の部品が購入できなくなりますのでご注意ください．

hiroki-0001, 2021-08-30:
@yasuohayashibara  8/27のメカ班ミーティングでガラエポの購入数について議論したので、こちらに結論を書いておきます。

**黒色GFRP  T2  200mm × 300mmを3枚購入し、黒色GFRP T3, A2027 T5の2種類に関しては在庫確認だけする方針で連絡しようと思います。**

変更に伴い、[以前添削していただいた本文](https://docs.google.com/document/d/1q_4cEqjKXNEYhzL0yYBiqp898IkN4fs5Rz3gTj1qXXI/edit?usp=sharing)を書き換えましたので、問題点がある場合は指摘をお願いします。チェック後、谷津商事さんのほうにお問い合わせします。

hiroki-0001, 2021-09-02:
文章を再訂正したので確認していただきたいです。
今回のお問い合わせでは購入目的ではなく、そもそも黒色のGFRPを取り扱っているのかをとりあえず確認したいので『取り扱いがあるかどうかを質問するような文章』にしました。
話が何度も変わってしまうような形になってしまい申し訳ありません。

yasuohayashibara, 2021-09-03:
コメントいれました．

hiroki-0001, 2021-09-03:
ありがとうございます！こちらの文章で問い合わせてみます。

hiroki-0001, 2021-09-18:
谷津商事さんの方から返信が来ました。
返信内容については[こちらのドキュメント](https://docs.google.com/document/d/1q_4cEqjKXNEYhzL0yYBiqp898IkN4fs5Rz3gTj1qXXI/edit?usp=sharing)にまとめておきます。
また今回送られた資料などに関しては、[Googleドライブ](https://drive.google.com/drive/folders/1kDY6LEpT9WaE-a9FRrlwxkGpHGLbnfmK?usp=sharing)の方にまとめておきました。

MasatoKubotera, 2021-09-18:
@hiroki-0001 
返答にあった**ES-3230**ですが，下記のリンクから色が確認できました．

[image]

http://www.matsuda-eng.co.jp/prod.html

---

167機で現状使っている緑色GFRPの品番が分かりましたので添付します．
A31-782538

[image]

MasatoKubotera, 2021-09-18:
上記のES-3230は緑黒のような感じがします．
添付されていた資料の中で**ES-3520J**がかなり黒に近いかと思われます．

https://www.risho.co.jp/product/products2/un_clad/property/es-3520j.htm

[image]
http://www.matsuda-eng.co.jp/prod.html

MasatoKubotera, 2021-09-18:
T3については，現状使用している[黒FRPプレート](https://store.shopping.yahoo.co.jp/robotena/m0145.html?pid=ibx&bid=1&sc_e=mshp_nc_con_77_pc_NC00034308)と比較すると1.4倍ほどコストかかるので要検討です．

MasatoKubotera, 2021-12-31:
Ver4.1の試作については現状のもので足りますが，量産分がないため正月休み中に決めたいと思います．

MasatoKubotera, 2022-01-11:
足裏として使用している[黒FRPプレート（200x300x3.0mm, ガラスエポキシ, カーボン配合）](https://store.shopping.yahoo.co.jp/robotena/m0145.html?pid=ibx&bid=1&sc_e=mshp_nc_con_77_pc_NC00034308)についてもT2と同様なものを使用していきたいと思います．

使用枚数比は1機体あたり，T2：3枚，T3：1枚です．

最低購入数
T2：3枚×7=21
T3：1枚×7=7

余裕をもって
T2：3枚×7+9=30
T3：1枚×7+3=10
でも良いかもしれません．

MasatoKubotera, 2022-01-16:
購入案　詳細

型番　寸法(縦x横x厚)　数量
ES-3520　200x300x2　30
ES-3520　200x300x3　10

MasatoKubotera, 2022-01-20:
ES-3520ですが，一般公差が±0.25となっていますが研磨品は，T2で±0.04，T3で±0.05になるようです．
足裏に使用する T3は一般公差でも構いませんが，リンクに使用するT2は研磨品の方がよさそうです．
https://www.risho.co.jp/product/products2/un_clad/property/tolerance.htm

RISHOのHPからガラス布基材　エポキシ樹脂　積層板のラインナップを確認しましたが，ES-3520しか考えられませんでした．https://www.risho.co.jp/product/products2/un_clad/index.html

MasatoKubotera, 2022-01-20:
その他の候補
- [住友ベークラフト](https://www.sumibe.co.jp/product/hpp/circuitry/sumilite-pl/index.html)　
[ガラスエポキシ積層板/紙フェノール積層板カタログ（PDF 3,547KB）](https://www.sumibe.co.jp/ww-attaches/805.pdf)

Keita1Mori, 2022-01-21:
追加でその他の候補
[村上電業](https://www.murakamidengyo.com/cart/epoxyglass.html)

MasatoKubotera, 2022-01-25:
決定稿
Ver4.1で用いるGFRP

---

[エポキシガラス（黒色）](https://www.murakamidengyo.com/cart/epoxy-kuro001.php#jump-dab132382837038e40a1e765b3e62737) 村上電業

[image]

エポキシガラス黒 厚み2幅200高さ300 数量30
エポキシガラス黒 厚み3幅200高さ300 数量10

---

板の詳細
[日東シンコー エポキシガラスクロス積層板 SL-EC](https://www.nittoshinko.co.jp/web-box/upload/pdf/E0209_1712R10.pdf)
厚さ公差：±0.25

その他資料については[ドライブ](https://drive.google.com/drive/folders/1B5po4f1wR8JWmAhhmDA1m-E44rXwFpgb?usp=sharing)(チーム外秘)に保管している．

MasatoKubotera, 2022-02-09:
到着しました．
すべてではありませんが，ほとんどの板に保護フィルムが貼付されていました．
板の表面状態を確認は下図で確認できます．

表面状態を見る限り，T2とT3は微妙に違う気がします．
T3については，以前から使用している[黒FRPプレート（200x300x3.0mm, ガラスエポキシ, カーボン配合）](https://store.shopping.yahoo.co.jp/robotena/m0145.html?pid=ibx&bid=1&sc_e=mshp_nc_con_77_pc_NC00034308#)と同じような感じがします．

[image]「GFRP T2　保護フィルムあり」
[image]「GFRP T3　保護フィルムあり」

[image]「GFRP T2」
[image]「GFRP T3」

closed 2022-02-14 by MasatoKubotera