SUSTAINA-OP/SUSTAINA-OP #65: 単純穴(ガバ穴)の径調査/統一

MasatoKubotera, 2021-06-19:


MasatoKubotera, 2021-07-08:
CADの作成者が多く共通化されていない．そのため，単純穴(ガバ穴)の径が同じねじ系でも複数存在している．
また，適切な単純穴(ガバ穴)の径が定められていないため，調査する必要がある．
はめあいの方に次いで，組み立て時にガタや径が適切な大きさではない(大きすぎるなどの)ために個体差が発生する恐れがある．

HirokiHasegawa777, 2021-08-23:
https://docs.google.com/spreadsheets/d/1HX8agHPXO6uKsF9gblbU09bVT7DCAgCpy_jCZBSQtf4/edit#gid=0
上記のスプレッドシートに調査結果をまとめました。

調査の結果3.1、2.7のバカ穴が多かったため、
**M2-2.1　M2.6-2.7　M3-3.1**　といったように**0.1広げた径で統一**することになりました。

yasuohayashibara, 2021-08-25:
穴の直径に対して指摘しておきます．

穴の直径はどの程度穴を精度高く加工できるかということに依存します．
重なり合う穴の中で，一つでもボール盤で穴を開ける場合は±0.05mmの精度を出すことは難しいと思います．
M3の場合，精度高く加工できる方が担当するのであればΦ3.2，あまり経験値の高くない方が担当する場合はΦ3.5がよく使用されます．
（あまり詳しくないので，調査していただけれと思います．）

逆に全ての穴をCNCで加工するのであれば，セッティングさえきちんとできていれば，まったく広げなくても大丈夫の場合があります．（M3の場合はΦ3でOK）
材料が樹脂かジュラルミンかにもよりますが，適正な穴のサイズを見極めることが必要です．

ギリギリのサイズで作ると，結局追加工で穴のサイズを広げることになるおそれが高いです．
また，追加工により特定の部品の組み合わせできてしまった場合，よく理解していない人が組み立てると，無理にねじを押し込もうとして結局ネジ溝を削るおそれがあります．
（組み合わせを変えないということであれば大丈夫ですが，おそらく組み替えると思いますので）
特定の組み合わせにならないようにするためには，少し余裕をもたせたほうが良いです．
もしくは，ネジ穴の位置の精度を規定して，それ以外は不良品として取り扱うことができれば，穴を小さくすることができます．