SUSTAINA-OP/SUSTAINA-OP #67: カメラ_外装 はめ合い修正

MasatoKubotera, 2021-06-19:
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/66

MasatoKubotera, 2021-06-19:
カメラ_外装を修正する際に，下記内容も出来れば修正したい．
1. USBMini-Bの逆のコネクタは使用しないため，覆う．
2. M2のタップ加工が施されているが，インサートナットに変更．

TakuyaNagamine, 2021-07-01:
下の写真のように隙間を作り、丸みを帯びていた部分を削ってみました
[image]
Format: [image]

TakuyaNagamine, 2021-07-01:
下の写真のように隙間を作り、丸みを帯びていた部分を削ってみました
[スクリーンショット (25)](https://user-images.githubusercontent.com/83112606/124076414-2d440180-da81-11eb-9df0-badd704cf31a.png)
[image]

TakuyaNagamine, 2021-07-01:
下の写真のように隙間を作り、丸みを帯びた部分を削ってみました
[image]

TakuyaNagamine, 2021-07-01:
[image]

TakuyaNagamine, 2021-07-01:
# カメラ_外装の改良
## 改良部分
・ねじ穴周辺
・レンズはめ込み部分
・その他
## 改良内容
・はめ込んだあとに、取り出しやすくするために隙間を作った
・3Dプリンターが曲線を再現するのは難しいとのことなので曲線を極力減らした構造にした
[image]

TakuyaNagamine, 2021-07-01:
現在3Dプリンターが使用不可とのことだったので、使用可能になった後に実際に作成してみる

TakuyaNagamine, 2021-07-01:
@MasatoKubotera 
「M2のタップ加工が施されているが，インサートナットに変更．」
とのことですが，このインサートナットに種類の指定はありますか？

MasatoKubotera, 2021-07-02:
@TakuyaNagamine 

インサートナットは<a href="https://jp.misumi-ec.com/vona2/detail/221000748262/?searchFlow=combo2products&KWSearch=UD-%e2%80%8b42001&HissuCode=UD-42001&PNSearch=UD-42001">UD-​42001</a>でお願いします．M2用のインサートナットです．(Brains に在庫はありません)

@sekitorimisaki さんは，下記のように寸法を求めたようです．
>腕先のナットを下のリンクをもとに寸法を決め、インサートナット化しました。
>http://www.tokai-mmc.co.jp/insert/seihin_ud2.html

<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/60">腕先 3Dプリンタパーツのインサートナット化 #60</a>でも3Dプリンタパーツにインサートナットを埋め込んでいるので参考，相談してみてください．

TakuyaNagamine, 2021-07-02:
インサートナットに対応するように加工してみました
[image]

TakuyaNagamine, 2021-07-05:
# 試作品 Part1
## 良かった点
・はめた後に素手で外しやすくなった.
## 改良点
・外しやすくなったものの,完全に垂直でないとはめることが難しかった（プロファイル0.3でプリントしたので0.1や0.2にすれば解決する可能性アリ）.
## Part2に向けて
・プロファイルを0.2にすると3.5時間,0.1にすると7時間と非常に時間がかかるので,0.3でもはめ込みやすくなるように全体的な幅を調整する.
[image]
[image]

TakuyaNagamine, 2021-07-05:
@MasatoKubotera 
カメラ_外装のレンズの部分のはめ合いについて確認しようと思ったのですが,それを確認しようとすると
GanKenKunに付いている既存のカメラ_外装を一度外さなければなりません.
私だけでは判断しかねるのですが,どうすればよいでしょうか？

MasatoKubotera, 2021-07-05:
@TakuyaNagamine 

初プリントおめでとう👏
きれいな仕上がりだと思います．Part2に向けて頑張ってください．

ここで，プロファイルの共有方法について紹介します．
1. 「プロファイルを管理する」を選択．
2. 自分で作成したカスタムプロファイルを選択し「書き出す」

[image]
[image]

上記を行うことでcuraprofaile形式でプロファイルが共有・保管することが出来ます．
良いパラメータをなくさないように保管してください．Googleドライブに上げてもらっても構いません．
また，複製することで一つのプロファイルから派生して作ることもできますので試してみてください．

>カメラ_外装のレンズの部分のはめ合いについて確認しようと思ったのですが,それを確認しようとすると
GanKenKunに付いている既存のカメラ_外装を一度外さなければなりません.
私だけでは判断しかねるのですが,どうすればよいでしょうか？

上記の件ですが，新規に外す場合は163機で分解調査を行うので一緒に分解することができます．また，分解後でよければ165機の物があります．こちらは @Keita1Mori @NaokiIwasawa が場所を把握していますので声をかけてみてください．

TakuyaNagamine, 2021-07-05:
@MasatoKubotera 
了解しました！

yasuohayashibara, 2021-07-06:
> GanKenKunに付いている既存のカメラ_外装を一度外さなければなりません.

新しいカメラもあると思いますよ．
レンズの部分は力をかけると光軸がずれますので，ご注意ください．

みなさんへ
別件ですがこちらのカメラ特注品で最も時間のかかる部品ですので，不足する場合は早めに購入を検討したほうが良いです．

TakuyaNagamine, 2021-07-06:
# カメラ_外装とレンズ部分とのはめ合い
## 結果
・カメラのレンズ部分ははめやすくなり,取り外す時もレンズを傷つける可能性が低くなったと思われる.
## これから
・インサートナットが入荷したらつけてみる
## 備考
・以前「はめることが難しい」と書いたが,工夫すればはめられないことはなく,あまりはめやすくすると外れやすくなってしまうので,このままでもよいのかもしれない.
[image]
[image]

TakuyaNagamine, 2021-07-18:
# インサートナットの取り付け
## 結果
・「USBMini-Bの逆のコネクタは使用しないため，覆う」とのことだったが覆ってしまうと,外す時にどうしても傾いてしま　　　　　　　　　　　　　
　い,レンズを傷つけることが分かったので,覆わずにそのままにする.
・レンズの部分が少しきつく,倒れたときに衝撃を与えてしまいそうなので少し広げる.
## 備考
・ねじによってある程度固定されているので「はめ合いを修正する」というよりも「衝撃を与えず,安全に外せる構造にする」というようにした方がいいのかもしれない.
[image]
[image]

yasuohayashibara, 2021-07-20:
お疲れさまです．

はめあいということでは，別のissueにしたほうが良いかもしれませんが，指摘しておきます．
上の写真で六角形の穴の左上の部分とレンズが干渉している（あたっている）ように見えます．
逆に右下が広いので，中心があっていないかもしれません．
転倒したときに大きな力が加わるおそれもありますので，少し隙間があったほうが良いと思います．
ただし，調整するのが大変な場合であったり，組立にスキルが必要だと思われる場合は，この六角形を広げたほうが良いかもしれません．（穴のサイズを小さくしなければならない理由は無いと思いますので．．．）

あと，大量生産する前に，できれば製作した頭部をロボットに搭載して，ボールの位置が正しく検出できるかをチェックした方が良いと思います．
カメラが傾いて搭載されていたりするとボールの位置が正しく検出されないことがありますので，再度作り直しになるおそれがあります．
従来のロボットではあまり検証してきていないように思いますが，この期に行っておいたほうが良いと思います．

検証には画像処理に強い方にも手伝ってもらった方が良いと思います．
別のissueにしたほうが良いかもしれません．

NaokiIwasawa, 2021-07-27:
163の再組立てで @TakuyaNagamine 君が製作したカメラ_外装を実装しました。
インサートナットは問題なく締結できましたが、衝撃等によって抜ける恐れもあるかと思いました。全身の組み立てが終わり次第検証の方をよろしくお願いいたします。

TakuyaNagamine, 2021-07-27:
@NaokiIwasawa 
了解しました！

MasatoKubotera, 2021-07-27:
@TakuyaNagamine @NaokiIwasawa 

>衝撃等によって抜ける恐れもある

インサートナット周りのPLAが薄いと感じました．もともと，インサートナットを入れずタップ加工して締結していたためもう少しインサートナット周りを厚くしても良いかもしれません．

インサートナット周りを厚くすると，カメラ_メインフレーム1_KBCR-S02MUも変更することになるかと思います．

---

カメラの認識についてですが，こちらの部品はあくまでカメラ外装であって(はめあいがきついことを除いて)このパーツの精度によりカメラ認識が変わるということはないかと思います．

[image]

---

[頭バンパーの必要性 #85](https://github.com/citbrains/OpenPlatform_ver3_0/issues/85)で提案していますが，カメラ_外装は前方への転倒時に地面に接触し衝撃を受けます．そのため，硬いPLAではもろに衝撃を受けその衝撃がカメラ_メインフレーム1_KBCR-S02MUにも伝わっていると考えられます．
時間があれば，素材をTPUに変更してプリントして検証してくくれると嬉しいです．
TPUの充填率によって転倒の衝撃で凹み，カメラ本体に衝撃が伝わる恐れがあるので注意が必要です．

TakuyaNagamine, 2021-08-04:
# 試作品part2
・試作品part1を実際に取り付けた結果や先輩たちのアドバイスに基づいて改良した.
## 改良点
・レンズをはめ込む部分の隙間を大きくした.
・インサートナット周りを厚くした.
・それに基づいて **カメラ_メインフレーム1_KBCR-S02MU** も変更した.
・USBMini-Bの逆のコネクタの部分を埋めなおした.
## 写真
・カメラ_外装
[image]

・カメラ_メインフレーム1_KBCR-S02MU
[image]

MasatoKubotera, 2021-08-06:
@TakuyaNagamine 
**カメラ_メインフレーム1_KBCR-S02MU**と**カメラ_メインフレーム**を締結しているねじが現状M2x12ですが，M2のねじを廃止したいため**M3に変更**をお願いします．
ネジ長については，最長10mmとなってしまうため締結方法の変更が必要となります．

現状では，**カメラ_外装**と**カメラ_メインフレーム1_KBCR-S02MU**と**カメラ_メインフレーム**の3つの部品を連結していますが，**カメラ_外装**と**カメラ_メインフレーム1_KBCR-S02MU**，**カメラ_メインフレーム1_KBCR-S02MU**と**カメラ_メインフレーム**の2つずつで固定することでねじ長10mmに抑えていただきたいです．

それに伴って，このissueの名前を変更させていただきます．

TakuyaNagamine, 2021-08-06:
@MasatoKubotera 
使用するインサートナットも変更するということでしょうか？

MasatoKubotera, 2021-08-06:
@TakuyaNagamine 
記入が漏れていて申し訳ないです．
インサートナットについても変更してください．
使用するインサートナットについては，[UD-43040](https://jp.misumi-ec.com/vona2/detail/221000748262/?HissuCode=UD-43040&PNSearch=UD-43040&KWSearch=UD-43040&searchFlow=results2products&list=PageSearchResult)でお願いします．

このインサートナットは，[腕先 3Dプリンタパーツのインサートナット化 #60](https://github.com/citbrains/OpenPlatform_ver3_0/issues/60)の長さが大きすぎた問題に対応したインサートナットであり，腕先の3Dプリントパーツと共用になります．

TakuyaNagamine, 2021-08-06:
了解しました！

TakuyaNagamine, 2021-08-11:
## カメラ_メインフレーム1_KBCR-S02MU・カメラ_外装の再設計
・ねじの変更及びインサートナットの変更により改良した.
# 改良点
・カメラ_外装をUD-43040に対応するようにした.
・カメラ_メインフレーム1_KBCR-S02MUとカメラ_外装をM3×6mmのねじで固定できるようにした.
　(アセンブリ済み)
# 写真
・カメラ_外装
[image]
・カメラ_メインフレーム1_KBCR-S02MU
[image]
・アセンブリ時
[image]
# 今後
・カメラ_メインフレーム1_KBCR-S02MUとカメラ_メインフレームをM3×6mmのねじで固定できるようにする.

TakuyaNagamine, 2021-08-12:
@MasatoKubotera 
 **M2のねじを廃止する** とのことですが,**KBCR-S02MU**と**カメラ_メインフレーム1_KBCR-S02MU**を取り付ける際に使用するM2のねじも変更するということでしょうか?
[image]

MasatoKubotera, 2021-08-12:
@TakuyaNagamine 
その点についても記載漏れでした．．．すみません．
カメラ基板の穴がφ2.3になりますのでその部分については，例外としてM2x6mm：4本を使用します．
[KBCR-S02MU](https://www.shikino.co.jp/products/product-kbcr-s02mu.php)

これ以外の場所についてはM2ねじについては廃止します．

TakuyaNagamine, 2021-09-02:
@MasatoKubotera 
カメラ_メインフレームと頭_メインフレームを繋げる部分の穴をねじ穴にしようとしたのですが,赤線で囲った部分の設定はどのようにすればよいでしょうか?
[image]

TakuyaNagamine, 2021-09-02:
@MasatoKubotera 
先輩にご指摘いただいた部分を改良しました!
・[試作品データのリンク](https://drive.google.com/drive/folders/1-ELrFyQRYSPTypYH1MUbgS2P9SowftKe)

MasatoKubotera, 2021-09-05:
@TakuyaNagamine 
改良ありがとうございました．
上記リンク内で共有されている，改良した下記のモデルをver4_0ブランチに統合しました．

(カメラ_メインフレーム，頭_メインフレームについては，[カメラ_メインフレーム・頭_メインフレーム ネジ径の変更 #131](https://github.com/citbrains/OpenPlatform_ver3_0/issues/131)で行われていた．)

- カメラ_メインフレーム.ipt
- カメラ_メインフレーム1_KBCR-S02MU.ipt
- カメラ_外装.ipt
- 頭_メインフレーム.ipt

[image]［反映後の本体.iam］

統合後に気づきましたが，カメラから出るUSBケーブルの逃げがありませんでした．
申し訳ないですが，カメラ_外装.iptだけ修正をお願いします．
[image]［USBケーブルの逃げがないカメラ外装］

MasatoKubotera, 2021-09-06:
@TakuyaNagamine 
修正ありがとうございます．
ver4_0ブランチに反映しました．

[image]

MasatoKubotera, 2021-09-06:
@TakuyaNagamine 
このissueで取り扱う内容，ver4_0への反映が終了しましたので確認次第，closeしてください．

TakuyaNagamine, 2021-09-07:
確認したのでcloseします

MasatoKubotera, 2021-09-26:
旧名：カメラ_メインフレーム1_KBCR-S02MUにおいて設計ミスが判明した．
詳細は下図の画像に書きこみしている．

この修正についてはすでにモデルに反映している．
[image]

MasatoKubotera, 2021-10-03:
[image]
再設計に伴って，インサートナットが必要となった．そのためアセンブリでインサートナットを挿入している．

MasatoKubotera, 2021-10-03:
再設計が終了したため，closeする．

closed 2021-10-03 by MasatoKubotera