SUSTAINA-OP/SUSTAINA-OP #69: カメラケーブル固定方法の見直し

MasatoKubotera, 2021-06-19:
参考：https://github.com/citbrains/OpenPlatform_ver3_0/issues/66

MasatoKubotera, 2021-07-08:
>3Dプリントパーツ(TPU)のCameraCable_Assistは，底面積が小さいためプリントが非常に難しい．USBをはめ込む構造をPOM部品の頭_サブフレームに付け加えたい．

MasatoKubotera, 2021-07-08:
実施する場合は，こちらのissueの作業状態による[カメラUSBケーブルの変更 #104](https://github.com/citbrains/OpenPlatform_ver3_0/issues/104)

MasatoKubotera, 2021-08-19:
現状のカメラケーブルの径に合わせて制作してください．

hiroki-0001, 2021-08-20:
カメラケーブルの径ですが計測したところ、3.7mmほどでした。写真を送付しておきます。
[image]

hiroki-0001, 2021-08-20:
こちらの径にあわせて、再設計してください。

yasuohayashibara, 2021-08-21:
あまり追えていないので勘違いがあるかもしれませんが．．．

１）カメラケーブルに関して
カメラケーブルですが，すでに林原も型番を把握していませんが，たまたま見つけて購入したものですので，これに最適化して設計するのは危険かもしれません．
もし最適化するのであれば，予備も含めたケーブルの確保を行ったほうが良いかもしれません．

２）コネクタに加わる力に関して
コネクタを囲う形で頭を製作するということだと思いますが，USBコネクタ周辺はある程度のマージンを持たせるようにしてください．
マージンが無いと固定時や衝撃が加わったときにUSBのコネクタに大きな力がかかるおそれがあります．
それぞれの部品がある程度誤差があり，衝撃力により変形することを前提に設計してください．
変形を許容しない部分がコネクタ部の場合は，応力はコネクタ部に集中します．

重要なのはケーブルを引っ張られたときにコネクタ部に力が加わらないことで，そのためにケーブルの途中を固定することが良い解決策です．ケースを付けることで，ケーブルの途中が固定されると，固定を忘れることもないので良いと思います．

hiroki-0001, 2021-09-02:
今回のissueでは下の写真の、『CameraCable_Assist』が形状ゆえ作成しづらいため、別の方法でカメラケーブルを固定する方法を探すために立てられました。
[image] 

現在考えられている固定方法としては、『頭サブフレーム』を写真のように、ケーブルを埋め込めるような溝を作ることでケーブルを固定する方法です。
[image]

この方法で固定する場合は以下のようになります。
[image]

hiroki-0001, 2021-09-17:
上の固定方法だと、ケーブルを完全に固定できないという意見をいただいたので、下の図のように以前まで使用していたケーブル固定パーツよりも簡単な構造のものを作成しました。

[image] 

このパーツを用いてケーブルを固定した場合、以下のようになります。

[image]

[image] 

このパーツをTPUを用いて、作成しようと考えています。

hiroki-0001, 2021-09-17:
こちらのパーツのモデルは、[googleドライブのこちら](https://drive.google.com/drive/folders/1z8Xgz0Li_ftzWolzXfrxGLQNqOgKW6LW?usp=sharing)にあげておきました。
現在は統合できないそうなので、できるようになったから統合します。

MasatoKubotera, 2021-10-04:
Googledriveで共有された部品をモデルに統合しました．
[image]