SUSTAINA-OP/SUSTAINA-OP #70: サーボID統一/修正作業

MasatoKubotera, 2021-06-19:


hiroki-0001, 2021-07-08:
現在GankenKunに使用されているサーボモータ(b3m1040,1170)にはそれぞれの部位ごとでIDが振り分けられている。過去に作成したID振り分けの画像が間違っていたので、それの修正。及び備品管理の再登録をおこなう。

hiroki-0001, 2021-07-08:
備品管理の方の再登録は終了したため、次はGankenKunの各ID振り分けの画像がわかりづらいため、修正した画像を作成する。出来が良ければそれを現在GankenKunの書くプログラムで使用されている画像と取り替えることも考える

hiroki-0001, 2021-08-10:
@KiyoshiroKawanabe 連絡遅れました。前回のミーティングでお願いしたissueに関してです。

[image][image][image]

こちらが現在使用されている、サーボモータの各ID振り分け図になります。

現状のままだと

- 現在は使用されていない首のピッチ軸(ID20)が存在する
- それぞれの回転軸(ヨー、ロール、ピッチ)が判別しづらい

などの問題があるためそれら改善した画像を作成していただきたいです。

hiroki-0001, 2021-08-12:
@KiyoshiroKawanabe GankenKunの画像です。

[image]
[image]
[image]
[image]
[image]
[image]

KiyoshiroKawanabe, 2021-08-16:
矢印なしのもの作成しました．

[image]
[image]
[image]
[image]

MasatoKubotera, 2021-08-17:
@KiyoshiroKawanabe @hiroki-0001 
ID4とID5
ID13とID14
逆な気がしますがどうですか？

KiyoshiroKawanabe, 2021-08-20:
[image]
[image]
[image]
[image]

hiroki-0001, 2021-08-20:
@KiyoshiroKawanabe @MasatoKubotera 
実機を見ながら確認しましたが、修正前のほうが正しいです。
細かい指摘になりますが、ID4, ID13を示す番号の位置が若干ずれています。
正しい位置は以下の画像のとおりです。
[image]

MasatoKubotera, 2021-08-20:
@KiyoshiroKawanabe @hiroki-0001 
ロール・ピッチ・ヨーで色分けした方が良い気がします．

2番と11番についてもかなり紛らわしいことになっているので，サーボモータ付近に番号を配置するのはやめた方がいいです．

KiyoshiroKawanabe, 2021-08-21:
色変更、矢印追加しました。
[image]

hiroki-0001, 2021-08-23:
こちらの画像を3分割したものはありますか？(左側・正面・右側)
その画像を追加でこのissueにあげてください。
画像の追加を確認次第、このissueを閉じたいと思います。

KiyoshiroKawanabe, 2021-08-24:
[image][image]
[image]
[image]

hiroki-0001, 2021-08-24:
ありがとうございます。分割画像もきたのでcloseします、

closed 2021-08-24 by hiroki-0001