SUSTAINA-OP/SUSTAINA-OP #71: サーボケーブル根本保護の提案

MasatoKubotera, 2021-06-19:
XHケーブルの根元を保護する<a href="https://www.rp-heliport.com/products/detail/11281">JST-XH バランス端子保護クリップ</a>を見つけた．

現在，根元を曲げすぎることによるケーブルの断線が多く見受けられる．
購入するには至らないと感じるが，このような部品を3Dプリンタ/TPU素材で作成できるのでないだろうか．

また，ピンセットではないと取り外せないコネクタを解消できるのではないだろうか．

aiba0921, 2021-06-26:
よろしくお願いします。

aiba0921, 2021-06-26:
[image]
ｑｑｑｑ

aiba0921, 2021-07-02:
ケーブル用のカバー制作のために各機体のケーブルの付き方を確認していたところ肩のケーブルで気になる部分がありました。
・ケーブルがほぼ直角に外に出ているためケーブルに負荷がかかっている可能性あり。
・ケーブルが肩のカバーの外側にあるため機体が横に倒れた時にケーブルが下敷きに
　なる可能性あり。
[image]
[image]

MasatoKubotera, 2021-07-02:
@aiba0921 
報告ありがとうございます．機体番号を記載してくると助かります．

ケーブルの付き方ですが，現状機体によって異なっています．
この原因は，整備を行った際にケーブルの取り回し方が決まっておらず，結束バンドで固定する箇所が定まっていないからです．今後サーボケーブルに関しては，組み立てマニュアルに記載し統一していく方針となっています．

下記のissueがサーボケーブルに関連するissueです．
- <a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/18">ケーブルを指定の長さで製作していただけないかを打診する #18</a>
  現状，購入時に付属するケーブルを適切な長さに再加工していますが，コネクタの圧着が適切に行えていないため通信できないことが多々あります．そのため，指定した長さでケーブルを作っていただけないか打診するissueです．

- <a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/22">サーボケーブル固定方法の見直し #22</a>
  結束バンドで固定するのでは毎回固定位置が異なってしまうのではないかというissueです．固定パーツを作成するという案を出しましたが，「重くなる」という理由で却下され，組み立てマニュアルに記載するという流れになりました．

- <a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/86">結束バンドの固定箇所確保 #86</a>
  @aiba0921 が指摘してくれたところもそうですが，結束バンドを使って固定できない箇所というものがあるというissueです．

aiba0921, 2021-07-03:
・来週の予定
　各機体のケーブルの確認

aiba0921, 2021-07-09:
報告遅くなってすみません。
今週は細かく確認できていませんが、先週の土曜日時点で確認しているのはコート上の機体では161,164,166の3機でした。
issue確認の後にモーション確認を164で行ったのですが、164の歩行が不安定だったこともあり実際に右側に横倒しになりました。その際ケーブルが期待の下敷きになってしまう事を確認したので、ケーブルの配置や長さを調整した状態でのモーション確認を今後行っていく必要があるかもしれません。

hiroki-0001, 2021-07-09:
@aiba0921 今はケーブルの配線方法の議論をしていると思いますが、配線方法とケーブルの根元部分の改良パーツの作成は少し論点がずれてきてしまうので、これより行うケーブルの配線の検証に関する議論はhttps://github.com/citbrains/OpenPlatform_ver3_0/issues/22で行っていただけると助かります。
急に違う場所での議論をお願いしているので、よくわからないようであればいつでも聞いてください。

aiba0921, 2021-07-18:
一応ケーブルの保護カバーをinventorの方で作ってみました。
[image]
[image]
ただ既製品がケーブルを上下から挟み込むのに対して、僕が作ったものだと現状では端子の外側からはめる形になるのでそのあたりを今後工夫していこうと思います。

yasuohayashibara, 2021-07-22:
ケーブルは現在最も多くの問題を抱えていますので，このような議論は良いですね．

おそらくケーブルで一番頻繁に起きている問題は，圧着の部分が以下の要因で外れるもしくは切れることによるものだと思います．

- 引張
- 繰り返し屈曲

上記のメカニズムは屈曲は防げますが，引っ張りが防げないかなと思いますので，もし使用するのであればケーブルを抱え込んだほうが良いと思います．
あと，外部からの力により，コネクタにモーメント力が加わらないかも検討した方が良いと思います．この部品を介してコネクタ部に力が加わると今度はコンタクトが変形するおそれがあります．

aiba0921, 2021-07-27:
先週の土曜日での指摘を受けてケーブルを抱え込めるようにしてみました。ただこの形だとカバーの取り外しが少々大変になると思うのでそこを今週中に改善する予定です。
[image]

aiba0921, 2021-08-06:
前回の課題であったカバーの取り外しをしやすいように改善しました。
[image]
[image]
[image]

hiroki-0001, 2021-08-06:
@aiba0921 このカバーの設計には3Dプリンターを使用すると聞いたのですが、現状の構造だとカバーとケーブルを固定するパーツ(写真の赤丸で囲っている部分)が小さすぎるので3Dプリンターで製作することが難しいと思います。
なのでこの部分のパーツを大きくする。または別の構造にしたほうがいいと思います。

[image]

aiba0921, 2021-08-10:
[image]
前回のミーティングで指摘されたフィレットの部分を削除し、ケーブルとカバーを固定するパーツを厚くしてみました。

aiba0921, 2021-09-24:
現在のbrainsで使用している積層式の3Dプリンターでは制作が難しいため光学式の3Dプリンターが導入された場合そちらで行う可能性あり。それまでは保留。

MasatoKubotera, 2021-11-19:
積層でもできそうです．．．
https://twitter.com/shiba_8ro/status/1461301583702999040?s=20
https://twitter.com/shiba_8ro/status/1461277948254646276?s=20

MasatoKubotera, 2022-01-17:
技術力がないため継続が困難だと思います．

closed 2022-01-17 by MasatoKubotera