SUSTAINA-OP/SUSTAINA-OP #72: LIPOバッテリーマニュアルの作成

MasatoKubotera, 2021-06-19:


MasatoKubotera, 2021-06-19:
管理体制の引継ぎも同時に行っていきたい．
- Google フォームによる電圧値報告
- Google スプレッドシートによる管理

MasatoKubotera, 2021-07-08:
@RyodaiArai @YukiShigematsu 
夏休みのBootCampで扱いたい内容となっています．
BootCampが8月の第1週だったと思います．完成後に一度，教える側(2年生以上)やハード班でも確認や勉強を行いたいと思っています．そのため，期限を7/21(水)とさせてください．

自分が今までに作成した資料をベースにしていただいても大丈夫です．
ドライブフォルダ：[Googleドライブ](https://drive.google.com/drive/folders/1pXoXR7TPv6wP7U26g3zNSM22tgsqoe1Y?usp=sharing) (チームメンバー以外アクセス不可)
以前作成した資料(ドキュメント版)：[210222_LIPOバッテリーマニュアル](https://docs.google.com/document/d/1u5dbN46fd6FFWAcv7LeVlAgN-MhtFxxyGSeRMIpn-PI/edit?usp=sharing)
以前作成した資料(Markdown版)：[lipo_battery_reference](https://github.com/MasatoKubotera/lipo_battery_reference)

MasatoKubotera, 2021-07-08:
LIPOバッテリーは危険物です．
チームとして管理体制や使用方法を確認するためにも重要な資料になるかと思います．

YukiShigematsu, 2021-07-19:
バッテリーマニュアル
https://docs.google.com/document/d/1u5dbN46fd6FFWAcv7LeVlAgN-MhtFxxyGSeRMIpn-PI/edit#
に
「慣らし充電」と「バッテリーチェッカーの使い方」の項目を追記しました。

MasatoKubotera, 2021-08-15:
マニュアルについては追記が終わり，ハード班でバッテリ勉強会が行われたためこのissueを閉じます．

closed 2021-08-15 by MasatoKubotera