SUSTAINA-OP/SUSTAINA-OP #74: 胴体_サブフレーム_1_10no

hiroki-0001, 2021-06-20:


MasatoKubotera, 2021-06-20:
エンドミルの逃げ加工を変更する必要性がありそうだ．

AD58-3104, 2021-07-09:
統一化するのでなく胴体_サブフレーム_1_10を修正する場合コネクタの通り道の部分を広げる事になると思います。このパーツは長辺方向に圧縮応力を受ける場合が多いと思いますが、その時の強度を下げないためにコネクタと干渉している部分をアーチにすると良いかなと思います。
そもそも殆ど力が掛からない部分なので単に干渉している部分を削るだけでも問題ないとは思います。

MasatoKubotera, 2021-07-09:
誤解させてしまっていると感じたので説明を付け加えます．

**胴体_サブフレーム_1_10**は，サーボケーブルのコネクタ部分が写真のように干渉するため修正する必要があります．
修正方法については，
>コネクタの通り道の部分を広げる事になると思います。このパーツは長辺方向に圧縮応力を受ける場合が多いと思いますが、その時の強度を下げないためにコネクタと干渉している部分をアーチにすると良いかなと思います。

で自分も良いかと思います．

>統一化するのでなく

という点ですが，**胴体_サブフレーム_1_11**が**胴体_サブフレーム_1_10**に酷似しており，**胴体_サブフレーム_1_10**を使用することで，パーツ種類が減るため，統一化と呼ばれるものを行っていきたい考えです．

MasatoKubotera, 2021-08-15:
[XHコネクタ データシート](https://akizukidenshi.com/download/ds/jst/xh.pdf)より，ケーブルを通すためには12.3x5.7の隙間が必要であることが分かった．
これに対応するために，胴体_サブフレーム_1_10.iptの穴加工部分を変更しました．
歪みの影響を少なくするために，追加でくり抜く部分は最小限にしています．
[image]
[image]

---
また，形状が似ている**胴体_サブフレーム_1_11**を廃止し，**胴体_サブフレーム_1_10**を使用することにした．
[image]

---
[胴体_サブフレーム_1_9の再設計 #89](https://github.com/citbrains/OpenPlatform_ver3_0/issues/89)でも議論されていた歪みについても**胴体_サブフレーム_1_10**における歪み個所の幅をを2mmから4mmに増大することで対応しました．
[image]

---
修正したCADについては，ブランチ[fix_torso_frame](https://github.com/citbrains/OpenPlatform_ver3_0/tree/fix_torso_frame)にPushしました．
@NaokiIwasawa 確認後統合お願いします．

MasatoKubotera, 2021-08-17:
ver4_0ブランチに統合されたため，issueを閉じます．

closed 2021-08-17 by MasatoKubotera