SUSTAINA-OP/SUSTAINA-OP #75: 上腕_メインフレーム_1_1の再設計

hiroki-0001, 2021-06-20:
6mmネジを挿入しするとパーツ内部でねじ同士が干渉してしまう問題と
[image]

サーボモータのケーブルのコネクタが引っかかってしまうため再設計が必要である。
＊一応押し込めば入るがケーブルの根本の負荷が大きすぎるため、再設計は必須。
[image]

AD58-3104, 2021-07-06:
この部分の再設計が必要なのには同意するのですが、腕全体の設計を変えてしまう方が良いように感じます。

MasatoKubotera, 2021-07-08:
肩以下の腕を再設計することは[腕ユニット 修正 #59](https://github.com/citbrains/OpenPlatform_ver3_0/issues/59)で扱います．現状再設計するリソースがないため，こちらのパーツを修正することになる可能性があります．

AD58-3104, 2021-07-09:
腕ユニット全体の再設計をするリソースがないのであればこちらのパーツの再設計も行わなくて良いのではないでしょうか。現状腕を修正する事に緊急性が無いので余計な手間になってしまう気がします。

MasatoKubotera, 2021-07-09:
腕ユニット全体の再設計は現状問題が指摘されておらず，
>腕全体の設計を変えてしまう方が良いように感じます。

は，リソース的に行えませんが，

>サーボモータのケーブルのコネクタが引っかかってしまうため再設計が必要である。
＊一応押し込めば入るがケーブルの根本の負荷が大きすぎるため、再設計は必須。

と指摘されているため，このパーツだけは修正したいと考えています．

AD58-3104, 2021-07-09:
全体設計の際にこのパーツ部分が手戻りの様になってしまうので、腕パーツを取り外す頻度によっては今急いでやる必要もないのかなと考えました。コネクタを抜き差しする回数が多いならこのパーツだけ再設計するのに賛成です。

aiba0921, 2021-08-17:
@MasatoKubotera
「6mmネジを挿入しするとパーツ内部でねじ同士が干渉してしまう問題」は現状では165のみで起きている問題ですか。それともほかの機体にも起きている問題ですか。

MasatoKubotera, 2021-08-17:
@aiba0921 上腕_メインフレーム_1_1は，現状の機体において同じバージョンのパーツが用いられているので，165機体に限った問題ではありません．
腕に関しては，足ほど重要視されておらずバカ穴になっていても放置されているのが現状です．

aiba0921, 2021-08-17:
@MasatoKubotera 
干渉している問題に関して163,165の再組立て時やその進捗報告のissueにはこの問題のことはあまり話題に上がってきませんでした。自分も163の左腕の組み立てを担当していましたが、「干渉してフレームを固定できない」というような感覚はあまり感じませんでした。自分の勘違いかもしれませんが、出来れば実機での(使われているネジの長さなどの)確認を一度お願いしたいです。

MasatoKubotera, 2021-08-17:
@aiba0921 
この部分については，実機で確かめることがかなり難しいです．

実機で確かめることができない理由として，下記の２つが挙げられます．

・**ネジ穴の垂直精度が出ていないものが多い**
側面のタップ加工である為，まずボール盤で側面に治具を用いて下穴(φ2.5)を空けます．
この際に，ドリルの軸と穴を空ける面の垂直精度が167機以前での機体では出ていません．
お互い，もしくは片方が外側にそれていると干渉は起きません．（この要因で干渉が起きないのは偶然です）
内側にずれてる機体では，ごくわずかですが面と頭間に隙間があります．
このことに関しては，切削・追加工の技術面についても向上しなければ解決することが出来ない問題でもあります．

・**干渉が小さすぎる**
inventorでネジ長6ミリのネジを締結し，干渉解析を行うと下図のようになり，0.142mm^3の干渉が見つかります．（白い丸のマークは干渉を表しているものではありません）
[image]

よって目に見えないため，実機で確かめることは難しくなりますが，干渉を起こしているため再設計になります．

aiba0921, 2021-08-18:
[image]
[image]
自分も、inventorで上腕サブフレーム_1_1の外側から上腕メインフレーム_1_1のネジ穴の底までの長さを計測してみたところ4.5mmだったのですが使用するネジをM3の6mmから4mmに変更することはできますか。（問題点としては若干外れやすくなる可能性あり）
[image]
もしくは、見た限りの干渉ならネジ穴の位置を1mm程度矢印の方向にずらすことで解消できないでしょうか。（こちらを行った場合サブフレームの方も微調整が必要になってくる。加えてずらしすぎると、はめ合わせる部分の周辺の強度が下がる可能性があるのでずらせても１ｍｍ～1.5ｍｍと思われます。）

aiba0921, 2021-08-20:
1mmだと結構ギリギリになってしまいそうだったので0.5mmにしてみました。
[image]
「サーボケーブルのコネクタが引っかかる」という問題に関しては解決できそうなデザインができました。
[image]
コネクタの縦×横がだいたい13mm×6mmなのでおそらく問題なく通せると思います。

aiba0921, 2021-08-23:
https://drive.google.com/file/d/1TR_zKGe9lkIOgS_JnmESDV1SrjbH4lfY/view?usp=sharing
前回のミーティングで指摘されたネジ穴とスケッチのを拘束して修正したデータをドライブに上げました。
（スケッチの拘束については、今後修正する際に必要になりそうな紫の線は一部残してあります。）

aiba0921, 2021-08-25:
https://drive.google.com/file/d/1R4qk7PVOgvhAMSgm9O47r_NMqeeH3rh5/view?usp=sharing
上腕_メインフレーム_1_1の再設計に伴い修正した上腕_サブフレーム_1_1のデータをドライブに上げました。

aiba0921, 2021-09-01:
議論の参考になればと考え、岩澤先輩からいただいた案をもとに制作した上腕_サブフレーム_1_1のデータをドライブに載せました。
https://drive.google.com/file/d/16lsK4SzfOAGvKvd166jAPmtLItQGU39A/view?usp=sharing

aiba0921, 2021-09-02:
GankenKun163機 整備記録 #2の
https://github.com/citbrains/OpenPlatform_ver3_0/issues/2#issuecomment-882180369
で上腕_サブフレーム_1_1の歪みが報告されていました。見た限りでは二股に分かれている辺り（今回削り方が議論される辺り）から歪みが発生していました。

aiba0921, 2021-09-02:
GankenKun165機 整備記録#3
https://github.com/citbrains/OpenPlatform_ver3_0/issues/3#issuecomment-861338462
ではサーボとのネジ締結部分からの歪みが報告されていました。

aiba0921, 2021-09-03:
下記最新版の上腕_サブ_フレーム_1_1のデータのリンクです。
https://drive.google.com/file/d/1_VzazTTsuMYzO8fWmDGXecpo3bxB45qn/view?usp=sharing

aiba0921, 2021-09-04:
下記最新版の上腕_サブフレーム_1_1、上腕_メインフレーム_1_1のデータです。
https://drive.google.com/drive/folders/1982YnKWhxFMXXBl-jlNbW4MxDRxqdoru?usp=sharing

MasatoKubotera, 2021-09-06:
@aiba0921 

上腕_サブフレーム_1_1において，構造解析のftesファイル参照エラーが発生しました．
[image]

また，ダイレクト編集が行われていたため寸法が確定していませんでした．
[image]

上腕_メインフレーム_1_1の設計仕様（XY平面に描かれていないこと）やSideBumperが上腕.iamを参照しており，入れ替える，最初からiamを作る子をしましたがエラーが解決できませんでした．そのため，ドライブで共有していただいたしていただいたiptについては使用できませんでした．

ver4_0ブランチに反映するための解決案としては，上腕とSideBumperをすべて再設計してアセンブリまでしていただく必要があります．
また，上腕_メインフレーム_1_1を整えて再度作成する必要があります．

aiba0921, 2021-09-07:
修正した上腕の各パーツのデータをドライブに載せました。
https://drive.google.com/drive/folders/1qVQ9rSJ6Exh4YuE854BTzq6EIJg_BTbz?usp=sharing
本日の行った大きな変更点としてはサイドバンパーの赤丸で囲た部分の消耗が激しいとのことだったのでこの部分を中心に再設計しました。
[image]
[image]

MasatoKubotera, 2021-09-07:
@aiba0921 
昨日と同様に，構造解析のftesファイル参照エラーが発生しました．
構造解析のファイルをフォルダ内に格納するか，解析を反映していない場合は，スタディを削除してください．

SideBumperについて
日本語名称への変更は避けるようにお願いします．
今後，全て英語名に変更予定です．
また，大幅に形が変わる場合以外で名称の変更をするときには相談してください．

アセンブリを開くときに**サイドバンパー3.ipt**がないと警告が出ました．
サイドバンパー.iptに参照変更することで開くことが出来ましたが，共有前に共有したファイルだけで開けるかどうか検証お願いします．

サイドバンパー.iptｎのスケッチ1ですが，「派生」が使われていないために角度拘束，寸法拘束に無理があるかと思います．
[image]

干渉は確認されなかったので，良いかと思いますが今後変更されることもありますので，余裕があれば派生を用いてみてください．
[image]

aiba0921, 2021-09-08:
@MasatoKubotera 
構造解析のスタディの削除、サイドバンパーからSideBumperへの名称の変更、アセンブリの修正を行ったデータをドライブに載せました。
https://drive.google.com/drive/folders/1K7r2j95KA0W8HWnTMFHFwm7CkPGOKTvS?usp=sharing

MasatoKubotera, 2021-09-08:
@aiba0921 
参照エラーについては確認できませんでしたので，
>構造解析のスタディの削除

については対応できていました．
>サイドバンパーからSideBumperへの名称の変更

も確認しました．

引き続き作業お願いします．

aiba0921, 2021-09-08:
スケッチ派生したSideBumperと対応させた各パーツのデータを載せました。
https://drive.google.com/drive/folders/1HgDEU2xGC0m2vlXlL7RU2urKmpdraB-M?usp=sharing

aiba0921, 2021-09-10:
SideBumperのスケッチにおける拘束で変更があったので修正します。（数か所に固定拘束を使用していたため）

aiba0921, 2021-09-11:
@MasatoKubotera 
SideBumperの修正しました。
https://drive.google.com/drive/folders/19SYjLh7Eur3nc1J8V5jN8h2sc6DrL2u8?usp=sharing

MasatoKubotera, 2021-09-15:
@aiba0921 
修正ありがとうございます．
大変遅くなりましたが，上記で共有していただいた部品３点についてver4_0ブランチに統合しました．

[image]

MasatoKubotera, 2021-09-15:
サーボフリーホーンの拘束ミス

[B3M フリーホーン側の設計ミス #138](https://github.com/citbrains/OpenPlatform_ver3_0/issues/138)のissue内で提言していたフリーホーンの拘束について，ver3.0で肩ロール，腕ピッチの２か所で拘束が間違っており，-0.2mm 幅が狭いことが分かった．

Ver4.0では拘束並びに，ホーン間幅について修正している．
幅については，下図の青で選択された２つの部品を修正している．
[image]

Ver3.0 
[image]
[image]

Ver4.0 
[image]
[image]

MasatoKubotera, 2021-09-15:
下図で青色に選択されている**Shoulder_vertical_frame_02**(旧名：肩_サブフレーム_1_1)について，Ver3.0ではA2017 T1.5部材を用いていたが，A2017 T2に変更している．

変更理由は，[163機整備記録](https://github.com/citbrains/OpenPlatform_ver3_0/issues/2#issuecomment-882180369)で大きく変形していることが報告せれていたこと，T1.5部材の廃止である．

これに合わせて，**Shoulder_vertical_frame_01**(旧名：肩_メインフレーム_1_1)も変更している．

[image]

MasatoKubotera, 2021-09-17:
ミーティングでの共有が終了したのでcloseします．

closed 2021-09-17 by MasatoKubotera