SUSTAINA-OP/SUSTAINA-OP #76: 胴体_メインフレーム_1_12の再設計

hiroki-0001, 2021-06-20:
SLCG5.5-6を追加工しない場合、以下の写真で示す雄ねじ部分が干渉するため、胴体_メインフレーム_1_12に雄ネジが干渉しないように設計する。

[image]

AD58-3104, 2021-07-09:
こちらは主に圧縮力が掛かる部分ですが、ねじの頭が干渉している部分はパーツ全体の圧縮への強さにはあまり寄与していない箇所なので強度等を考慮せずに雄ねじの頭の部分だけ削れば良いと思われます。(作業者の方へ)

hiroki-0001, 2021-07-23:
MIRパーツを作ると、組み立て・予備部品の問題でめんどくさくなるため、MIRパーツを作らないように

MasatoKubotera, 2021-07-30:
上記で指摘したスペーサについては，追加工を必要としない代替品が見つかった．
https://github.com/citbrains/OpenPlatform_ver3_0/issues/127#issuecomment-889121201

MiyawakiKaito, 2021-08-23:
この問題は胴体メインフレーム_1_12の厚みを変えればいいのでしょうか。何も聞かずに進めてしまっているので、もし間違っていたら教えてくださるとありがたいです。

AD58-3104, 2021-08-24:
既にコアタイムの方で答えて貰っているとは思いますが一応...。
全体の厚みを削ってしまうと流石に強度的に問題があると思うのでやらない方が良いです。方針としては干渉付近のみ厚みを削る、もしくはねじの頭の干渉を避ける為の穴の形を変える(穴を大きくするなど)等が考えられます。理想としては強度があまり変わらず、干渉が無くなるのが良いです。

MasatoKubotera, 2021-08-25:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/76#issuecomment-889792904 で胴体_メインフレーム_1_12に干渉しないスペーサーが見つかったため，作業を行う必要性がなくなりました．
そのため，このissueは閉じます．

closed 2021-08-25 by MasatoKubotera

MiyawakiKaito, 2021-08-25:
@AD58-3104 
申し訳ありません。ご返答いただいているとは気づかず、またissue自体が作業する必要がないものだったため、コメントの方を消してしまいました。今後同じような問題が起きた時はそのような方針をとらせていただきます。