SUSTAINA-OP/SUSTAINA-OP #77: 胴体_サブフレーム_1_3の修正

hiroki-0001, 2021-06-20:
胴体_サブフレーム_1_3のはめあいの逃げ加工がおかしい。またコックピットの再設計により飾りネジを使用する場所も異なるので、そこも修正する。

[image] 

[image]

FujisawaShuuichirou, 2021-07-03:
このissueを担当します。

FujisawaShuuichirou, 2021-07-19:
[image]

cad図修正しました。

FujisawaShuuichirou, 2021-07-24:
修正したものgoogle ドライブにアップロードしました。
リンク：https://drive.google.com/drive/folders/1_wHLrMVLHdFm5pgbswUhiH7Pb6QKehVh

NaokiIwasawa, 2021-08-18:
# 再修正

修正されたものを確認したところ、元からスケッチにエラーがあったので直しました。

## エラー内容
穴を基にしたスケッチがあったが、その穴が消されエラーを出していた
## 対処
押し出し後の面からのスケッチされていたので、胴体_基礎_1_3で同様のスケッチを書き直した。

[image][image]
左：修正前の余分な穴　右：修正後
[image][image]
左：エラーがあったスケッチ　右：問題の後書きスケッチを胴体_基礎_1_3に移行

NaokiIwasawa, 2021-08-18:
修正が終わったのでcloseします
@FujisawaShuuichirou 修正お疲れさまでした！ありがとう

closed 2021-08-18 by NaokiIwasawa