SUSTAINA-OP/SUSTAINA-OP #78: 胴体_メインフレーム_1_1の再設計

hiroki-0001, 2021-06-20:
胴体_メインフレーム_1_1の写真の4点固定の内の1点の固定部分の面積が小さすぎる。(写真、赤丸部分)　また、歪みも大きいため、肉抜きが適切でない可能性が高い。

[image]

MasatoKubotera, 2021-06-20:
このissueは，<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/45">スラストベアリングの改良 #45</a>と<a href="https://github.com/citbrains/OpenPlatform_ver3_0/issues/25">GankenKunの保管方法が妥当か検討する #25</a>に関連性がある．

AD58-3104, 2021-07-09:
こちらのissueを担当します。モデルの修正作業は順番があると思いますのでまずは有限要素解析から行っていきます。パーツのゆがみは長辺方向に反る形のものであったのでその方向に対する曲げ応力に強い形状を探す予定です。

Keita1Mori, 2021-09-06:
# 最終案

[image]

## 変更点
- 以前問題になっていたスペーサと接する半径を大きくした。
[image]

- 胴体_メインフレーム_1_1の両側もスペーサを取り付けた際に出っ張りがあるためはみ出さないように変更した。
[image]

- スペーサの固定をおネジだけで支えるのではなくスペーサの面を含めて固定するとより強度が増すと思いスペーサ取り付け部の穴の周りを0.5mm削りました。
[image]
- 機体解体時の調査で胴体_メインフレーム_1_1の真ん中が曲がっていたため中心の円の部分をハニカム構造に変更した。
[image]

MasatoKubotera, 2021-09-07:
@Keita1Mori 
はにかむ構造への変更・検討ありがとうございます．
上記でのハニカム構造の実現は難しいと思われます．
理由として，切削輪郭が多く（長く）切削に時間を要すことが挙げられます．（ファイルが共有されていなかったため，切削時間検証はできませんでした．）

また，ハニカム構造を取り入れる場合，部品中央全体を行った方が良いかと思います．分解調査では，真ん中からくの字に曲がっていたので．．．

内角については，フィレット加工をお願いします．切削するときに過度にエンドミルが入らない場合限界まで切削されますが，質量や構造解析で正確な値が取れません．

>スペーサの固定をおネジだけで支えるのではなくスペーサの面を含めて固定するとより強度が増すと思いスペーサ取り付け部の穴の周りを0.5mm削りました。

スペーサについて追加工しない[MSU-306-03](https://jp.misumi-ec.com/vona2/detail/221006202577/?searchFlow=combo2products&KWSearch=MSU-300-03&HissuCode=MSU-306-03&PNSearch=MSU-306-03)に変更されましたが，おねじ部分は3mmです．サーボモータ，部品と干渉するところについてはポケット加工が出来ないものかと思います．
アセンブリまで行い検証お願いします．

MasatoKubotera, 2021-09-09:
@Keita1Mori 
Googleドライブで共有されたファイルをver4_0に反映しました．
反映する際に，スペーサーのポケット加工は対応するスペーサが提示されていなかったため埋めました．また肉抜きについてはコックピット幅の拡大に伴い多少変更しました．ハニカム構造の肉抜きに関しては，今後も要改良箇所です．

yasuohayashibara, 2021-09-10:
ちょっと余談にはなるのですが，ハニカム構造はカーボンで挟んであげると，軽くて頑丈になります．
実はxegaの足先はハニカムサンドイッチ構造になっています．（今はもしかして違うかも．．．）
https://citbrains.studio.site/45

何度も曲がって歩けなくなり困っていましたが，上記の構造にしてから全く問題が発生していません．
ただし，自作するのは無理ですので，あくまで参考までということで．

AD58-3104, 2021-09-12:
フィレットに関してですが、加工面以外にもメリットがあります。基本的に内側の角はフィレットをつける方が応力集中が少なくなり強度が増します。(今回のハニカムだと小さすぎて対して変わらないとは思いますが)

Keita1Mori, 2021-09-18:
# 報告
## 設計案
スラストベアリングの幅の拡大により胴体_メインフレーム_1_1とスラストを固定するための六角スペーサの位置変更を行いました。
[image]
指摘されたハニカム構造の六角のサイズを4mm⇒6mmに変更しました。

MasatoKubotera, 2021-09-18:
@Keita1Mori 
中央部分について，以前の形状を残すことは考えなくてよいです．
中央部分すべて均一にハニカム構造を導入できればと思います．

yasuohayashibara, 2021-09-19:
最終的にどのような形になるかが想像がつかないところはありますが，上記の構造は中心部分の強度が低いことが懸念されます．これも有限要素解析をすると明確にはなりますが，従来曲がっている場合は改善しないおそれがあります．
（鋭角部に丹念にRをつけると改善すると思いますが，丸穴を開けるのと変わらなくなっていきます．）

あと，おそらく今までの努力はロボットを作りやすくするためのものだったと思いますが，上記の構造を切削するためには，かなりの時間が必要となり，多くの努力が要求されることが予想されます．バリを取る作業も大変だと思います．切削で気になるのは，１つ１つの穴が小さいため，外形を切削するのではなく，ポケット加工しなければならなくなるおそれがあることです．穴を全て削り取ることになりますので，膨大な時間を要する加工になるおそれがあります．

どこかで議論したような気もしますが，穴を切削する場合，中心に小さな柱が残るような加工をすると，それが倒れてエンドミルに絡みついてエンドミルが折れることがあります．両面テープで固定していますので，小さな部品は切削途中で外れることがあります．これがエンドミルに絡まるとエンドミルが破損します．

これを防ぐためには全てを削り取ることが必要で，ポケット加工せざるをえなくなります．エンドミルの破損は経済的な問題もありますが，セットアップを一から行わなければならなくなりますので，時間的なロスがより深刻な問題となります．
あと，途中から始めるときは，パスの再制作をしなければならなくなるときもありますので，完成させるために多くの手間が必要になることも予想されます．

MasatoKubotera, 2021-09-25:
こちらの再設計については，[スラストベアリングの改良 #45](https://github.com/citbrains/OpenPlatform_ver3_0/issues/45)と大きく関係があり，クロスベアリングの導入により形状の変化の可能性があります．

MasatoKubotera, 2021-09-29:
[スラストベアリングの改良 #45](https://github.com/citbrains/OpenPlatform_ver3_0/issues/45)にてクロスベアリングの導入を行っている．
胴体_メインフレーム_1_1についても，このクロスローラベアリングにあった形状に変更した．この時に，クロスローラベアリングを固定するためにカレイナっとを片脚8個ずつ圧入する．また，板厚もT3からT4に変更した．

前に議論されている肉抜きについては，ハニカム構造は切削時間の面からみて今回は見送る．

[image]

yasuohayashibara, 2021-09-29:
見た目での印象ですが，抜きすぎかなという気がします．
こちらも構造解析してみないと分かりませんが，おそらく比較的大きな力が加わる部分ですので，曲がったり振動したりするかもしれません．
この部分は塑性変形だけでなく弾性変形（振動）も気にしてあげたほうが良いところです．

あと，T4は他でも使用しますでしょうか．
ver. 3.0では使用していなかったと記憶していますので，材料の種類が増えるのは避けたほうが良いかもしれません．
抜く部分が多ければT5でも良いかもしれません．
なお，再設計が多くなる場合は，無視していただいて構いません．

MasatoKubotera, 2021-10-19:
再設計が終了したためcloseします。