SUSTAINA-OP/SUSTAINA-OP #79: リンク筋交いの使用材料について

hiroki-0001, 2021-06-20:
リンク_ふくらはぎ_サブフレーム_1_2(予備案)の材質をPOMにしても問題ないかどうか検証をおこなう。
[image]

MasatoKubotera, 2021-06-20:
https://github.com/citbrains/OpenPlatform_ver3_0/issues/38
上記issueと関連性が高いため参照すること．

MasatoKubotera, 2021-06-20:
この検証については，過去に行われているため過去のissueを見返して判断する必要がある．

MasatoKubotera, 2021-07-08:
使用部材により軽量化が可能である．

aiba0921, 2021-08-23:
GankenKun163機 整備記録 #2 
の内容において過去に実際にPOMのパーツが使われていたという記録がありました。
[image]
[image]

yasuohayashibara, 2021-08-23:
当時の担当者（OBの中島君）が試しに作ってみたとのことでした．
十分な評価はできていませんが，壊れて使えなくなったということは無かったようです．

aiba0921, 2021-08-23:
過去の整備においては完全に壊れたことはなくてもガバ穴になってしまい使えなくなったことはあるようです。

yasuohayashibara, 2021-08-23:
ネジ部に集中的に応力が加わる構造になっていますので，その部分だけ強化するようにフランジ付きのインサートナットを入れるという方法があるかもしれません．
ただし，大変なので数gのことであれば，従来どおりで良いかもしれません．
この部品はワイヤー放電加工できますので，アルミの場合でも本学であれば手軽に作れます．

aiba0921, 2021-09-02:
メカ班で議論し、インサートナット化が技術的、厚み的に不可能であるため使用材料はアルミに決定しました。よってcloseします。

closed 2021-09-02 by aiba0921